rsta2/circle #388: AARCH64 bootloader and unaligned memory access

toptensoftware, 2023-06-29:
I'm working on a new version of the flashy bootloader and was testing it with aarch64 and came across an issue where aarch64 builds would randomly work vs not work.  eg: just putting in a dummy no-op function that wasn't called would make the program start working.

I think it turned out to be an issue with unaligned memory and the fact the MMU isn't enabled when the bootloader runs.  I say "I think" because I'm by no means an expert on ARM architecture, but after adding `-mstrict-align` to the AARCH64 build the problem seems to have gone away.

Question: 

* Does this sound correct and perhaps the current version of the bootloader is only working by sheer luck?
* Should the aarch32 version have a similar `-mno-unaligned-access`.

rsta2, 2023-06-30:
Yes, it sounds correct. The problem is, that the bootloader runs without MMU and caches enabled and unaligned memory accesses are not allowed in this mode. Newer GCC versions generate unaligned accesses under some conditions. I think, you should add both options.

toptensoftware, 2023-06-30:
Thanks for the insight Rene.  

The new version of Flashy is basically a complete re-write with lots of new features and improvements. I've put it into a separate repo and will publish it as an npm package to make it easy to use.  I'd already added the switches in that that project.

For the old bootloader, I've created PR #389 in case there's future changes to it and this issue pops up again.

rsta2, 2023-06-30:
Thanks Brad for the PR! I'll need some time left to check out your new bootloader version. I've answered your email.

rsta2, 2023-07-01:
Flashy v2 is supported in Circle on the *develop* branch as a separate project.

closed 2023-07-01 by rsta2