rsta2/circle #390: aarch64-none-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory

AlexandreBourrieau, 2023-07-03:
Hello,

I have some some difficulties to make file using WSL on windows.

I have : 
 - installed WSL2 + ubuntu 22.04.6 ;
 - Update & upgrade
 - Dowloaded "arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf" and extract it
 - Copy all the file from arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf/bin to /usr/bin

after that, when i go to cicle-master path and execute ./makeall clean I got : 

```
    CLEAN  /mnt/f/Dev Elec/Raspberry/DAQ_/circle-master/tools
aarch64-none-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
aarch64-none-elf-g++: fatal error: cannot execute 'cc1plus': execvp: No such file or directory
compilation terminated.
```

Can someone help me please ? Thaks you !

rsta2, 2023-07-03:
Personally I never used WSL, but I've heard, that filenames and paths, which contain white spaces, may cause such problems. You could try to put the toolchain and the Circle project into directories with paths, which do not contain white spaces and start again.

AlexandreBourrieau, 2023-07-03:
Thanks I will try that.

AlexandreBourrieau, 2023-07-11:
Hello, not really but I used another toolchain in windows with Pycharm Clion and mgywin64 and it worked. I can compile 64 and 32bits. But I needed to use a old toolchain because with more recent one I got same error...
Thanks you for your help anyway ;)

closed 2023-07-11 by AlexandreBourrieau

rsta2, 2023-07-12:
Thanks for info.