rsta2/circle #393: Pi Zero 2 W failing to Initialize WLAN

DerEffi, 2023-07-13:
Hello,

First of all thank you for providing this environment to easily develop bare metal apps for the pi.

I'm currently running some smaller tests on my Raspberry Pi Zero 2 W for wifi connections and did not manage to get my app or even the hello_wlan sample to start properly.
I've downloaded the firmware and hostap and built the addons SDCard, fatfs and wlan without errors but when I put the kernel image on the sd card I get the following output in repeat:
```
logger: Circle 45.2 started on Raspberry Pi Zero 2
00:00:00.66 timer: SpeedFactor is 1.51
00:00:01.80 dwhci: No device connected to root port
00:00:01.83 sdhost: emmc1: sdhost-bcm2835 loaded
00:00:02.34 emmc: Found a valid version 3.0x SD card
00:00:02.48 wlan: ether4330: chip 43430 rev 1 type 1
00:00:02.94 wlan: ether4330: firmware ready
00:00:02.94 wlan: ether4330: addr E4:5F:01:XX:XX:XX
```
After that the pi reboots due to `m_WLAN.Initialize()` returning false.

Are there any known issues on the Zero 2 W or did I forget something?

Thanks in advance
DerEffi

rsta2, 2023-07-13:
Have you tried this with Raspberry Pi OS in the same hardware configuration? What is odd here is, that the system reboots. This should not happen, even when an error occurs. Just an idea, but can it be that your power supply is not strong enough? The RPi needs some more power, when the WLAN is connected. There are no known issues here and I tried my RPi Zero 2 W again with the *hello_wlan* sample without problems.

DerEffi, 2023-07-13:
Hi Rene,
thank you for the fast answer.

tl;dr: Yes it was a power issue, sample is now working.

I thought for some reason, since the pi zero 2 doesn't draw that much power (even under load barely more than 500mA) it would be enought to connect it to a USB port. But I probably shouldn't power it through the TTL Adapter. I will set it up properly now...

closed 2023-07-13 by DerEffi