rsta2/circle #399: release-45.2 usb samples compile with errors

ArnoldUK, 2023-09-07:
I'm not sure if it is this release only but worked perfectly before. All usb samples (keyboard,mouse etc..) are not compiling due to undefined references in `dwhcidevice.cpp:736` references.
```
arm-none-eabi-ld: ../../lib/usb/libusb.a(dwhcidevice.o): in function `CDWHCIDevice::TransferStage(CUSBRequest*, bool, bool, unsigned int)':
/home/pi/projects/circle/lib/usb/dwhcidevice.cpp:736: undefined reference to `CScheduler::Get()'
arm-none-eabi-ld: /home/pi/projects/circle/lib/usb/dwhcidevice.cpp:736: undefined reference to `CScheduler::Yield()'
make: *** [../../Rules.mk:209: kernel.img] Error 1
```

rsta2, 2023-09-08:
These samples have to be compiled without the system option `NO_BUSY_WAIT` defined. `NO_BUSY_WAIT` can be used only, when the scheduler is in the system, which is not the case here.

ArnoldUK, 2023-09-08:
Yes, I figured it out now and undefined the NO_BUSY_WAIT option. Compiles without errors now, thanks.

closed 2023-09-08 by ArnoldUK