rsta2/circle #410: GetClockTicks without wrap

ThexXTURBOXx, 2023-11-13:
The current implementation of `CTimer::GetClockTicks` may wrap.
However - at least when using the physical counter - this could be simply avoided by also using the value of `ARM_SYSTIMER_CHI` which already exists `bcm2835.h`.
When using the physical counter, a simple implementation (inspired by [LLD's kernel](https://github.com/rockytriton/LLD/blob/main/rpi_bm/part17/src/kernel/timer.c#L34-L45)) could look like this:

```cpp
u64 CKernel::GetClockTicksHiLo() {
  PeripheralEntry();

  u32 hi = read32(ARM_SYSTIMER_CHI);
  u32 lo = read32(ARM_SYSTIMER_CLO);

  // double check hi value didn't change when retrieving lo...
  if (hi != read32(ARM_SYSTIMER_CHI)) {
    hi = read32(ARM_SYSTIMER_CHI);
    lo = read32(ARM_SYSTIMER_CLO);
  }

  PeripheralExit();

  return static_cast<u64>(hi) << 32 | lo;
}
```

I am just pointing this out because I was wondering why something like this is not offered yet by circle.
It is probably because of a missing counterpart when not using the physical counter (?)
If you want me to create a PR, I can gladly do so.

rsta2, 2023-11-13:
Probably nobody asked for this so far. ;) Yes, please create a PR. Branch off 
from the develop branch. The new method in CTimer should have this prototype:

static u64 GetClockTicks64 (void);

It should support all Raspberry Pi models and all system options, which are 
supported by CTimer::GetClockTicks().

rsta2, 2023-11-14:
PR #411 implements this feature and has been merged to the *develop* branch.

rsta2, 2024-02-28:
This function is in Circle 46.

closed 2024-02-28 by rsta2