rsta2/circle #419: Question about log when testing with Qemu

endinferno, 2023-12-18:
Recently I am trying to figure out how network driver works in raspi3B+, so I want to add some logs when testing with Qemu.
I hope that it can directly print to screen, but I find that Clogger is not working when I run the sample programs.
So how to make CLogger work when I run the sample programs?

rsta2, 2023-12-18:
The logger should work with QEMU with the samples without modification. You 
need a VNC viewer to see the output (connect localhost:5900). The most simply 
example, which uses the logger, is sample/03-screentext. There is a small 
delay before the output starts, while on a real RPi the Act LED blinks. If you 
cannot get it working, I need more info (QEMU version and command line, 
toolchain version, Circle configuration).

rsta2, 2024-02-05:
Closed due to inactivity.

closed 2024-02-05 by rsta2