rsta2/circle #420: Help with setting PREFIX

tastycode, 2023-12-21:
I'm new to certain tooling stacks and I'm having trouble understanding the instructions for compiling circle. 

I have tried several variants of `PREFIX` and `PREFIX64`, but it tells me the prefix is invalid no matter what. I'm also confused about the trailing dash, in this scereenshot below, you can see that none of the releases have a - after the words `eabi` or `elf`

It would really help if you provided examples of the package names one might have downloaded and what the corresponding prefix for them, as well as clarify PREFIX vs PREFIX64. I assume that this is for 32bit raspberry pis vs 64bit raspberry pis and the value of the toolchain needs to conform to the building machines architecture. 

I'm excited to build some cool things with Circle!

rsta2, 2023-12-21:
Here we go. You go to the following website:

https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

On this website you go down to and open "Downloads: 12.2.Rel1", because this is still the recommended toolchain for Circle. You look for the host/target combination, you want to use, for example:

* arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi.tar.xz (for x86_64 Linux host and AArch32 bare-metal target)
* arm-gnu-toolchain-12.2.rel1-x86_64-aarch64-none-elf.tar.xz (for x86_64 Linux host and AArch64 bare-metal target)

The host is the system, where you do your developing on. The target is your Raspberry Pi computer. RPi 1-2 only support AArch32, RPi 3-4 (including RPi Zero 2 W) additionally support AArch64. You have to decide, if you want to build 32- or 64-bit target images.

Now that you have the toolchain archive on your hard disk, you extract it, for example to:

/home/user/tools

A new directory appears (for AArch32 here):

/home/user/tools/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi

When you look into the subdirectory bin/ of this path, you will find the toolchain binaries, for example:

/home/user/tools/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc

You strip the "gcc" from the path and that's your toolchain prefix for Circle:

```
PREFIX = /home/user/tools/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-
```

If you want to build 64-bit kernel images, you have to set `PREFIX64` instead. That means `PREFIX` is for AArch32 target images and `PREFIX64` for AArch64 target images. The architecture of your development host does not count here.

You can also use the `configure` tool to set-up your configuration (for the RPi 4, AArch32 here):

```
$ ./configure -r 4 -p /home/user/tools/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-
```

configure automatically determines, if AArch32 or AArch64 is used, and writes the right prefix variable (`PREFIX` or `PREFIX64`) to the configuration file Config.mk.

tastycode, 2023-12-21:
Thank you. I will try this out. It might be worth noting in the README to avoid using the `.pkg` method which is also available on that page.

rsta2, 2023-12-22:
The .pkg files may work too, but I didn't try this.

rsta2, 2024-02-05:
Closed due to inactivity.

closed 2024-02-05 by rsta2