rsta2/circle #423: Waveshare 15.6inch Capacitive Touch Screen

jatkins123, 2024-01-24:
Hi,

First thank you for creating Circle, it's amazing and has been very useful to me.
I'm interested in purchasing a Waveshare 15.6inch Capacitive Touch Screen but unsure as to whether it would be compatible with Circle's touchscreen support. I have found that the official RPi touchscreen works successfully with Circle but unsure about this one.
It states that touch support is provided by USB and connects to the Pi's HDMI port for display.

Many thanks,
jatkins123

rsta2, 2024-01-25:
I guess, this screen will work with Circle, but this info is only because of 
this wiki entry:

https://www.waveshare.com/wiki/15.6inch_HDMI_LCD

I haven't tried this display by myself. So don't make me responsible, if it 
does not work.

The screen requires some special entries in the config.txt file (see the wiki).

Thanks for appreciating Circle!

jatkins123, 2024-01-25:
Thanks for quick reply, I will take all responsibility if it does not work (in that case I will use it as a screen without touch).

closed 2024-01-25 by jatkins123