rsta2/circle #426: Sample "07-usbstorage" fails with Pi5

pottendo, 2024-02-03:
Hi,
trying to run _07-usbstorage_ I get
```
logger: Circle 45.3.1 started on Raspberry Pi 5 (AArch64)
00:00:00.13 timer: SpeedFactor is 7.69
00:00:00.23 xhci: HW reset failed
00:00:00.23 usb: Cannot initialize xHCI device \#0
```
on a RPi 5 / 4GB setup. 
Happens with or without `usbpowerdelay=1000` in `config.txt`
Config:
```
$ cat ../../Config.mk
PREFIX64 = aarch64-none-elf-
AARCH = 64
RASPPI = 5
DEFINE += -DARM_ALLOW_MULTI_CORE
DEFINE += -DSERIAL_DEVICE_DEFAULT=0
```
bye, pottendo

PS: here config.txt:
```
#
# Enable 64-bit mode (AArch64)
#
# This file must be copied along with the generated kernel image
# onto a SDHC card with FAT file system, if 64-bit mode is used.
#
# This file also enables the possibility to use two displays with
# the Raspberry Pi 4 and selects the required mode for the USB host
# controller of the Compute Module 4.
#

arm_64bit=1
#dtoverlay=uart0
#dtparam=uart0_console

[pi02]
kernel=kernel8.img

[pi2]
kernel=kernel8.img

[pi3]
kernel=kernel8.img

[pi3+]
kernel=kernel8.img

[pi4]
armstub=armstub8-rpi4.bin
kernel=kernel8-rpi4.img
max_framebuffers=2

[cm4]
otg_mode=1

[pi5]
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2
gpu_mem=16
kernel_address=0x80000
kernel=kernel_2712.img
usbpowerdelay=1000
```

rsta2, 2024-02-04:
Unfortunately I cannot reproduce this here. Which firmware version do you use 
in the EEPROM of the RPi 5?

BTW. The usbpowerdelay= option belongs to cmdline.txt, but this will not 
change the situation here, because the problem occurs earlier, before this 
option takes effect.

pottendo, 2024-02-04:
Hi, I retrieved the `booloader_version`:
```
$ vcgencmd bootloader_version 
2024/01/24 12:16:01
version 1aeae380dd5256134eea87a7e0367a9939f806b0 (release)
timestamp 1706098561
update-time 1706967189
capabilities 0x0000007f
$ sudo raspi-config 
BOOTLOADER: up to date
   CURRENT: Wed 24 Jan 12:16:01 UTC 2024 (1706098561)
    LATEST: Wed 24 Jan 12:16:01 UTC 2024 (1706098561)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.

```
thx, pottendo

rsta2, 2024-02-04:
Thank you. I've found a problem in the `HWReset()` function of the xHCI driver. It failed, when the controller was already running. I fixed it on the *rpi5* branch. Can you please try again?

rsta2, 2024-02-04:
I can reproduce this problem with one USB flash drive. Please give me some time 
to sort it out. Thanks for testing.

rsta2, 2024-02-05:
There is another update on the *rpi5* branch. With this all my USB flash drives do work at the RPi 5. Can you please try your USB device again?

pottendo, 2024-02-05:
hi,
with the same USB key as initially tried it still delivers
```
logger: Circle 45.3.1 started on Raspberry Pi 5 (AArch64)
00:00:00.13 timer: SpeedFactor is 7.69
00:00:00.58 xhcidev: Cannot set device address (4)
00:00:00.88 kernel: Compile time: Feb  5 2024 18:13:48
00:00:00.88 kernel: USB mass storage device not found
```
but with another one (USB3 adapter for a microSD card) it works:
```
logger: Circle 45.3.1 started on Raspberry Pi 5 (AArch64)
00:00:00.13 timer: SpeedFactor is 7.69
00:00:00.88 usbdev3: Device ven5e3-749 found (SS)
00:00:00.89 usbdev3: Product: Generic USB3.0 Card Reader
00:00:00.94 usbdev3: Interface int8-6-50 found
00:00:00.94 usbdev3: Using device/interface int8-6-50
00:00:01.37 umsd: Capacity is 59320 MByte
00:00:01.37 xhciroot: Port 3: Device configured
00:00:01.37 kernel: Compile time: Feb  5 2024 18:13:48
00:00:01.38 kernel: Dumping the partition table
00:00:01.38 kernel: # Status Type  1stSector    Sectors
00:00:01.39 kernel: 1 00     0C         8192    1048576
00:00:01.39 kernel: 2 00     83      1056768  120430592
00:00:01.40 kernel: 3 00     00            0          0
00:00:01.40 kernel: 4 00     00            0          0
```
maybe this has worked always - don't know. 
Still the problem remains with this very old USB key - not so important, but definitely a regression as this USB key works with Circle with my models Pi3 and Pi4.

You may close this issue anyway - as you can't reproduce it.
I can try to instrument circle to get more information out - let me know if this is something you want to do...

thx, pottendo

rsta2, 2024-02-05:
Thanks for testing. There seems still be a problem with the USB support for 
the RPi 5 (also see the "Supporting the Raspberry Pi 5" discussion). It works 
under some conditions, but not always. That's why I will keep this issue open. 

Because the RPi 5 USB support in Circle is new, I do not have much info about 
it. If you have more info, what happens with your older USB3 flash drive, I'm 
interested in it of course.