rsta2/circle #429: Unsuportted USB touch screen driver - Big Tree Tech HDMI7

dolence, 2024-02-10:
Hi! I've discovered a [nice display](https://biqu.equipment/es/products/bigtreetech-hdmi5-v1-0-hdmi7-v1-0?variant=39984058105954) that are mainly used in 3D printers projects using Raspberry Pi. They are affordable and have an audio jack output, which is very nice to use with MiniDexed and other audio projects. 

The thing is, touch isn't supported on Circle. Touch screen calibration tool doesn't recognize this screen, as you can see on the image above:
[image]

I knwo this display work under Linux, as it work on Klipper/Klipper Screen 3D printer firmware and also on my Ubuntu host. They don't need calibration. I managed to capture the data between the usb display and my computer using Wireshark. Is there anything I can do to try to get this working?

[usb_touch.zip](https://github.com/rsta2/circle/files/14230337/usb_touch.zip)

rsta2, 2024-02-11:
The USB interface 3-1-0, which is provided by this touchscreen, seems to be uncommon to me. But there is some chance, that it will work with Circle, if you add this interface to the USB device factory for HID devices. Can you apply the following patch and try again:

```
diff --git a/lib/usb/usbdevicefactory.cpp b/lib/usb/usbdevicefactory.cpp
index fe183b1..478a959 100644
--- a/lib/usb/usbdevicefactory.cpp
+++ b/lib/usb/usbdevicefactory.cpp
@@ -94,7 +94,8 @@ CUSBFunction *CUSBDeviceFactory::GetDevice (CUSBFunction *pParent, CString *pNam
 		pResult = new CUSBMouseDevice (pParent);
 	}
 	else if (   pName->Compare ("int3-0-0") == 0
-		 || pName->Compare ("int3-0-2") == 0)
+		 || pName->Compare ("int3-0-2") == 0
+		 || pName->Compare ("int3-1-0") == 0)
 	{
 		CString *pVendor = pParent->GetDevice ()->GetName (DeviceNameVendor);
 		assert (pVendor != 0);
```

dolence, 2024-02-11:
Hey, thank you! Unfortunately the patch didn't worked...

[image]

rsta2, 2024-02-11:
OK, here another patch (contains the previous one):

```
diff --git a/lib/usb/usbdevicefactory.cpp b/lib/usb/usbdevicefactory.cpp
index fe183b1..478a959 100644
--- a/lib/usb/usbdevicefactory.cpp
+++ b/lib/usb/usbdevicefactory.cpp
@@ -94,7 +94,8 @@ CUSBFunction *CUSBDeviceFactory::GetDevice (CUSBFunction *pParent, CString *pNam
 		pResult = new CUSBMouseDevice (pParent);
 	}
 	else if (   pName->Compare ("int3-0-0") == 0
-		 || pName->Compare ("int3-0-2") == 0)
+		 || pName->Compare ("int3-0-2") == 0
+		 || pName->Compare ("int3-1-0") == 0)
 	{
 		CString *pVendor = pParent->GetDevice ()->GetName (DeviceNameVendor);
 		assert (pVendor != 0);
diff --git a/lib/usb/usbhiddevice.cpp b/lib/usb/usbhiddevice.cpp
index eb41e5d..5cd6246 100644
--- a/lib/usb/usbhiddevice.cpp
+++ b/lib/usb/usbhiddevice.cpp
@@ -105,7 +105,8 @@ boolean CUSBHIDDevice::ConfigureHID (unsigned nMaxReportSize)
 	}
 
 	if (   GetInterfaceClass ()    == 3	// HID class
-	    && GetInterfaceSubClass () == 1)	// boot class
+	    && GetInterfaceSubClass () == 1	// boot class
+	    && GetInterfaceProtocol () > 0)
 	{
 		if (GetHost ()->ControlMessage (GetEndpoint0 (),
 						REQUEST_OUT | REQUEST_CLASS | REQUEST_TO_INTERFACE,
```

dolence, 2024-02-11:
It worked! Or kind of... 
Now the display got recognized, touch screen tool starts and I can pass through the calibration process. Tried with my LVGL app and unfortunately this happened:

[image]

dolence, 2024-02-11:
100% working after inputing the touch screen data into cmdline.txt. This display outputs 0-4096 for X and Y coordinates. Thank you very much for your time! I'm very glad it worked!

rsta2, 2024-02-12:
Great that it is working now. I committed the patches to the *develop* branch and pushed it to the repo. It will be part of the next release. Thanks for testing.

dolence, 2024-02-12:
Yes! On Circle and LVGL demos everything is running fine. Your driver update solved it. However, I'm still having those messages saying the vertical or horizontal coordinates are higher then screen resolution on MiniDexed (circle-stdlib). But yeah, it's probably on my side now. Today I will try to get it sorted.

dolence, 2024-02-13:
By any chance, could the cmdline.txt touchscreeen values not being parsed on circle-stdlib?

rsta2, 2024-02-13:
Good! Thanks for info. Yes, all cmdline.txt options must be one the first line, 
separated by a space.

rsta2, 2024-02-28:
This modification is in Circle 46.

closed 2024-02-28 by rsta2