rsta2/circle #441: How to do GPIOPin button Interrupt correctly

dirkarnez, 2024-03-07:
I have gone through some examples and documentation and came up with this:
```cpp
#include "kernel.h"
#include <circle/timer.h>
#include <assert.h>

#define GPIO_BUTTON	17	// connect a button between this GPIO and GND

static const char FromKernel[] = "kernel";

CKernel::CKernel (void)
:	m_Screen (m_Options.GetWidth (), m_Options.GetHeight ()),
	m_Timer (&m_Interrupt),
	m_Logger (m_Options.GetLogLevel (), &m_Timer),
	m_GPIOManager (&m_Interrupt),
	m_ButtonPin (GPIO_BUTTON, GPIOModeInputPullUp, &m_GPIOManager)
{
	m_ActLED.Blink (5);	// show we are alive
}

CKernel::~CKernel (void)
{
}

boolean CKernel::Initialize (void)
{
	boolean bOK = TRUE;

	if (bOK)
	{
		bOK = m_Screen.Initialize ();
	}
	
	if (bOK)
	{
		bOK = m_Serial.Initialize (115200);
	}
	
	if (bOK)
	{
		CDevice *pTarget = m_DeviceNameService.GetDevice (m_Options.GetLogDevice (), FALSE);
		if (pTarget == 0)
		{
			pTarget = &m_Screen;
		}

		bOK = m_Logger.Initialize (pTarget);
	}

	if (bOK)
	{
		bOK = m_Interrupt.Initialize ();
	}

	if (bOK)
	{
		bOK = m_Timer.Initialize ();
	}

	if (bOK)
	{
		bOK = m_GPIOManager.Initialize ();
	}
	
	return bOK;
}

TShutdownMode CKernel::Run (void)
{
	//m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);

	m_ButtonPin.ConnectInterrupt (InterruptHandler, this);
	m_ButtonPin.EnableInterrupt (GPIOInterruptOnRisingEdge);

	while (1)
	{
		u32 nNumber = m_Random.GetNumber () % 100; //0 到 4,294,967,295
		// static const char ScreenMsg[] = "Hello screen!\n";
		//m_Serial.Write (ScreenMsg, sizeof ScreenMsg-1);

		static const char SerialMsg[] = "in the loop!\n";
		m_Serial.Write (SerialMsg, sizeof SerialMsg-1);

		CString Source;
		Source.Format ("%u", nNumber);
		m_Serial.Write (Source, Source.GetLength());
		
		CTimer::SimpleMsDelay (1000);
		// char Buffer[100];
		// int nBytesRead = m_Serial.Read (Buffer, sizeof Buffer);
		// if (nBytesRead > 0)
		// {
		// 	m_Screen.Write (Buffer, nBytesRead);
		// }
	}
	
	m_ButtonPin.DisableInterrupt ();
	return ShutdownHalt;
}

void CKernel::InterruptHandler (void)
{
	static const char SerialMsg[] = "Hello UART!\n";
	m_Serial.Write (SerialMsg, sizeof SerialMsg-1);
}

void CKernel::InterruptHandler (void *pParam)
{
	CKernel *pThis = (CKernel *) pParam;
	pThis->InterruptHandler();
}
```
But the button (i connect just a push button between GPIO 17 aka pin 11 and GND) is not working, not sure whether my circuit is wrong or my code is wrong. Expected "Hello UART!" printed in serial for button press. the while true loop is perfectly working. Thanks!

rsta2, 2024-03-07:
That should be OK. I did a quick test on a RPi 5 with your code, and it works! Perhaps your wiring is not OK? Also the GPIO interrupt is triggered immediately, because you are using `GPIOInterruptOnRisingEdge`. Perhaps `GPIOInterruptOnFallingEdge` would be better. But this shouldn't be the problem here. Which RPi model are you using?

dirkarnez, 2024-03-07:
Thanks @rsta2 , i am using raspberry pi 4b, the circuit is this. no preference on `GPIOInterruptOnRisingEdge` or `GPIOInterruptOnFallingEdge`, but it seems mine does not work (even i short circuit the GPIO 17 and GND to mimic pressed buttion. 
[image]

rsta2, 2024-03-07:
Your setup looks OK. I tested your program here on a Raspberry Pi 4B again, 
and it works well. Have you tested a different GPIO pin (e.g. GPIO18)? It 
wouldn't be the first time, that a GPIO pin is broken from earlier tests.

dirkarnez, 2024-03-07:
Thx, i tried three other pins still not working, i am using headless mode, does it affect the GPIO? Did you have special settings in for example cmdline.txt, config.txt ?

dirkarnez, 2024-03-08:
[image]
It is:
```bash
logger: Circle 46 started on Raspberry Pi 4 Model B 4GB (AArch64)
logger: Revision code is c03111, compiler has been GCC 13.2.1
```
Is it possible that my [compiler](https://github.com/xpack-dev-tools/aarch64-none-elf-gcc-xpack/releases/download/v13.2.1-1.1/xpack-aarch64-none-elf-gcc-13.2.1-1.1-win32-x64.zip) (on Windows) may do something differently? this is the zip of my sd card contents [sdcard.zip](https://github.com/rsta2/circle/files/14535813/sdcard.zip), does it work for you too? Thanks!

rsta2, 2024-03-08:
You have the same revision of the RPi 4B like me, but 4GB RAM. I have only 2GB, but this shouldn't make a difference. I tried your SD card contents, and I got the same behavior like you. The GPIO interrupt does not work. But interrupts generally work, because the timer can calculate the SpeedFactor.

The recommended toolchain for Circle builds can be downloaded [here](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads). You should still use the GNU C 12.2.Rel1 based toolchain (see more at the bottom of the page). I think [this](https://developer.arm.com/-/media/Files/downloads/gnu/12.2.rel1/binrel/arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-aarch64-none-elf.zip) is the right download for you.

With the Linux version of this toolchain I build the attached *kernel8-rpi4.img*. Can you please extract it from the .zip archive and put it on the same SD card for a test? It does work here (with GPIO18).

[kernel8-rpi4.zip](https://github.com/rsta2/circle/files/14536759/kernel8-rpi4.zip)

dirkarnez, 2024-03-08:
Thanks for the patience, i have tried your binary and also my build using your recommended toolchain. Both does not work, i wonder if it is firmware mismatch (using tag "Step46") or my configure being wrong (`./configure --raspberrypi 4 --realtime --prefix aarch64-none-elf- --multicore --c++17` )🤣🤣

rsta2, 2024-03-08:
Please try configure without the --realtime option. You cannot write to serial 
or screen from a (GPIO) interrupt handler in REALTIME mode. But this does not 
explain, why my binary does not work for you. It is built without --realtime.

rsta2, 2024-03-08:
Rebuilt everything means, you did a "./makeall clean" from project root and 
"make clean" for your program before rebuild? No, the GPIO addresses are the 
same for any RAM size.

dirkarnez, 2024-03-08:
i deleted the circle repo, redo git clone circle, reconfigure and recompile everything

rsta2, 2024-03-08:
that should be sufficient

dirkarnez, 2024-03-08:
Could you please recommend some software / hardware tricks to check if the pins or the Pi itself is really broken so i can save money to buy another one 🤣🤣. Apparently it looks good since the RX / TX pins are working for serial print, and i am desperate for using circle in my final year project 🤣🤣

dirkarnez, 2024-03-11:
When i pressed the button for interrupt, the `"Hello UART!\n";` may not be printed,  "in the loop!\n"; keeps printing per period of time. But `"Hello UART!\n";` may appear multiple times later like they are from the queue.

To keep the button working more responsively, i have to press the button in high speed.

dirkarnez, 2024-03-12:
@rsta2 i see and thx so much bro! Let me close this

closed 2024-03-12 by dirkarnez

rsta2, 2024-03-12:
Great!