rsta2/circle #446: How to remove all text not from own CSerialDevice

dirkarnez, 2024-03-22:
[image]

Hi bro, when i do serial printing for my own CSerialDevice, when some error / exception happened from my code i can see irrelevant text getting printed too (like the screenshot), may i know is there any `#define`(s) or settings to block all of these? Because i am using the serial output for graph-plotting and i cannot have any other text, including raw revision code printing since Circle 46.

Also since i am using `SCREEN_HEADLESS` mode, can i also omit any screen text-printing too? it seems they are redirected to serial.

Thank you so mcuh!

rsta2, 2024-03-22:
Unfortunately I do not know, where this text is coming from. Circle does not 
send this text.

If you do not want to see the output from the logger, just delete the CLogger 
member from CKernel. If class of the system tries to access the logger then, a 
dummy instance is created, which ignores the message. But debugging is difficult 
in this case. 

Circle by itself does not access the screen or the serial interface. The 
sample programs do this, but they explicitly write on these devices and it can 
be seen in the programs, where this is done.

The Raspberry Pi 4 and 5 support more than one serial interface, so you could 
use one interface for the plotter and one for the logger there.

dirkarnez, 2024-03-22:
It is very scary that each time on error / exception, different documentation-like text comes out :P

I have tried to remove CLogger but you are right i need it for debugging, as long as the debugging text is managable. However, can i remove 	`CScreenDevice m_Screen;`? the application is in running in `SCREEN_HEADLESS` mode and i really want to get rid of all the things related to Screen, making the following code in samples suspicious XD
```Cpp
	if (bOK)
	{
		CDevice *pTarget = m_DeviceNameService.GetDevice (m_Options.GetLogDevice (), FALSE);
		if (pTarget == 0)
		{
			pTarget = &m_Screen;
		}

		bOK = m_Logger.Initialize (pTarget);
	}
```
and cmdline
```
logdev=ttyS1
```
Can this line redirecting Screen output to serial?

Thank you so much!

rsta2, 2024-03-22:
How it seems, this text comes from the *git-shell*. Because I do not know your setup, I do not have an explanation for this. Perhaps it is not triggered from the Circle application?

You can absolutely remove the `m_Screen` member and the related initialization, if you don't need it. You also don't need the `SCREEN_HEADLESS` option any more then, because it only implements `CScreenDevice` in a way, that all output operations are ignored. When there isn't a screen, you cannot write to it, and nothing need to be ignored.

`logdev=ttyS1` redirects the logger output to the (first) serial device. In case there is no screen, you can also change the code to this:

```
	if (bOK)
	{
		bOK = m_Logger.Initialize (&m_Serial);
	}
```

dirkarnez, 2024-03-23:
Thank you so much, going to edit the code.
The terminal i use is [Tera Term](https://teratermproject.github.io/index-en.html). I tried serial plotting from Arduino IDE too.

rsta2, 2024-03-23:
I tried addon/fatfs/sample with reading your file with buffer size 1024, and it 
worked without problems. Can you check the file system of your SD card? After a 
crash it may be corrupted in a way.

dirkarnez, 2024-03-24:
OMG indeed my card is corrupted haahaha, it works so good now after using another SD card. However, may i also know how to also disable this text, please? I want to format the logger message so customized plotter can distinguish logger message and ADC values. Once again thank you so much!
[image]

closed 2024-03-24 by dirkarnez

rsta2, 2024-03-24:
Good, that it works now. If you want to remove these lines, you have to edit *lib/logger.cpp*. An other way is not supported and not intended. ;)

dirkarnez, 2024-03-24:
I see! Will there be an enable / disable option 🤣?

rsta2, 2024-03-24:
Unfortunately not. Circle is not very chatty, but these two lines should be 
allowed. The third line from the timer can be suppressed by defining 
NO_CALIBRATE_DELAY in Config.mk:

DEFINE += -DNO_CALIBRATE_DELAY

BTW. The first two lines are the only one, which come from the logger directly, 
so you could filter out all lines starting with "logger:".

dirkarnez, 2024-03-24:
Ok hahahaha