rsta2/circle #454: Ability to set HDMI output refresh rate

pizzarollsroyce, 2024-06-08:
I don't believe that Circle exposes a way to set the HDMI video output refresh rate to 50hz, which I'd like for a project I'm working on. 

I tried looking for a mailbox register I could poke to do this, but I'm not able to find anything. 

I'll note that if I boot into Raspbian, I can set it to 1280x720@50hz, so the hardware appears to support this.

rsta2, 2024-06-08:
The only way to set the HDMI mode with Circle are the firmware settings in 
config.txt:

https://elinux.org/RPi_config.txt#Video_mode_options

Unfortunately this does not work on the Raspberry Pi 5 and your wanted target 
mode is not listed and thus probably not supported by the firmware.

Raspberry Pi OS uses a kernel mode video driver now, not the firmware video 
support any more, and supports different options.

rsta2, 2024-06-09:
You are welcome. Yes, the method, you have described, should work.

closed 2026-03-03 by rsta2