rsta2/circle #463: Building circle-stdlib w/Circle: This header is not available in freestanding mode.

davefilip, 2024-07-24:
Although circle-stdlib is mentioned several places in the nodes, using any of the circle-std samples throws lots of compiler errors with:

rsta2, 2024-07-24:
I can only suggest to  report this in the circle-stdlib project. The new 
Circle 47 does not apply the freestanding mode any more, when STDLIB_SUPPORT 
is 2 or greater, so this problem shouldn't happen with it.

davefilip, 2024-07-24:
Thanks Rene!  I just subscribed to their mailing list.

So I downloaded Circle 46 (I think a few weeks ago), and have been using that, so do you think that any changes between 46 and 47 might solve my problem?  Should I try that first?  Because I don't know what changed with 47, let me know if you think that might make a difference.  Thanks!

rsta2, 2024-07-24:
Circle 47 solves the single problem, you have described. Other major changes 
are described in the main README.md file of Circle or listed here in detail:

https://github.com/rsta2/circle/releases/tag/Step47

rsta2, 2024-07-25:
> Should I report this to the StdLib team or the NewLib team?  Or is this
> something that can be fixed within Circle?

Yes, please report it. It cannot be fixed in Circle.

davefilip, 2024-07-25:
Great, will do, thanks.  Closing this as it is not a Circle bug.