rsta2/circle #465: I2S simultaneous input+output

zanellia, 2024-08-05:
Hey there! Love the project - thanks a lot for putting this together!

Q: is it possible to both read from an input I2S device and write to an output I2S device at the same time?

rsta2, 2024-08-05:
Yes, simultaneous I2S input and output is supported on Raspberry Pi 1-4. You have to specify `DeviceModeTXRX` to the [constructor](https://circle-rpi.readthedocs.io/en/47.0/devices/audio-devices.html#_CPPv4N19CI2SSoundBaseDevice19CI2SSoundBaseDeviceEP16CInterruptSystemjj7booleanP10CI2CMaster2u811TDeviceModej) of the class `CI2SSoundBaseDevice` and maybe to enable the slave clock with `bSlave = TRUE` (depends on your hardware). On Raspberry Pi 5 it is currently not supported. Thanks for appreciating Circle!

zanellia, 2024-08-07:
Thanks a lot for the detailed reply! For context, we are trying to connect a high quality audio codec (AK4621) to a Raspberry 4 (or CM4), we’ll use the evaluation board supplied by AKM (AKD4621-B) where we can program all the operation modes with DIP switches without the need for specific I2C command and communication protocols, the same evaluation board does expose the most relevant signals for the I2S data, either input and output. Circle will be the bare metal framework we’ll use for the audio data processing, input to output, simultaneously.

With these assumptions here are some questions (I apologize in advance if this gets a bit involved/verbose)

1. we’ll process 2 audio input (L+R)  channels and output the result to 2 output (L+R) channels at a fixed sampling rate of 48/96Khz with 24 bits per sample, is this feasible?

2. we’ll share 4 I2S electrical signals between the AK4621 and the RPi, is this feasible?
LRCK -> PCM_FS or frame sync signal
LCRK is output from AK and input for RPi
BICK -> PCM_CLK or bit clock
BICK is output from AK and input for RPi
SDTO -> PCM_DIN or serial data input
SDTO is output from AK and input for RPi
SDTI -> PCM_DUOT or serial data output
SDTI is output from RPi and inout for AK

3. at power up, probably before the full boot of the bare metal "OS", the CODEC will start to send clocks and data to the RPi I2S: should we wait to start the CODEC only after the full boot?

5. input and output data to and from the AK4621 must be MSB first, MSB justified, the AK uses the 2nd complement format. Is this ok for the RPi?

rsta2, 2024-08-07:
You are welcome. The answers for your questions are:

1. Yes.
2. Yes. You have to configure the `CI2SSoundBaseDevice` class with `bSlave = TRUE`.
3. The affected GPIO pins 18-21 are configured as inputs on boot on RPi 4 and CM4, until the class `CI2SSoundBaseDevice` is constructed, so it's not necessary to start the CODEC late. I cannot see a problem here, when it is started, before the driver is active.
4. Yes.

closed 2024-08-22 by rsta2