rsta2/circle #469: Circle WLAN w/Raspberry Pi AP / hostapd

davefilip, 2024-09-08:
Rene,

I am having some troubles with the WiFi support in addon/wlan, using Circle as a WiFi client.  I can connect reliably to some Access Points (AP), but not others.

The complicating factor is that the immediate problem I am trying to solve is connecting to a Raspberry Pi running the full OS (e.g., Raspberry Pi OS 11 / Debian Bullseye) running hostapd, which connects exactly 50% of the time.  By that I mean that every other connection works, and every other connection fails.

So why don't I talk to the hostapd folks?  I have, but after trying a few things, they have been unable to help.

The frustrating thing is that using Circle WLAN, I am able to connect every time to a NETGEAR Orbi AP.  What I can't figure out is what the incompatibility is between Circle WAN and hostapd.

When it does connect, it is solid and reliable (I've left it running overnight, and even see console log messages about renewing the IP (I'm using DHCP).

When it does NOT connect, I get a lot of (more than 300) "CTRL-EVENT-DISCONNECTED - Disconnect event - remove keys" on the console log (see first image), and on the (RPi running Bullseye) hostapd I just see a lot of (more than 300) 'disassociated' errors with the MAC address of the RPi running Circle in the system log.  So hostapd thinks that Circle WAN is disconnected and unable to connect.

So is this a Circle problem or a hostapd problem?  Not sure, but I was not able to make progress in the hostapd forum, so I'm wondering if there is any known, or perhaps similar with yet a different AP, issue with Circle WAN connecting to certain APs?

On the Circle WAN side, I have tested and seen the exact same problem with Circle running on a RPi 3, 4 and Zero 2W.  On the hostapd side, I have tested and seen the exact same problem on both RPi 3 and 4.

Unfortunately this is a bit of a problem for me because my use case was to have a number of Circle-based RPi Zero 2Ws monitoring various things in real time, and reporting back wirelessly to a RPi running Raspberry Pi OS (with an Apache web server, Python, Java, etc.) that I can connect to with my phone.  The gist is that I want to have multiple Zero 2Ws running a Circle-based OS for real time data collection, all reporting back to one full Linux / Raspberry Pi OS.

Since it connects every other time, the reason that is a challenge is that all of these will all be headless RPis on a floating barge in the middle of a lake with no Internet connection.  So the start-up sequence has to be carefully planned and coordinated, including any reboots.

So what does work is:

Desktop -> NETGEAR AP
Circle WLAN -> NETGEAR AP
Desktop -> hostapd

And what works every other time is:

Circle WLAN -> hostapd

I'm writing for any thoughts on figuring out what the incompatibility is, since it only fails every other time, to determine what Circle WAN is not providing to hostapd to keep it happy.

However, since this is wireless -- and I can't run a wireless network sniffer (can I?) -- I am not sure how to further debug this going forward?  And I have already posted all of this on the hostapd forum, and nothing that was suggested has helped.

Any thoughts or idea?

Thanks,

Dave.

[image]
[image]

rsta2, 2024-09-08:
Hi Dave, I successfully tested Circle WLAN with some APs (FRITZ!Box, D-Link and Hama), that I had here. I never tested with Raspberry Pi OS in AP mode. I will do this, but this will take some time. Unfortunately I have currently no idea, what happens there. I ported WPA-Supplicant to Circle some time ago, but did not use Hostap. I must get familiar with the field again.

Rene

davefilip, 2024-09-08:
Thanks Rene!  I’ve successfully used hostapd for a few years on a few diffeerent RPis, connecting from a desktop (macOS), iPhone, iPad (OK, all Apple!), and from other RPIs (also running Raspberry Pi OS).  Circle was the first time I’ve had any problems, but reasonable that it was not in your test suite!

One additional note: the latest supported version of Raspberry Pi OS (Bookwork / Debian 12) replaces dhcpcd (used since the earliest days) with NetworkManager.  This is significant only in that NetworkManager can be configured as a RPi AP without installing hostapd.  However, at least from my admittedly limited testing, Circle WLAN does not seem to see or connect at all to a RPi running Bookwork as an AP (although I can still connect to it from my Apple devices).

So although this might muddy the waters, I mention it because if I could get Circle WLAN to connect reliably with Raspberry Pi Bookworm / Debian 12 / NetworkManager, that would give me a work-around.  I have been focused mostly on hostapd simply because I have used it extensively in the past, and quite frankly because I am still running my RPi OS images on Buster (Debian 10) and Bullseye (Debian 11), but I could upgrade those images to Bookworm (Debian 11) if it provided as solution.

For instructions on how I have set up hostapd on a Raspberry Pi: https://howto.colornetlabs.com/howto_wap/

And for Circle, to confirm, I am building using Step 47.  And, it should go without saying, I have not modified any of the add-on/wlan code myself.

Thanks in advance for any testing / debugging / advice you can offer regarding all of this.

rsta2, 2024-09-09:
I've already tested it with NetworkManager on Raspberry Pi OS 11. There were two problems: First Circle WLAN recognizes the AP as WPA, not WPA2. After changing the configuration for Circle WLAN to WPA, it tries to connect to the AP. An association is established, but in the 4-way WPA handshake the WLAN driver always reports an error in the fourth step. I don't know, how to continue here and will test with hostapd now. Thanks for the link to the instructions!

davefilip, 2024-09-09:
Rene,

Thanks for the testing and feedback!  Actually, my preference is for hostapd only because I’ve used it longer and had more experience with it … but most importantly I can install it on different computers, not just RPis … but the good news is that it does successfully connect 50% of the time, so the basic handshake protocol is working.  Just something failing on reconnects.

The every-other-time fail I suspect is because Circle is not doing something (or doing something) that makes hostapd think it is a ‘bad’ connection, and then blacklists the MAC address.

One more hint: if I disconnect Circle WAN (e.g., unplug the connected Circle booted RPi) and then reset hostapd before a Circle WLAN new connection (systemctl restart hostapd), then the new Circle connection seems to always succeed.

And one final clarification (minor point, but just to be clear): If I rest hostapd (systemctl restart hostapd) and am running my Circle-based OS on multiple RPis, each one can connect successfully in succession, but when any one tries to re-connect without a restart of hostapd, it will fail.

[Although you probably got this before, just wanted to be clear that resetting hostapd is not necessary before ANY new Circle WLAN connecting, just one that had previously connected.  The problem is specifically linked to the MAC address of a prior connection.]

Thanks again for any help you can provide in understanding why reconnections are failing.  Although I have not seen this problem with hostapd and any other WIFI client, it may be possible that the bug is in hostapd, but if we can understand what that is, then I can go back to the hostapd forum to get more assistance from them (hopefully either a patch or a configuration work-around).

Cheers,

Dave.

rsta2, 2024-09-09:
Dave, I got hostapd running and I can reproduce the problem. It does not happen that often, but from time to time. You find a log file attached. I'm using Raspberry Pi OS 11 on a RPi 4B to run hostapd 2.9. The Circle test program is the *hello_wlan* in the most recent version from the *develop* branch. I enabled the debug messages from WPA-Supplicant [here](https://github.com/rsta2/hostap/blob/3848d62a32d544c8601c105032f5e19f48d6f8bd/wpa_supplicant/main_circle.cpp#L44) (set it to `MSG_DEBUG`) and the event reporting from the WLAN driver [here](https://github.com/rsta2/circle/blob/6fe2064f22dbb5ed843f82502aabc3d5c2d8a38e/addon/wlan/ether4330.c#L29) (set to 1).

Rene

[wlan.log](https://github.com/user-attachments/files/16931364/wlan.log)

davefilip, 2024-09-09:
Rene,

Thanks, glad you were able to reproduce it.  Do you have any thoughts as to why in only happens sometimes and not others?  Any difference in the handshake?

I am a bit surprised that it does not happen “that often”, since I can easily reproduce every other time.

So for testing, did you do disconnects / reconnects in software?  Or did you unplug / re-plug the RPI running Circle, which is what I did?  If the former, the might provide a hint.  I always did the unplug / re-plug, since in my targeted environment, power might not be aways reliable.

I was testing with both:

========================================================
Raspberry Pi OS / Debian 10 / Buster

$ hostapd -v
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen ***@***.***> and contributors
========================================================
Raspberry Pi OS / Debian 11 / Bullseye

$ hostapd v2.9
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen ***@***.***> and contributors
========================================================

The latter matching your same configuration.

My eventual use case is to have several RPi’s running my Circle-based OS and the RPi running full Raspberry Pi OS running headless on the barge on the water, and connecting to the full Raspberry Pi OS from my phone to view results.  Therefore, not a lot of diagnostics when it doesn’t connect.  Although I guess I could just turn the activity LED on when it does connect and keep power-cycling until it turns on (which feels a bit messy?).

Any other thoughts?

Regards,

Dave.

rsta2, 2024-09-09:
Dave, I was removing the power from the client, running the Circle sample, and 
put it on to restart again. Unfortunately I have currently no idea, what is 
causing this problem. I will do more tests in the evening. I will be able to 
tell more exactly then, how often it happens.

Rene

davefilip, 2024-09-09:
Rene,

Thanks for the update.  I’ve just added the following to the XXXKernel::Initialize() for WiFi:

	if (!m_WLAN.Initialize()) {
		s_pThis->m_Screen.Write(ERR_INIT_WIFI, strlen(ERR_INIT_WIFI));
		s_pThis->m_Screen.Write(ERR_KERNEL_PANIC, strlen(ERR_KERNEL_PANIC));
		return false;
	} else if (debug) {
		s_pThis->m_Screen.Write(INF_INIT_WIFI, strlen(INF_INIT_WIFI));
	}

	if (!m_Net.Initialize(FALSE)) {
		s_pThis->m_Screen.Write(ERR_INIT_NETWORK, strlen(ERR_INIT_NETWORK));
		s_pThis->m_Screen.Write(ERR_KERNEL_PANIC, strlen(ERR_KERNEL_PANIC));
		return false;
	} else if (debug) {
		s_pThis->m_Screen.Write(INF_INIT_NETWORK, strlen(INF_INIT_NETWORK));
	}

	if (!m_WPASupplicant.Initialize()) {
		s_pThis->m_Screen.Write(ERR_INIT_WPA, strlen(ERR_INIT_WPA));
		s_pThis->m_Screen.Write(ERR_KERNEL_PANIC, strlen(ERR_KERNEL_PANIC));
		return false;
	} else if (debug) {
		s_pThis->m_Screen.Write(INF_INIT_WPA, strlen(INF_INIT_WPA));
	}

And then after each connect attempt I pull the power, count to 5, and then plug the power back in (at the USB connector).  So this will stall and not pass to  XXXKernel::Run until it gets a WiFi connection (it will keep trying for a bit, but after it fails, it never gets connected).

Following this approach, my experience is alternate pass/fail, as in:

First Attempt - Connect
Second Attempt- Not Connect
Third Attempt - Connect
Fourth Attempt - Not Connect
Fifth Attempt - Connect
… etc …

Let me know if you find out anything more, after more testing.

Cheers,

Dave.

rsta2, 2024-09-09:
Dave,

here are my test results. I did 10 tries. The RPi 4B, running hostapd, was active the whole time. For each try I attached the power, waited for the result, removed the power and started again for the next try. The results:

```
1	OK
2	ERR	Connect after 17s, IP after 32s
3	ERR	Connect after 17s, IP after 32s
4	ERR	Twice ERR, Connect after 28s, IP after 32s
5	OK
6	OK
7	ERR	Connect after 17s, IP after 32s
8	ERR	Connect after 17s, IP after 32s
9	ERR	Multiple ERR, Connect after 48s, IP after 123s
10	OK
```

4 of 10 tries were immediately successful. All tries were successful after up to 2 minutes.

I did also some tries, where I restarted hostapd between each try. All tries were immediately successful here.

I guess, the problem is, that when the WLAN client is restarted within a relatively short time, the AP does not know, that it was restarted, because it thinks, it is still connected. That's why the re-connect fails and this triggers these "DISCONNECTED - remove keys" messages. That's not nice, but actually it should not be a problem.

Rene

davefilip, 2024-09-09:
Rene,

OK, thanks for testing and sharing.

However, are you saying that when it did not connect, it would eventually connect if you let it keep trying?

If so, I did not see that, and it would just die with a no key offered error.

However, it sounds like you are saying that if I just wait long enough between disconnecting and reconnecting, it should be fine?  So maybe “counting to 5” between restarts is just too short?  And maybe I should just go and get a beer before restarting?

Regards,

Dave.

rsta2, 2024-09-10:
Dave,

yes, this was my test result for the 10 tries. When I was let it keep trying, the connect was established after some seconds.

And yes, the connection timeout on the AP is longer than some seconds. Because the WLAN client is powered off, the AP does not notice, that the client is gone and when it tries to re-connect, it will get an error, because for the AP it is still connected.

Rene

rsta2, 2024-09-10:
I think, the client timeout in hostapd is about 90 seconds. So if you wait 2 minutes, it should be enough.

davefilip, 2024-09-10:
Rene,

Thanks - yes, that seems to work, so sorry to bother you with all of this!  I probably should have thought about delayed reboots.

So the gist is that because a Circle-based OS boots so quickly, it confuses hostapd, because it thinks that another device with the same MAC address is trying to connect.  It all makes sense now.

Thanks for finding this work-round — which is to just chill and wait about 2 minutes before attempting to reboot.  A bit messy, but I can make it work!

N.B.: From my Mac to hostpd I can connect and disconnect and re-connect, etc., in a matter of seconds, which I tried when I first discovered this with Circle, and I have probably done many times previously, and that does not confuse hostapd.  But admittedly not exactly an apple-to-apples comparison, and it does take more than 90 seconds to re-boot a Mac, so I was performing an potentially invalid test.

Going forward, of course, the key will be getting Raspberry Pi OS 12 with NetworkManager, since that is the way Raspberry Pi — and in general Debian — is going.

One final quick question, only if you have a quick answer, if not, I won’t bother you further:

When using a Logitech keyboard w/Circle, I frequently get:

    dwhci: Transaction failed (status 0x202)

messages on the console.  However, it does not appear to ‘break’ anything, and I understand it is a USB timing issue?  I have tried setting usbspeed=full in cmdline.txt, which just caused my keyboard not to work.  And advice on how to get rid of these annoying messages?  I’ve seen references in some of your other posts online, particularly related to MIDI, but I haven’t found anything that works.  But again, it doesn’t appear to actually prevent anything from working.

Thanks again for all your time and help for finding the (simple if perhaps not just obvious) work around to my WiFi connection problems with hostapd.

Cheers,

Dave.

rsta2, 2024-09-10:
Dave,

the problem is, that when the Circle app is simply powered off, it does not disconnect from the AP. Circle can disconnect from the AP with [this call](https://github.com/rsta2/circle/blob/749fe99301846217439a40ac7bda547a986db0f7/addon/wlan/bcm4343.cpp#L45), but this happens only, when the instance of `CBcm4343Device` is destroyed, which is normally not done in the *hello_wlan* sample.

This "Transaction failed" message is caused by an inaccurate USB timing and can be ignored. If you want to have a better USB timing you can enable the `USE_USB_FIQ` system option, but you cannot use the FIQ for other purpose then.

You are welcome.

Cheers,

Rene

davefilip, 2024-09-10:
> Understood, but because the problem I initially described also happens with a reboot or shutdown, for which I am explicitly calling the reboot() and halt() functions (as defined in startup.h), I presume those functions are two low level to call all of the delete (::~) functions for each instantiated object.  Actually, if I remember correctly, I think I found those in the assembly code?
> I know that I described simply pulling the power plug when I first reached out, as I provided that as the simplest example to replicate the problem, but I initially found the problem after calling reboot() [as I normally don’t go around indiscriminately unplugging USB power cables].
> So to prevent the problem on a reboot() — otherwise I must always so a shutdown, wait 2 mins, and then do a power cycle — do you recommend I find the address of the CMcm4343Device device and explicitly delete it before calling reboot()?  If so, these are the devices I am seeing:
> 00 00606EC0 ready    main
> 01 00618680 ready    wifireader
> 02 0062C700 ready    wifitimer
> 03 00640CC0 ready    net
> 04 00651140 sleep    wpa_supplicant
> 05 006AF040 block    @6af040
> 06 00678400 sleep    dhcp
> 07 006B0140 block    @6b0140
> 08 006C0A00 block    httpd
> 09 006D4A80 sleep    ntpd
> 10 0081C100 run      @81c100
> So is ’net’ an instantiation of CMcm4343Device?  Or is that abstracted, and is there another way that I need to find the addressor the CMcm4343Device device?
> Apologies if this should be obvious and in the documentation, which I have spent a lot of time reading, and I have previously looked for but not found a way to cleanly shut down the network.  Feel free to reply with a ‘RTFM’ and a link to where I should start reading.
> [... cut ...]
> Nonetheless, once I’m done writing and debugging, I can also just set the log level to 1, and the messages disappear (along with lots of other log messages I’m generating while writing and debugging a bare metal OS).  So I will simply live the with annoyance until I am done writing new code, and then change the log level.

Apologies: halt(), reboot() and poweroff() all in sysinit.cpp, although still peppered with some assembly instructions (they are not in pure assembly code).  I misspoke earlier.

Also, I listed tasks and not devices, I copied the wrong bit of the output (I have a  test page that displays both devices and processes); these are the only registered devices I am seeing:

Character Device : string
Block Device     : emmc1
Block Device     : emmc1-2
Block Device     : emmc1-1
Character Device : tty1

So please disregard my last message, as I was having a bit of a brain freeze, but I am feeling much better now … ;-)

But I am interested in a way of cleaning shutting down the WiFi connection, if possible, so that reboot() is clean, which I understand requires getting the address of the CBcm4343Device  device and deleting it?

And apologies again for my last message, which contained bits of gibberish …

rsta2, 2024-09-10:
Dave,

if you use this version of [main.cpp](https://github.com/rsta2/circle/blob/master/sample/38-bootloader/main.cpp), which does not contain explicit calls to `halt()` and `reboot()`, you should be able to return from `CKernel::Run()` with `ShutdownReboot` or `ShutdownHalt` and `main()` will in turn return with `EXIT_REBOOT` or `EXIT_HALT` and because `CKernel` is instantiated locally in `main()`, the destructor of `CBcm4343Device` will be called, because it's a member of `CKernel`. This should shutdown the WLAN connection.

I know, that in most versions of *main.cpp* there is a comment, that one cannot return from `main()`, because some destructors are not implemented. In the meantime a number of destructors have been implemented and there is a good chance, that it will work now. It depends on the classes, which are members of `CKernel` in your application. Let me know, if it does not work (e.g. because you get failed assertions). We will find an other solution then. I tested it with the *hello_wlan* sample and it works there.

Rene

davefilip, 2024-09-11:
Thanks, got it!

I actually moved all of my business logic out of the kernel and call reboot() and halt() from a user command shell I’ve written … to keep the kernel relatively small and basic, a slight deviation from the Circle model / samples … but I now understand what needs to be done.

Thanks again for all your help, and I think I understand everything that I need to right now.

Cheers for writing and making available such a great stable, functional and well documented environment for writing a “bare metal” OS!  I tried a few earlier attempts that failed mostly because I spent way too much time dealing with differences in RPi hardware between models, and you’ve abstracted away all of those differences.  What you’e really provided is what most other OSs would call the HAL, or Hardware Abstraction Layer.  Along with some cool functions for managing memory and tasks.  Thank you!

rsta2, 2024-09-19:
Only serious USB "Transaction failed" errors are logged on the *develop* branch now. So these messages should not occur any more in your setup.

davefilip, 2024-09-20:
Rene,

Thanks for making that change!  Since USB timing can be a bit flakey, make sense that you only want serious errors (a.k.a., errors that could result in the loss of data) logged.

I’ve changed the subject because I might have something somewhere between an enhancement request and a Circle programming question … ;-)

As far as I can tell, and I have been looking through the code … albeit with find/grep to hopefully get to the right place, so I may be missing something … Circle (as of 47) does not appear to have an ICMP / PING “client”, e.g., to determine (which a quick time-out) whether or not another node is accessible on the network.

Of course, you do provide an ICMP / PING “server” that responds to ping requests from another computer.  But I can’t find anything to SEND an ICMP / PING request?

Can you confirm if that is the case, or am I just not looking in the right place (given how much functional is included in Circle, it seems like an obvious omission, which is not a criticism in any way, but just unexpected giving how much is already included).

If not, before I spend a lot of time trying to build it, can you confirm whether … as far as you know … NetworkLayer::Send() should be able to send an ICMP / PING packet?  If so, I admit that I need to spend some more time researching the ICMP packet format, but just wondering if you think it should be possible.

Of course I’ll need to get responses coming back, but don’t want to go too far down the rabbit hole unless I know that the Circle networking layer should be able to handle sending non-socket packets like ICMP.

And apologies if this is included somewhere, and I just haven’t found it (either in the doc or the source code).

Cheers,

Dave.

rsta2, 2024-09-20:
Dave, you are right, there is no ping client in Circle yet, and yes, you can use `CNetworkLayer::Send()` to send ICMP packets. But as you wrote, you need a method to pass through the replies. ICMP is specified in [RFC 792](https://www.rfc-editor.org/info/rfc792).

Rene

davefilip, 2024-09-20:
Rene,

Thanks for confirming both points.

Although I should have no problem sending ICMP / PING packets, I suspect — although have not yet gone deep into the code — that I will need to patch some code within the Circle network layer to handle the responses, which is what I have been trying to avoid since the start  (due to problems upgrading later on).  Which is why I would suggest / encourage — assuming that this would be of value to others, which I think it would? — adding a ping client to core Circle at some point in the future.

My use case for wanting ping is to have a “cluster” of RPis running my custom OS “discover” each other (by broadcasting MQTT messages to a shared topic), so that I can then run commands and/or draw graphics on any node in the cluster from any other node.  The challenge is that when a node goes off-line, socket based things (like HTTP / HTTPS, even just a simple ’telnet’-like socket) tend to hang.  Because ICMP is so light weight, it can tell me quickly if another node is still active.

I’ve been using MQTT long before Circle, and yes, I could write a simple MQTT ‘ping’ (call/response) as well , but even MQTT will hang uncomfortably long when nodes come on and go offline.

I honestly am very thankful for all you’ve included in Circle thus far, and I have no idea how much time you can spend on it going forward, but I hope you don’t mind me sending suggestions from time to time, along with potential use cases, to give you ideas on what you might think about adding in the future.

Cheers,

Dave.

rsta2, 2024-09-21:
Dave, I will have a look into the "ping client", when I have some time left. I'm always open for suggestions. Stay tuned!

Cheers,

Rene

davefilip, 2024-09-22:
Rene,

Sorry to write again so soon with something new, but when you get a chance, not urgent, I was wondering if you could share your thoughts about a problem I'm having with mutex and call-backs within Circle.

I have a bit of code which inherits from CMQTTClient, and is mostly based on mqttsampleclient.cpp from Step 35.

If I send ten (10) MQTT messages with a one second pause between each one, everything works as expected.  If I send ten (10) MQTT messages in rapid succession, however, the MQTT messages received get somewhat garbled (after about the 4th or 5th).

Which is probably not entirely unexpected, since the OnMessage() call-back function is not synchronized.  So I've created a CMutex at the class definition:

    CMutex *Lock;
    ....
    Lock = new CMutex(); <- Called when class is instantiated

And then at the top of the OnMessage() call-back function, I put:

    Lock->Acquire();

And at the bottom of the OnMessage() call-back function, I put:

    Lock->Release();

However, the messages still get garbled.  But here is what is really confusing me: if I remove the Release() at the bottom of OnMessage() function, all of the messages still get processed (although some are garbled).  I would expect that to block after the first Acquire() was not released.

So is there any reason why CMutex Acquire() / Release() can't be used in the same function of a call-back?

Separately, I also do an Acquire() / Release() when instantiating the class and initializing my data, and if I remove the Release() there, the OnMessage() call-back function does hang (as is expected).  So doing the Acquire() outside of OnMessage() without a Release() does work as expected.

So I am assuming that using Acquire() / Release() in the OnMessage() call-back function should cause single-thread execution.  But is there any reason why that would not work?

Does CMQTTClient create a separate Task for each call-back?  Or is it all part of the same Circle Task?

I am writing mostly to confirm whether or not my understanding of what I think should work is correct, or if I am not understanding how this should work.

Thanks,

Dave.

rsta2, 2024-09-23:
Dave, there is a bug in *mqttclient.cpp*, which causes an invalid payload delivered in `OnMessage()`, when multiple MQTT publish messages from the server arrive in one network packet. I will fix it soon. Thanks for reporting this!

One also has to set [this parameter](https://github.com/rsta2/circle/blob/749fe99301846217439a40ac7bda547a986db0f7/sample/35-mqttclient/mqttsampleclient.cpp#L35) to the maximum number of MQTT packets queue-able on receive, but because of the bug, this is currently not sufficient.

The whole MQTT client processing (including the call-backs) is running in [one task](https://github.com/rsta2/circle/blob/749fe99301846217439a40ac7bda547a986db0f7/lib/net/mqttclient.cpp#L361), and a `CMutex` instance can be acquired [multiple times](https://github.com/rsta2/circle/blob/749fe99301846217439a40ac7bda547a986db0f7/include/circle/sched/mutex.h#L38) from the same task. So `Acquire()` had no real effect in your case, but this wasn't the problem (see above).

I will let you know, when the bug is fixed.

Rene

rsta2, 2024-09-23:
OK, it's fixed on the *develop* branch.

davefilip, 2024-09-23:
Rene,

Thanks for the fix!  Yes, I discovered the bug in mqttclient.cpp after more testing,  I did try playing around yesterday with the queue size, but as you mentioned, that didn’t help due to the bug.

I also found that using CSemaphore does work for synchronization within the same Task, whereas CMutex does not.  But apparently this makes sense, because the call-back is within the same Circle Task, and CMutex works at the Task level (still trying to wrap my head around the concept of a Circle Task, which seems more like a Linux/Unix Thread than a Linux/Unix Process???).

I haven’t downloaded the latest Develop branch, and remember that going from Step 46 to Step 47 broke integration with Stephan’s newlib (circle-stdlib).  Do you happen to know — and if not I can test here! — whether what is currently on Circle Develop builds cleanly with Stephan’s latest newlib?  If not, I might need to surgically replace just mqttclient.cpp (and maybe .h?) into Step 47, which should work???

Thanks again for the quick fix!

Regards,

Dave.

rsta2, 2024-09-23:
Dave,

you are welcome.

Yes, a Circle task is more like a Linux thread, because all tasks share the same address room. Each Linux process has its own memory space. But Circle has a cooperative scheduler, not a preemptive one as Linux. 

When you checkout the *develop* branch manually in *circle-stdlib/libs/circle*, you should be able to build circle-stdlib and circle-newlib together with the latest patches. I tested this here with `configure -r 3 -p aarch64-none-elf- --opt-tls` and it worked. I'm not aware of a problem in circle-stdlib with Step47.

There is a ping client now in *test/ping-client* on the *develop* branch too.

Rene

davefilip, 2024-09-23:
Rene,

Cool - thanks again!

I have verified that the mqttclient.cpp patch works - kudos for turning around so fast!

I wasn’t sure if the Develop Branch was becoming Step 48, and might create any issues with Stephan’s newlib, which is why I asked.  Step 47 works great with his latest newlib.

Also, understood about the shared memory space between Tasks.  The cooperative scheduler is one of the reasons why I landed with Circle, as I’m building my custom OS primarily for IoT / GPIO types of projects that are too big for a microcontroller like Arduino, and don’t need the overhead of Linux, but also can’t react fast enough to real time (electrical) events on the GPIO header.  Cooperative scheduling means I only lose control of the processor when I want to.  [And of course device events like keyboard and network interrupts, but with Circle I can control them in software.]

Oh, and I also have over a half dozen RPi 1Bs and a couple of 2Bs in a drawer that are waaaay to slow for the lastest version of the Raspberry Pi OS, but are still useful with my Circle-based OS.

I’ll take a look at the ping client and let you know!

Regads,

Dave.

rsta2, 2024-09-23:
Your are welcome. Yes, I think a cooperative scheduler has advantages in a 
system, where the developer has full control, what happens. And Circle is a 
small system compared to Linux.

Rene

davefilip, 2024-09-27:
Rene,

The ping client seems to work well, but with one question: Is the assumption that only one Circle Task can ping at a time?

Although I haven’t tried playing with the “identifier” parameter - not exactly clear on what that does, or if it should always be ‘1’  - but it gets confused if I have multiple threads ping’ing at the same time (packets get received by one task and ignored if not the correct IP address (i.e., from the “other” Task), but then the other task can’t read the discarded responses).

No really a big problem is this is a limitation - I think I can work around that for my immediate need, and otherwise this seems to work well.  But just wondering if there is a way to ping different IP addresses from different Circle Tasks?

Use case is that I have a background Task that is keeping track of nearby nodes it discovered by occasionally ping’ing them, but then from an interactive shell I can’t ping something different at the same time.  Again, not a huge problem for me, not sure how often this would come up, and I really appreciate what you have provided.

Cheers,

Dave.

rsta2, 2024-09-28:
Dave,

this is only a test program, which has to be modified/extended for specific use 
cases. With the modification in lib/net/networklayer.cpp it is possible now to 
send ICMP packets and to receive any ICMP packet, which comes in. This does 
not enforce any behavior of the ping client. Also the ping client is not part 
of the network library, so you can easily modify it. 

The test program pings only one server at a time, but this can be modified. The 
Identifier allows multiple ping sessions from one computer to another computer 
at a time. The session can be distinguished using this parameter. But I think, 
you don't need this.

For your use case I think, you have to maintain a list of IP addresses of the 
computers to be polled. Then from time to time you send a ping request to each 
of the computers on this list, one after another. Then you sleep for a short 
time (e.g. one second) and afterwards you read all ICMP packets, which did 
arrive. When you have a match (IP address of sender == IP address of a 
computer, which got a ping request), you know, that this computer is alive. 
After doing this for all received ICMP packets, you can sleep for a while, 
until you check again. This can be implemented in one  background task.

Rene

davefilip, 2024-09-28:
Rene,

Thanks, and if I wasn’t clear, my question is related to currency (two simultaneous but unrelated operations).  For example, if I want to have:

Task 1: Ping 192.168.1.1 with 10 packets
Task 2: Ping 192.168.1.2 with 10 packets

If I run Task 1 by itself, I get 10 packets back from 192.168.1.1

If I run Task 2 by itself, I get 10 packets back from 192.168.1.2

Now, if I run both simultaneously (currently doing a  ->Yield() between sending packets for timesharing), using CPingClient as currently written, I will get less than 10 ping responses from either.

What I believe happens in this case is that Task 1 receives some packets from 192.168.1.1 and some packets from 192.168.1.2, so as I have written the code around it, it will ignore the packets from 192.168.1.2.

What I believe happens is that Task 2 receives some packets from 192.168.1.2 and some packets from 192.168.1.1, so as I have written the code around it, it will ignore the packets from 192.168.1.1.

My question is how to write code using the new ping client so that Task 1 only receives packets echoed from 192.168.1.1, and Task 2 only receive packets echoed from 192.168.1.2.

Is this currently possible?  And if so, how?  So when you say:

> With the modification in lib/net/networklayer.cpp it is possible now to 
> send ICMP packets and to receive any ICMP packet

Is it possible to receive only receive specific ICMP echo responses from a particular source IP, or is that not possible / logical?

Or is a “traffic cop” needed to receive “any ICMP packet” and then determine what program / Task that needs to go to, and route it appropriately?

Here is where I am showing my ignorance of ICMP.

When using TCP, I can open a socket on a particular incoming port, and only receive packets sent specifically to that port.

When using TCP, I can open a socket to a particular port on another server, and only receive packets sent back to that socket (technically another, usually random local port > 1024 that was opened when I created the socket).

The “Transmission Control Protocol” makes all of that magic happen.

I admit to my ignorance when it comes to ICMP.  Specifically, with what you have provided thus far — which I understand is just a test, which is why it is in the test folder! — can I create a “session” (for lack of a better word), so that a particular task can send ICMP echo requests on that “session”, and only receive ICMP echo packets in response to that “session”, and no other “session”.

Again, not sure if “session” is the right word, but hopefully that makes sense?

Given all of that, I fully understand and appreciate that this is still “test”, and please consider this feedback — and not a complaint — asking if that is something that is either possible with what you have currently provided, or if can be done?

Thanks, and as always I appreciate everything you’ve done with Circle.

Regards,

Dave.

rsta2, 2024-09-28:
Dave,

yes, that's why you must have a single task, which does all the receiving. You 
cannot have multiple tasks to deal with this. That is, what I wanted to 
express here:

> For your use case I think, you have to maintain a list of IP addresses of
> the computers to be polled. Then from time to time you send a ping
> request to each of the computers on this list, one after [...]

> Is it possible to receive only receive specific ICMP echo responses from a
> particular source IP, or is that not possible / logical?

No, this is not possible.

> When using TCP, I can open a socket on a particular incoming port, and only
> receive packets sent specifically to that port.

ICMP is architectural a part of the IP layer. It does not offer ports like TCP.
 
> Again, not sure if “session” is the right word, but hopefully that makes
> sense?

With session I meant, that e.g. on Linux one can call ping multiple times for 
the same target computer. One session was one of those calls for me. This is 
handled by using different Identifiers in the ICMP echo header.

> Given all of that, I fully understand and appreciate that this is still
> “test”, and please consider this feedback — and not a complaint — asking if
> that is something that is either possible with what you have currently
> provided, or if can be done?

I think, your use case can completely implemented in one task with the 
available support of the underlying network layer. Of course, it would be 
possible to deliver received ICMP packets multiple times to multiple targets 
(e.g. tasks), but this would require multiple ICMP queues and a management of 
ICMP receivers (tasks) in the network layer. The Circle network library does 
not know anything about the calling task. To introduce such management here, 
is beyond the effort, which should be spent for such a feature like 
implementing ping in my opinion.
 
Rene

rsta2, 2024-09-28:
Dave, if you like, I can provide the source code for your use case as a one task solution as a prove of concept of my modifications on the network layer. I understand now, that it is probably not that easy to implement or modify the ping code without deeper knowledge about ICMP.

Rene

davefilip, 2024-09-29:
Rene,

> Dave, if you like, I can provide the source code for your use case as a one task solution as a prove of concept of my modifications on the network layer.

No, thank you, but not necessary, you have done enough already, which I appreciate.  From your prior e-mail:

The Circle network library does 
not know anything about the calling task. To introduce such management here, 
is beyond the effort, which should be spent for such a feature like 
implementing ping in my opinion. 

I agree entirely, and this is not important enough to become a priority for Circle.

I expect it to be infrequent that I would do an interactive ping from my command shell, just occasionally for troubleshooting … my bigger issue is keeping track of nearby nodes, in a background task, which I will do occasionally with 2 or 3 pings per node, and why I really wanted ICMP ECHO.  I will probably just use a semaphore around the pings in the background task, so worst case an interactive ping would need to wait for a few extra seconds before completing.  Not that big a deal, and a much simpler solution to a small problem.

Again, I was asking if Circle could do that — without a major effort or re-write — and the answer is no, because:

	The Circle network library does not know anything about the calling task. 

And I really appreciate you managing priorities around Circle, and not getting lost down the rabbit hole for less important things like this.

My question was partly out of ignorance of the Circle networking layer, which I have admittedly not studied in depth, and when I reached out earlier to ask how to write an ICMP ECHO ping, it was one of the things that I had no idea how to do (that is, after taking a brief look at the code, how to route the ping responses back to specific tasks).

Under BIG operating systems … Linux / macOS / Windows, all of which I use … I can open a half dozen windows and simultaneously ping a half dozen different nodes, and it all works without them interfering with each other.  To be perfectly honest, until our recent conversations about this, I just never thought about how ping worked, and I just enjoyed the magic that made it work  ;-)

Just like when I tap on my phone to remote start my Subaru in the parking lot, I never think about the iPhone App Networking / WiFi / Internet Routers / Subaru Cloud / Satellite Uplink / Satellite Downlink / Waking the Subaru Computer / Interface to the Starter Motor, and all of the authentication, encryption, and authorization layers at each step, which all work seamlessly to start my car within seconds, all from a simple tap.

Well, sometimes I think about it, but not for very long … ;-)

So obviously no need to compare and contrast Circle vs. BIG OS, and obviously multi-threaded ping is not important enough to re-factor the network management layer of Circle.

So all good, and I appreciate you taking the time to explain both the features and the limitations.

Cheers,

Dave.

rsta2, 2024-10-03:
Dave,

in Circle a task has a task object, where its class has to be derived from `CTask` (only the task object of the "main" task is an direct instance of `CTask`). This task object contains the saved state of the processor registers for this task, when it is not running. Furthermore a task has a state (e.g. ready, blocked, sleeping), a name and its own stack. You can associate a user data pointer with a task with `CTask::SetUserData(ptr, TASK_USER_DATA_USER)`, but this is rarely used. That's all.

Despite of this there is nothing more for a task. Especially it does not have its own address room, so its more a thread (known from Linux), not a process. There is only one process in Circle, the entire system.

 The main task usually runs code in the `CKernel` object. So if another task wants to communicate with it, you have to hand over (e.g. to the constructor) a pointer to the `CKernel` object to that task, so that it can call methods of `CKernel`. There is no other object for the main task, which runs within Circle.

If this does not fully answer your question, I would need to know more about, what you want to do with the (main?) task (your use case).

Rene

davefilip, 2024-10-03:
Thanks Rene!  That’s most of what I knew.  I had thought of the TASK_USER_DATA_USER, but I am using that for something else (sharing a vector class that contains a queue of MQTT messages, which are shared between Tasks).

I have been passing a pointer to CTasks created from the kernel, but that has been getting messy as I have CTasks that are created several levels away from the kernel.  And I was hoping that there might be a better way (i.e., if I could get a pointer to the object from the CTask itself).  I guess I was looking for a way to exploit the flat memory model of Circle.  I know I could never do anything like that in Linux across processes, but was hoping that in Circle there might be an easier way, since the address space is shared between tasks (and in Linux, I can share pointers between threads, which is how call-backs work).

So I now understand that is definitely not possible in Circle.  I was hoping I was missing something somewhere that might have made that possible, but couldn’t find anytime in the CTask class itself.

So this fully answers my question, except perhaps, the second part, which is whether or not there is any memory somewhere at a static address — like a scratchpad — that I can write to, and which Circle would leave alone.

For example, the ATAGS at (I think) 0x100 (not sure if it is the same location for all models?) that contains stuff written by the boot loader (which I am presuming Circle will never touch).  Is there any other location like that, which I can use as a static scratchpad, and which Circle will leave alone?

I actually have considered — although if feels wrong — of using the space allocated to ATAGS, since I assume (?) that after Circle is running, in no longer uses that memory?  Although feel free to tell me why that is a bad idea.

Just trying to better understand my options for IPC (Inter-Process Communication, although really, this is more ITC, since I’m dealing with tasks and not processes).

Cheers,

Dave.

rsta2, 2024-10-03:
Dave, isn't it possible to allocate some memory using "new" on the heap and 
pass a pointer to this (shared) memory to the constructor of each task, which 
has to operate on it? That's how I would do it. Circle will never touch this 
memory again, until your app calls "delete" on it or the system is shutdown.

Rene

davefilip, 2024-10-03:
Rene,

Yup, thanks, that would obviously work!

What I am trying to avoid is having to pass the address to each constructor.

This is what I have been doing, but again, becomes a bit messy when you are two or three or four levels removed, and have to pass the address from constructor to constructor to constructor to ...

But yes, I understand, there is no other way around it, and that is the best option.  I was looking for something a bit cleaner, like any task can go ‘here’ to get a pointer to the kernel task.  Just trying to keep things clean, and since the kernel is the father/grandfather/great-grandfather/etc. of every other task, I thought there might be a way.

I might be able to do something with an extern pointer, so I may play around with that.  I’m still learning what I can do with C++.  Before June of this year, I never wrote a single line of C++, and for the past 25+ years or so have been almost exclusively on Java on Unix / Solaris / Linux / macOS / etc. (albeit dabbling with PERL / Python / bash / etc.).  The reason I stayed away from Circle for so long was because I didn’t know C++, although I had programmed C for about 15 years several decades ago, and between that and Java (which is very similar to C++ in many ways), I’ve managed to get quite a bit accomplished.  June 2024 was when I finally decided that Circle provided enough hardware abstraction on a Raspberry Pi that I had to start learning C++. 

I mention all of that only because some of my questions might sometimes sound like they are coming from a Java + Unix mindset.

Thanks, that answers all of my (current) questions.

Cheers,

Dave.

rsta2, 2024-10-03:
Dave, you are welcome. When you are ready with this, I think, we can close this issue, because it is getting very long already and covers different topics. For other users it's better to have one topic per issue / discussion, because one can easier find that way, what he/she is searching for. When you have a new issue or question, please open a new topic.

Cheers,

Rene

davefilip, 2024-10-03:
Agreed, I previously tried changing the subject, but that did not change the topic (your reply came through with the old topic).

Can you close this topic from your end?  Or do I need to figure out how to do that?

Also, as an aside, this last question was tangentially related to the original topic: it is about solving the problem of the kernel starting a command shell (my code, running as a telnet server) task, staring an MQTT task (mostly my code as a wrapper but leveraging mqttclient.cpp) receiving a ’shutdown’ or ‘reboot’ message, and having a cluster node task (my code) read the MQTT message and shut down or reboot Circle by returning to the kernel, which then causes the kernel to return to main (how you have designed Circle to shut down).

Otherwise I’ve been calling halt() and reboot() directly, which as you’ve pointed out, does not properly shut down the WiFi connection.

Nonetheless, please mark this as closed, or let me know if I need to do that (don’t know how but can probably figure it out).