rsta2/circle #471: Circle appears to fail on official touchscreen with older controller boards

SteveSpencer, 2024-09-12:
Using an older "Official Touchscreen" with a Pi3B (32-bit arch) the backlight is not correctly set by the normal boot sequence. This leaves the display appearing blank, although it is displaying correctly.

I have tested this with two touchscreens and two different Pi3Bs, (four "platforms" in all), and this appears to be consistent. Through serial logging I was able to determine that the hardware was being initialised, the touchscreen was available (as touch1) and that it was reporting touch events as expected.

I modified a sample kernel.cpp to include explicit setting of the backlight value, and this resolved the problem.

By extending CKernelOptions and making a small change to CBcmFrameBuffer, I was able to then implement a cmdline.txt solution where the user can set the initial backlight level to a specific value.

Would there be interest in putting these changes into Circle generally?

rsta2, 2024-09-12:
> Would there be interest in putting these changes into Circle generally?

Yes, please send a Pull Request. The modifications must be merged into the 
develop branch. See CONTRIBUTING.md for more info!

rsta2, 2024-09-13:
There is a new *cmdline.txt* option `backlight=` on the *develop* branch.

rsta2, 2024-11-08:
This issue has been fixed in Circle 48.

closed 2024-11-08 by rsta2