rsta2/circle #476: Support for Speedlink Competition Pro Extra USB joystick

x48x4b, 2024-09-21:
Please add support for the Speedlink Competition Pro Extra USB joystick.

See related discussion  on 

https://github.com/randyrossi/bmc64/issues/277

Thanks in advance!

rsta2, 2024-09-22:
We could run some tests on your Raspberry Pi. I send you a kernel image to be installed on your BMC64 SD card, and you take a photo from the screen and send it here, if you like. I need to know, which RPi model you have?

x48x4b, 2024-09-22:
Thanks a lot! 
Your proposal is very fine for me.
I run my BMC64 installation  on a Raspberry Pi 3B+ (is that accurate enough?)
I guess you, will tell me how to boot into this test kernel.

rsta2, 2024-09-22:
Attached is the first test program. It is the *sample/27-usbgamepad* from the Circle repo. Please rename the file *kernel8-32.img* on your BMC64 SD card to something else, so that you can restore it later, and then extract the file in the attached .zip archive and put the *kernel8-32.img* file on the SD card. Start your RPi with the SD card with the joystick attached.

What happens? Normally the joystick should be detected and when you press some button or move a control, the current state should be shown in a log line. After testing this please start again and take a photo from the log messages, displayed on initialization.

After the test rename the file, you have renamed at the beginning back to *kernel8-32.img* to restore your BMC64 installation.

If it does not work with the BMC64 SD card or you don't want to touch it, you can also clone the Circle repo, go to the *boot/* directory and enter `make` there. This will download the necessary firmware files. Put them on an empty SD card along with the *kernel8-32.img* file from the attached archive and start your RPi with this SD card.

[kernel8-32.zip](https://github.com/user-attachments/files/17089839/kernel8-32.zip)

x48x4b, 2024-09-22:
Hello  & thank you for your effort!

I did it:

1. After first boot: "IMG_20240922_171008.jpg"

2. After second boot I did not see any "log messages".

I saw a rather similar output than after first boot.
Also I operated the joystick ("IMG_20240922_171100.jpg" ) and a button ("IMG_20240922_171300.jpg"). 

In my opinion this is reflected in the screen output.

Best regards

HK
[image]
[image]
[image]

rsta2, 2024-09-22:
Thanks for testing! From your images one can see, that your joystick is 
detected and it sends status reports. We can also see some reaction in the 
shown values, when a button or axis is operated. It would be interesting, if 
all controls show a reaction in the values, which make sense. "Buttons" is a 
bitmap, where each bit refers to one button. There are four "Axes" (0-255), 
where 128 should be the default position. I cannot say, what the "Hats" value 
refers to.

What I can say is, that the mapping of this joystick is not compatible to the 
standard mapping of the PS3 or XBox gamepads for example, of course. So it 
would need a special handling by BMC64. I did read in the BMC64 Issues, that 
there are different versions of this joystick. I don't know, if they have 
different vendor/device ID combinations. If not, how should BMC64 be able to 
detect the different versions to apply different mappings? If the general 
function of the joystick is OK (as questioned in the first paragraph), there is 
nothing I can do about it. Unfortunately I also don't know, how BMC64 handles 
gamepads without standard mapping.

x48x4b, 2024-09-22:
Thanks a lot, I have commented in https://github.com/randyrossi/bmc64/issues/277

Best regards

x48x4b, 2024-09-23:
BTW: I would appreciate if you could look into my last two posts in https://github.com/randyrossi/bmc64/issues/277

One possibility would be, that "circle" detects the joystick, but does not report it in the "devicenameservice".

rsta2, 2024-09-23:
The test program, you have tried, uses the same method to [detect the gamepad](https://github.com/rsta2/circle/blob/749fe99301846217439a40ac7bda547a986db0f7/sample/27-usbgamepad/kernel.cpp#L106). If there would be no entry in the device name service, your joystick wouldn't work with it.

But I think BMC64 uses a much older release of Circle. How I can see this it's Circle 40. There may have been important changes to the gamepad drivers since then. I built the test program again with this Circle version, and you can try with it, if your joystick is also detected there. If not, it would be a problem, which was fixed in the meantime in Circle.

If you want to try this with your own SD card (not with the BMC64 installation) make sure, that you do `git checkout Step40` before running `make` in the *boot/* subdirectory of Circle. The firmware must fit to the sent kernel image.

[kernel8-32.zip](https://github.com/user-attachments/files/17094711/kernel8-32.zip)

x48x4b, 2024-09-23:
Thanks a lot again! 

I directly used your test kernel in my BMC installation, with renaming the old one etc. as yesterday.

And indeed the output is red: "Gampepad not found" (see enclosed picture).

What would you do now?

Do you think the BMC64 would make a new release?
Or should I make a new BMC64 build myself? 

[image]

rsta2, 2024-09-23:
You are welcome.

You have to ask this in the BMC64 project. In any case BMC64 must be rebuild, if you want to use your joystick. Not sure, but I guess its [this commit](https://github.com/rsta2/circle/commit/e82c08ebf8d3f1dd06af6ea929c736de602d63bb), which could fix the detection problem. Perhaps it's easier to try to only apply this patch before build, but to update the whole Circle version.

x48x4b, 2024-09-23:
Thanks a lot!

1. How and where would you apply the patch?

2. I am just trying to build BMC64. I have come pretty far with it. 

But now I am stuck with this, after lengthy build output

...
checking for lex library... not found
configure: WARNING: required lex library not found; giving up on flex
checking for library containing yywrap... no
configure: error: Could not find either flex or lex!
make[1]: *** [Makefile:495: config.status] Fehler 1
make[1]: Verzeichnis „/home/hwkramer/Downloads/make_bmc64/bmc64/third_party/vice-3.3“ wird verlassen
make: *** [Makefile:1845: ../config.status] Fehler 2

rsta2, 2024-09-23:
From BMC64 project root enter:

```
cd third_party/circle-stdlib/libs/circle
git pull
git checkout -b gamepad-fix
git cherry-pick e82c08ebf8d3f1
edit lib/usb/usbdevice.cpp        # fix the merge conflict in the file header
git add lib/usb/usbdevice.cpp
git cherry-pick --continue
```

How it looks, you need the *flex* package installed on your machine. But I'm not a BMC64 expert.

x48x4b, 2024-09-23:
Thank you!

On my Fedora 40 machine there is Flex 2.6.4. The BMC64 build expects 2.6.3. There is a known issue about. I could fix that and build a working "kernel8-32.img" from the BMC64 sources. However it seems, that I still have the old "circle" version. (I just did a "git clone" from the BMC64 repo.) 

Is your above patching procedure to make sure that I have the current "circle" version in the build path?

rsta2, 2024-09-23:
This procedure only applies the patch, referenced  above, to the Circle version, which is currently used by BMC64, in the hope that it fixes the joystick problem. Everything else remains the same. Using the current Circle version 47 is probably not that simple, because of a few potential API breaking modifications in Circle in the meantime. Please note, that I'm not familiar with the BMC64 build process. I only had a quick look, where Circle is in the source code tree.

x48x4b, 2024-09-23:
Again, I was able to build a working "kernel8-32.img" from the BMC64 sources. This time with the patched version of "usbdevice.cpp". Unfortunately, the joystick was still not detected in my BMC64 installation.

rsta2, 2024-09-23:
OK, that's not nice, but it was only a guess. There can be more relevant 
modifications in the meantime. Now it's getting complicated. As I can see it, 
they are patching several components of circle-stdlib and Circle in the BMC64 
build process. To adapt this for the new versions probably requires much effort 
and time. Unfortunately I cannot really help you here. I can only fix known 
bugs in Circle, but I'm not responsible, if newer Circle versions are used in 
a project.

x48x4b, 2024-09-24:
Hello rsta2,

Thank you for all your kind help.
I understand your role in this and that you cannot do more.
That is perfectly okay for me.
It will be interesting to see if the BMC64 developpers will address that.
Their last official release dates from January 2022.
So let's hope that the core developpers are still in the project.

Best regards

HK

rsta2, 2024-09-24:
You are welcome.

rsta2, 2024-09-25:
> Would something like that work "out-of-the-box" in Circle Version 40?
> 
> https://www.reichelt.com/ch/de/shop/produkt/raspberry_pi_-_controller_snes-d
> esign-211468 (Raspberry Pi - Controller SNES-Design)

Unfortunately I don't know. There has not been feedback for this controller.

rsta2, 2024-09-25:
I have tested by myself:

* PS3 Sixaxis controller
* XBox 360 controller clone (not on RPi 4-5)
* Thrustmaster Dual Analog 4 (no standard mapping)

jsanchezv has tested:

* PS4 controller 
* Nintendo Switch Pro controller
* XBox One controller

All controllers must be connected via USB. Wireless mode (i.e. Bluetooth) is 
not supported.

Of course, there is no warranty.

x48x4b, 2024-09-27:
```
Hello,
I have looked at the controllers that you have tested.
And I am not so happy with the general design of them. 
IMHO they are not really suited for my retro games needs.
Therefore, I think about building my own stuff based on this encoder:
https://joy-it.net/files/files/Produkte/SBC-ZDE/SBC-ZDE_Datasheet_2023-12-22.pdf
I wonder, what you think are the chances, that this would be compatible with Circle version 40.
Best regards
```

rsta2, 2024-09-28:
From this datasheet I can only see, that's an USB controller. One can guess, 
that it provides a HID interface, like most USB controllers. But it depends on 
the firmware, if it needs to set the USB configuration, before the HID report 
descriptor is read. That's one of the problems with Circle 40. No datasheet 
will tell you this. So unfortunately I cannot say, if this controller works 
with Circle 40.

x48x4b, 2024-09-28:
Thank you for your advice.
Maybe I just have to order a cheap piece from China (e.g. https://www.ebay.de/itm/314739207738)
There it also says "... The circuit is compatible with USB1.0 / USB2.0, and supports Win98 / Win Me / Win 2000 / Win XP, and through the USB communication protocol testing (USB
Check and Hid View Tests), ... ."
I guess it suffices to connect the "naked" controller to see if it is recognized.

rsta2, 2024-09-28:
If it's cheap, you wasted less money, if it does not work. 

> I guess it suffices to connect the "naked" controller to see if it is
> recognized.

I think so.

x48x4b, 2024-10-18:
Fyi: Yesterday, I have received the cheap USB joystick controller from China: I have hooked it up with some breadboard parts and it actually works with my BMC64 setup (see picture).
[image]
As it does not seem that BMC64 intends to move to recent Circle versions, I will probably have to rip apart my Speedlink Competition Pro Extra USB Joystick and rebuild it with my  cheap USB joystick controller from China.

rsta2, 2024-10-18:
It's good, that you have found a solution. Thanks for info!

x48x4b, 2024-10-20:
This is the end product. There was no room for the controller in the joystick. I put it in the housing of an old light dimmer. Now I can use my joystick with my BMC64 setup.

[image]

closed 2024-10-20 by x48x4b

rsta2, 2024-10-21:
Congrats, looks great. Happy playing!