rsta2/circle #486: Raspberry Pi 4 : MultiCore support under QEMU

stephaneweg, 2024-10-11:
hi,
i am able to run Circle on qemu using the following command: 
"c:\Program Files\qemu\qemu-system-aarch64" -machine raspi4b -cpu cortex-a72 -smp 4 -m 2G -kernel c:\temp\circle.img

it boots correctly, except that when i call the initalize method of **CMultiCoreSupport** , it can not start the other core. there is a message "CPU core 1 did not start";  this log is reported when it checks, that the secondary cores responded

Have you experienced the same issue ?  what are the correct parameters for qemu ?

rsta2, 2024-10-11:
I can reproduce this with an older QEMU version. I think your command line is 
OK. I'm afraid there is no quick solution for this. I tried sample/26-
cpustress again on a real RPi 4B 2GB with multi-core support and it's working 
well there.  It is priority for Circle, that it works on the real hardware.

stephaneweg, 2024-10-12:
I know, multi core works well on my rpi too, i was wondering if there is a solution for qemu