rsta2/circle #497: TFTP client in Circle?

davefilip, 2024-11-11:
Just want to confirm that there is no TFTP client in Circle?

If not, shouldn't be too difficult, protocol is fairly simple (hence the first 'T'), basically just numbered UDP packets.  But since there is a server in /addon/, thought I'd ask.

I understand there is no inherent security in TFTP, however, my thought was to modify by placing some authorization encoded into the filename (similar to how you can GET http://username:password@server.com/ for some web servers, although I was planning on using a rotating authorization token instead of a password).

Nonetheless, wanted to confirm that there was currently no TFTP client in Circle?  And I do understand that the initial use case for the TFTP server was for updating kernel + boot files over the network.  Thanks.

rsta2, 2024-11-11:
Right, I can confirm, that there is only a TFTP server in Circle so far, not a 
client.

davefilip, 2024-11-11:
Thanks for confirming - I will close this ticket.

davefilip, 2024-11-11:
Question answered.

closed 2024-11-11 by davefilip