rsta2/circle #507: C11 support

andreas-jonsson, 2024-11-18:
Hi!

First I just want to say a big thank you for creating this amazing library. It's been a pleasure to work with.
Although I wish it would have been written in C. ;)

Talking about C... would it be possible to bump [this](https://github.com/rsta2/circle/blob/c243194670720c8a13e6a553654b6d26ec7845a0/Rules.mk#L205) to C11/GNU11?
Alternatively have an option for it just like with C++14?

rsta2, 2024-11-18:
Sure, I added a new make variable `C_STANDARD`, because I'm currently not sure, if the VC4 code builds with C11 in all configurations. You can enable C11 by setting
 `C_STANDARD = -std=c11` in *Config.mk*.

Thanks for appreciating Circle! I like OOP, so C was not a real option. ;)

andreas-jonsson, 2024-11-22:
Works well. Thank you

closed 2024-11-22 by andreas-jonsson