rsta2/circle #511: CBcmFrameBuffer can't set 32 bit color on RPI5

andreas-jonsson, 2024-11-22:
Hi!

Using the `CBcmFrameBuffer` object and setting the color depth (and resolution?) to 32 bit on RPI5. It seems to quietly, without errors fall back to 16 bit.

`Initialize` do not return any error and the object reports back that bit depth is 32 bit when it clearly still is 16 bit.

I'm a bit new to Circle so I don't know if I'm missing something here? But the code works just as expected on RPI3&4 so I assume something is different with the RPI5?

I found this post. Not sure if it is relevant to this: https://github.com/raspberrypi/firmware/issues/1904

rsta2, 2024-11-22:
The frame buffer support in the firmware of the RPi 5 is unfortunately not as comfortable as for earlier models. This is a problem, which is caused by a non-functioning property mailbox function in the RPi 5, which Circle uses to set the display depth. Unfortunately I don't know an other programmatic way to realize this. I guess it does not exist.

Could you please try the option `framebuffer_width=32` in the file *config.txt*? I think I read somewhere, that it may work to set the depth. I have an other RPi model currently on my desk and cannot change this quickly, so I cannot test this by myself at the moment.

rsta2, 2024-11-22:
You are welcome. Sorry, I mean `framebuffer_depth=32` of course, not the width. ;)

https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#framebuffer_depth

rsta2, 2024-11-22:
Unfortunately `framebuffer_depth=32` does not work on the RPi 5. The display image is not clean. So there is currently no way to use this depth with Circle on the RPi 5. :(

andreas-jonsson, 2024-11-27:
I'm sorry. I have been pulled in to another project and I don't have a easy to test environment setup atm. But I will follow up on this.

May I suggest that setting bit depth to anything else then 16bit on RPI5 will result in an error while calling `Initialize` or at the very least the `CBcmFrameBuffer` object don't falsely report 32bit?

rsta2, 2024-11-27:
No problem.

Yes, I will implement this.

rsta2, 2024-11-28:
Added a test for depth 16 on the *develop* branch.

rsta2, 2025-03-14:
Depth 32 works with Circle with the frame buffer on the Raspberry Pi 5, when the following options are added to the file *config.txt*:

```
framebuffer_depth=32
framebuffer_ignore_alpha=1
```

wmjenkinson, 2025-03-15:
Hello rsta2, does Circle now support a 32bit color depth if I use the following in the config.txt file:

[pi5]
framebuffer_depth=32
framebuffer_ignore_alpha=1
framebuffer_width=1920
framebuffer_height=1080
kernel=kernel_2712.img`

And set the following in the screen.h:
#ifndef DEPTH
#define DEPTH	32		// can be: 8, 16 or 32
#endif

If possible, could you send me an example.

rsta2, 2025-03-16:
Yes, depth 32 should work with Step48 from master branch with your settings. 
The develop version does not work at the moment, because it blocks depth 32 on 
the RPi 5. This test will be removed in the upcoming Step49, so that it works 
there too.

You can use sample/02-screenpixel, which draws a rectangle and a cross on the 
screen with the selected DEPTH. test/screen-ansi-colors shows some text in 
different colors.

wmjenkinson, 2025-03-16:
Thanks rsta2.  Got it working.  Could I ask, what is the format of the pixel?  is it argb???

rsta2, 2025-03-16:
You are welcome. Yes it's ARGB8888. This macro creates a pixel color value from its components (0-255):

```
#define COLOR32(red, green, blue, alpha)  (((blue) & 0xFF)       \
					| ((green) & 0xFF) << 8  \
					| ((red) & 0xFF)   << 16 \
					| ((alpha) & 0xFF) << 24)
```

Alpha should be 255.

wmjenkinson, 2025-03-18:
One more question rsta2. what resolutions does the RPI5 support in the Circle framework?

rsta2, 2025-03-18:
> what resolutions does the RPI5 support in the Circle framework?

This is controlled by the firmware. I only tested 1920x1080 on the RPi 5 so 
far.

rsta2, 2025-03-20:
32 bit depth works now on the RPi5, so this can be closed.

PabloIvnGl, 2025-03-29:
Sorry i'm come for other issue. while booting UEFI in rpi 5 D model, even changing the frambuffer to 24 or 16, the rpi don go further than UEFI, tryin to instal win 11 with wor boot iso BOOT.

rsta2, 2025-03-30:
This question is not related to Circle. Please ask it here:

https://forums.raspberrypi.com/