rsta2/circle #512: CSocket and ReceiveFrom Hang

davefilip, 2024-11-22:
Rene,

I am having some problems trying to use UDP Circle Sockets, and was wondering if you could offer any advice.  I have used Circle Sockets with TCP before, but this is my first UDP attempt.

My goal is to write a program to get a file from a TFTP server.  I believe I understand the protocol (it is ‘trivial’), and I have stand-alone client versions written in C and Java that work with my TFTP server, which is running on a Mac (eventually I hope to do Circle-to-Circle TFTP, but right now taking the Circle TFTP server out of the mix to simplify things, and just focus on the client side).

So I create and open a UDP CSocket:

	CSocket Socket = CSocket(m_Net, IPPROTO_UDP);
	int status = Socket.Connect(ServerIP, TFTP_PORT);

And the status is zero (and always is, even if I pass an invalid IP address, which I guess for UDP is merely setting up the socket and not connecting to anything, as it is a connectionless protocol?).

I then create a Read Request packet:

	char buffer[FRAME_BUFFER_SIZE];
	char *ptr = buffer;

	*(short *)ptr = bswap16(TFTP_READ);	// Request read
	ptr += 2;							// Length of short
	strcpy(ptr, m_RemoteFile);			// File to transfer
	ptr += strlen(m_RemoteFile) + 1;		// Length of filename w/null
	strcpy(ptr, TFTP_MODE);			// Binary (octet) mode
	ptr += strlen(TFTP_MODE) + 1;		// Length of mode w/null

Which I then send:

	size_t length = (ptr - buffer);
	int count = Socket.SendTo(buffer, length, 0, m_Server, TFTP_PORT);

Which appears to work (see the packet capture), and the count and length are always equal.

The problem is that I get stuck on the receiving side, waiting for a response:

	CIPAddress RemoteIP;
	u16 RemotePort;

	CLogger::Get()->Write(LogSource, LogNotice, "Before ReceiveFrom");
	count = Socket.ReceiveFrom(buffer, FRAME_BUFFER_SIZE, 0, &RemoteIP, &RemotePort);
	CLogger::Get()->Write(LogSource, LogNotice, "After ReceiveFrom, count is %d", count);

On the console, I see the ‘Before ReceiveFrom’ but not the ‘After ReceiveFrom’.  If I change ReceiveFrom to non-blocking, it always returns zero (no data available) immediately.

So I started up WireShark on my Mac to capture packets, and I see that Read Request looks good, and the server responds with one data packet — transmitted 5 times because it does not receive an ACK — but the Socket.ReceiveFrom call never completes, so my Circle code just waits.

I am attaching a screen shot of the packets sent and packets received.  Do you have any idea why my ReceiveFrom is not receiving the data packets?  When comparing the packet capture to a successful (non-Circle client), I believe the data packets look good, but for some reason Socket.ReceiveFrom is not returning any data.

Any advice on what I am doing wrong, or how to further debug this?

Thanks,

Dave
[image]

rsta2, 2024-11-22:
Dave, the problem is, that the TFTP server does not send from port number 69, 
but from a new port number. The socket at the client side is bound 
("connected") to the foreign port 69 and receives packet only from this port.

I think there is currently no way to deal with this in Circle, it needs a 
modification. Because you do not know your own port of your client socket, you 
cannot create another socket, which is bound this port and receives packet 
from any port of the sender.

I have to think about this, what the best solution in Circle is. This will 
take some time.

Rene

davefilip, 2024-11-22:
Rene,

OK, glad I can always come up with interesting problems … ;-)

Not sure that I understand entirely … so in the packet samples I had provided, the File Read Request was sent to port 69:
User Datagram Protocol, Sr Port: 60003, Dst Port: 69
But from Circle’s perspective, the outgoing port (>1024) was 60003.  When the TFTP server responds, it does so back to that same port:
User Datagram Protocol, Src Port: 63654, Dst Port: 60003
So is a Circle CSocket not a bidirectional socket?  Or is it assuming that the outgoing and incoming port have to be the same?

Otherwise, just throwing something out there, if the socket is NOT bidirectional, I assume that Circle internals know what the outgoing port is for the CSocket (in this case 60003), so can call something to get that from Circle internals and then open a DIFFERENT (NEW) CSocket on that port to receive the response?  So that I have one socket for sending and one for receiving?  Perhaps not the cleanest solution, but perhaps one that might work?

Or is there no way for me to get the ‘local’ (outgoing) port chosen by Circle when it connects to remote port 69?  By Circle internals, I mean there must be some data structure within Circle that keeps track of the outgoing port numbers?

Just trying to understand.  I know from conversations with Stephan that one of his regrets — and maybe regrets it too strong a word — is that he never got Unix sockets working in stdlib/newlib.

Thanks,

Dave.

davefilip, 2024-11-22:
… for example, in socket.h I see (currently private):

	u16 m_nOwnPort;

So for an outgoing socket connection, if I had access to this (which would of course require a new public function like GetLocalPort()), is this populated with the outgoing port number, to which the TFT server would be writing back?  If so, then I could open a new socket to read the response packets.

Or is that only set by a call to Bind(), when explicitly binding a CSocket to a local port number?

rsta2, 2024-11-22:
Dave, yes, that's an interesting problem.

> Or is it assuming that the outgoing and incoming port have to be the same?

Yes, when you establish the UDP socket with `Connect()`. So this was leading me to a possible solution:

If you call `Bind()` instead of `Connect()` after creating the `CSocket` object with some arbitrary own port number (e.g. 50001) and use `SendTo()` afterwards, you can still send to the TFTP port 69 and now you can receive with `ReceiveFrom()` from any sender and port. This should work. Can you please try this. 

> I know from conversations with Stephan that one of his regrets — and maybe regrets it too strong a word — is that he never got Unix sockets working in stdlib/newlib.

No need to regret. Stephan does a great job with circle-stdlib/newlib and this socket support in it was very difficult to achieve, because the underlying Circle networking support is limited. Also the cooperative scheduler is something, you cannot hide from the user, which is a problem, when you want to port some UNIX networking code without change. Nobody can say, what will be in some years. Perhaps networking support in Circle won't be that limited then and he can make another attempt. ;)

Rene

davefilip, 2024-11-23:
Rene,

Interesting … so I remove CSocket.Connect() and instead just CSocket.Bind() to a port number that I specify?

So to be clear, it means that I can now CSocket.Receive() or CSocket.ReceiveFrom() from the SAME socket?  But then I still CSocket.SendTo() with port 69?  So I don’t need to open a second CSocket to receive, and this will give me a single bi-directional socket?  If so, then Bind() is really what makes it possible to receive from the socket?

Also, can I assume that if I try to bind to an ephemeral socket that is already in use by another protocol, that the Bind() function fail?  E.g., can I just start @ 1025 and start working my way up until a Bind() is successful?  I’m hoping Circle won’t let me bind to an ephemeral port that is already in use?

Nonetheless, if that is a problem, I can probably just pick a static port < 1024 that I know I’m not going to use later on (e.g., some Windows service that ‘ll never need).

Yes, I agree Stephan did an incredible job on stdlib/newlib.  No doubt about it!

We have to remember that we are using Circle because Circle != Unix.  So for size and efficiency, Circle will (hopefully!) always be different from Unix/Linux.

I consider the cooperative scheduling a plus, because it means that I, as the programmer, can decide when control is given back to the operating system.  So if I am smart about it, I can have something closer to a Real Time Operating System (RTOS) when and where I want it, but not when and where I don’t.  E.g., shut down the network during a critical section of code, and then re-start the network when I am done.

Nonetheless, will give your suggestion a try tomorrow morning and let you know how it works out.

BTW - Will be opening another separate issue — probably tomorrow — with an occasional / sporadic issue I’m seeing with CMQTTClient disconnects.  I expect that one to be more of a discussion in terms of the best way to handle.  More details later, as I want to properly document.

Always happy with your quick and thoughtful replies!

Cheers,

Dave,

rsta2, 2024-11-23:
Dave,

> Interesting … so I remove CSocket.Connect() and instead just CSocket.Bind()
> to a port number that I specify?

Right. Connect() established a direct port-to-port association with UDP, while 
Bind() only sets the own port for sending packets and the port, to which UDP 
packets must be directed to be received via this socket.

> So to be clear, it means that I can now CSocket.Receive() or
> CSocket.ReceiveFrom() from the SAME socket?  But then I still
> CSocket.SendTo() with port 69?  So I don’t need to open a second CSocket to [...]

That's what I would expect.
 
> Also, can I assume that if I try to bind to an ephemeral socket that is
> already in use by another protocol, that the Bind() function fail?  E.g.,
> can I just start @ 1025 and start working my way up [...]

No, the Bind() caller must know, which sockets he has already in use. There 
are no system services in Circle, which use port numbers above 50000. Sockets, 
which get an own port automatically, get a port number above 60000 (up to 
60999). So using 50001 to 59999 should be safe. But if you need multiple ports 
for Bind(), you must maintain the port numbers on your own.
 
> Nonetheless, if that is a problem, I can probably just pick a static port <
> 1024 that I know I’m not going to use later on (e.g., some Windows service
> that ‘ll never need).

BTW. This 1024 border does not exist in Circle. There are only some well-known 
ports, used in Circle, and the mentioned range for automatic port assignment. 
You can show the network status using:

	m_Net.GetTransportLayer()->ListConnections (&m_Screen);
 
> We have to remember that we are using Circle because Circle != Unix.  So for
> size and efficiency, Circle will (hopefully!) always be different from
> Unix/Linux.

That's for sure. ;)
 
Rene

davefilip, 2024-11-23:
Yup, that works!  Just changed the CSocket function from Connect() to Bind(), and now I can receive packets.

Thanks again for the suggestion.  Not sure that anything else is needed, as this work-around does the trick.  Cheers for suggesting a simple fix!

davefilip, 2024-11-26:
Rene,

Re-opening this because I made one of the most obvious - and maybe one of the more common - errors when I said that I thought I had this working: I tested with a small file that was less than the packet size (512 bytes).

I fixed the original problem of doing a Bind() on the socket to a known port, so that I can receive data back from the server.

However, my ACK packets are not making it back to the server, because I had been trying to send ACKs to port 69, whereas the TFTP server is both sending and receiving data on a high numbered port.

So, the question is: how do I determine the high numbered port that the TFTP server is using to return data?  I assume this is something that I can get from the CSocket, but if so, I'm not entirely clear which property of the CSocket I should be looking at?

So in summary, I now understand that the TFTP server receives requests on port 69, but transfers data on a high port, so I just need to know what that high port is so that I can send ACK packets back to that port.  Otherwise the TFTP server keeps sending the same first packet over and over, waiting for an ACK (until it times out).

Hopefully this is clear.  If not, let me know and I will provide a packet dump.  I'm assuming this is (hopefully) an easy answer, bit I just don't know enough about the CSocket implementation, and CSocket is different enough from Unix sockets that I'm not entirely sure what needs to be done.

Or, I am using SendTo() specifying the IP and port number.  Do I need to switch to using Send() instead to send ACK packets, and will that magically know the remote port?  Just a thought I had while writing this.

Thanks in advance for explaining how this is supposed to work.

Cheers,

Dave.

rsta2, 2024-11-27:
Dave, you can use just `CSocket::SendTo()` or `CSocket::ReceiveFrom()` to set or get the port number, an UDP packet is sent to or received from. The socket must be bound to a local port number using `CSocket::Bind()` before.

```
/// \brief Send a message to a specific remote host
/// \param pBuffer	Pointer to the message
/// \param nLength	Length of the message
/// \param nFlags	MSG_DONTWAIT (non-blocking operation) or 0 (blocking operation)
/// \param rForeignIP	IP address of host to be sent to (ignored on TCP socket)
/// \param nForeignPort	Number of port to be sent to (ignored on TCP socket)
/// \return Length of the sent message (< 0 on error)
int SendTo (const void *pBuffer, unsigned nLength, int nFlags,
	    const CIPAddress &rForeignIP, u16 nForeignPort);

/// \brief Receive a message from a remote host, return host/port of remote host
/// \param pBuffer Pointer to the message buffer
/// \param nLength Size of the message buffer in bytes\n
/// Should be at least FRAME_BUFFER_SIZE, otherwise data may get lost
/// \param nFlags MSG_DONTWAIT (non-blocking operation) or 0 (blocking operation)
/// \param pForeignIP	IP address of host which has sent the message will be returned here
/// \param pForeignPort	Number of port from which the message has been sent will be returned here
/// \return Length of received message (0 with MSG_DONTWAIT if no message available, < 0 on error)
int ReceiveFrom (void *pBuffer, unsigned nLength, int nFlags,
		 CIPAddress *pForeignIP, u16 *pForeignPort);
```

davefilip, 2024-11-27:
Rene,

OK, got it - thanks.  Just seems counterintuitive to Bind() something that is connecting to an external server, but I understand, this is how Circle fills in the data structures for the socket.

I think I got a bit confused because of the prior issue and not having ports populated for CSockets in certain circumstances (e.g., m_pRequestSocket in TFTPDaemon is always 0.0.0.0, or no rForeignIP in a CSocket.Connect() without a Bind()).

So yes, no need to explain further, I am conflating different things and confusing myself … sorry … :-(

But the good news is that I now know how to use Circle CSockets with UDP going forward … :-)

And I have used Circle CSockets with TCP connections already … so I should be good going forward with Circle CSockets… :-)

Apologies for perhaps not getting what should have been obvious, and appreciate your patience.

Cheers,

Dave.

rsta2, 2024-11-27:
No problem. The Circle socket API is not fully compatible with BSD. So the 
meaning of "bind" can be somewhat "extended" here.

davefilip, 2024-11-27:
Fully understood, and we can’t lose sight of the fact that:

    Circle != Unix

I have no problem with Circle being a unique environment with a unique API.

The GNU project … which basically grew into Linux … had a self-referential acronym:

    GNU = Gnu is Not Unix

Was always curious, how did you come up with the name Circle?

Nonetheless, I will close this issue.

Regards,

Dave.

closed 2024-11-27 by davefilip

rsta2, 2024-11-27:
> Was always curious, how did you come up with the name Circle?

Because the name "Pi" in "Raspberry Pi" is equivalent with the circle number 
(PI ~ 3,1415926). People with English as mother language may have different 
associations for the word "Circle", but this wasn't my intention.