rsta2/circle #513: MQTT Client Crash on Disconnect

davefilip, 2024-11-23:
Rene,

I've been having an occasional problem with MQTT Client disconnects, and crashing Circle.  It is NOT easy to replicate, and for example, while I now continually have running (usually at least) four (4) Circle-based operating systems that each send a MQTT message to the same Topic once every minute, it sometimes it takes days before I see the problem on most nodes, but other times it can take only 20 minutes or so after rebooting a particular node.  I think it largely depends on how happy my WiFi network is at the time, and how good the wireless connection is between my office (2nd floor) network is to my basement network.

[N.B., Although I have upgraded my WiFi network 3 times, I still need to periodically reboot my WiFi network (maybe once every couple of months) to fix odd sporadic WiFi connection and disconnection problems.]

As per the attached screen shot, I periodically -- again, sometimes not for days at a time on most nodes, although also sometimes within an hour of rebooting a node -- get an MQTT Client disconnect -- which first calls back my disconnect function, and then crashes the operating system with an assert in the following section of code:

```
void CMQTTClient::Receiver (void)
{
	while (1)
	{
		assert (m_pSocket != 0);
		TMQTTReceiveStatus Status = m_ReceivePacket.Receive (m_pSocket);
		switch (Status)

```

It always seems to be 'Invalid Identifier' as the reason for the disconnect given to my call-back function, followed by the assert at line 386 (the MQTT socket is null).

I have been using MQTT long before Circle, and the reason I like it is because the protocol is very, very resilient to slow and shaky network connections, as well as being (compared to other message bus protocols) small and efficient.

So what is my point?  Only that I would rather see that when this happens, the MQTT connection is cleanly and peacefully closed, so that the IsConnected() function returns false.  I currently -- anytime my code talks to MQTT that has previously been instantiated and connected -- check the status of the connection with isConnected(), and if it is not connected, then re-connect it.  Throwing an assert that brings the entire operating system down means there is no way of recovering.

I would argue that it should be assumed that a connected MQTT server can be disconnected and/or send partial or broken or corrupt packets, and that the client should close the MQTT connection, but NEVER crash the operating system (via assert or otherwise).

Although I could probably modify the CMQTTClient class myself with my own code fork (I have done that for a few classes already), that would mean I lose any changes / improvements / fixes that are made to CMQTTClient in the official Circle branch (at least not without me having to compare and apply any changes manually to my own code fork).

Because you provide a call-back when MQTT is disconnected, I don't think it is unreasonable to simply mark MQTT as disconnected and not bring down the operating system, since the programmer therefore has a hook to know when it happens.  However, if you want to keep this existing behavior, perhaps a system property that defines either crash or cleanly close and continue?

Hopefully this makes some sense?

Thoughts?  Comments?  Thanks for considering.

Cheers,

Dave.

[image]

rsta2, 2024-11-23:
Dave, 

this failed assertion is caused by a bug in the MQTT client. The client must terminate the receiver by returning from it, after the connection has been closed, because an invalid packet identifier has been received in a packet. This is currently not implemented. I will provide a fix soon.

Another question is, why an invalid packet identifier is received at all? The MQTT client holds packets of higher QoS in a re-transmission queue, to be able to send them again, when no ACK arrives. When an ACK arrives for a packet, which is not in this queue, this error occurs. You said, you have some experience with MQTT. Did you see such aborts before?

Thank you for reporting this! It's an obvious bug in Circle and should be fixed, of course.

Rene

rsta2, 2024-11-23:
The fix for the failed assertion is on the *develop* branch. It's simple, but should work.

davefilip, 2024-11-23:
Rene,

Thanks for agreeing to provide a fix!

I have mostly used MQTT with the Eclipse Paho Java library, so I do not see the details of the exact error condition (i.e., ACK for invalid packet), but it is not unusual to see a dropped connection over WiFi, and “bad” connections (forget the exact exception thrown, but gist is that the connection with the MQTT server is bad or corrupt).

In my past experience, and also now with Circle, I am using the Java Mosquito server on a Raspberry Pi (different one for each project).  So it is it also possible that there is a bug in that server?

Rarely drops, if ever, over Ethernet, so I am assuming that it is because of a bad WiFi connection.  I have around 80 WiFi devices connected to my network, and occasionally some of them will temporarily drop off and then later reappear.  I suspect I might be overfilling a routing or ARP table somewhere in the router.  It’s the type of thing where once I starting ping’ing the dropped node from different places, it will (most times) magically reappear.  I suspect that this might be why sometimes I lose a MQTT connection.  If it gets too bad, I’ll reboot the WiFi router, and then all of my node magically reappear!

BTW - I also have a VPN (OpenVPN) connection to a MQTT (also Mosquito) server running in the Cloud (CentOS Linux on DigitalOcean external hosting), which very occasionally drops, but overall tends to be more reliable than WiFi in my home.  Which is very sad.

I appreciate you providing a fix, and will test once available.

BTW - my code base is still on Step 47 with the patches you had previously provided prior to releasing 48 — I haven’t migrated fully to Step 48 yet — so can I safely just copy mqttclient.cpp from the Develop branch — once you have a fix — into my code base?  I am not using any RPi 5s, or currently need any of the other 47 -> 48 enhancements.

[Insert warning here about taking responsibility for mixing Circle Steps with Development patches]

Cheers,

Dave.

davefilip, 2024-11-23:
Thanks Rene, will download, test, and let you know shortly.

rsta2, 2024-11-24:
Dave, thanks for info about your MQTT experiences! I think we should watch 
this "invalid packet identifier" error further. It can be a problem of the 
server, but also of the client in Circle. It should be possible to re-connect 
MQTT, when a disconnect occurs because of some error for the moment.

You can copy the fixed mqttclient.cpp from the develop branch to a checked out 
Step47 from master branch. This should build and work.

Rene

davefilip, 2024-11-24:
Thanks, I did a diff, see that it is a very simple fix, but appreciate that it is making its way into to Circle code branch going forward.

There are a couple of classes that I did code fork because I wanted to make some changes (like /addon/Properties*), but then I own those changes going forward (with proper attribution to the original source, of course!).  But CMQTTClient is one that I want to keep generic and within your code branch, so I can benefit from any enhancements/fixes going forward.

But yes, within my OS I can certainly implement a watchdog to re-connect if MQTT gets disconnected (now that it does not crash!), but it might also be useful if that could occur within Circle. Just a thought.   I mentioned that I am using the Eclipse Paho MQTT library, and something in that library does (not exactly sure where) is automatically tries to re-connect if the connection gets dropped, but then gives up after a certain point, presumably assuming that there is a non-transient problem so it should stop trying.  But not a must have feature, since there are other work-arounds (e.g., a watchdog) that can accomplish the same thing.

Self healing is always a good idea, but there are usually multiple places in the stack where that can be accomplished.

Cheers,

Dave.

rsta2, 2024-11-24:
I think I will stay with the current (non-re-connecting) solution. It's also a 
matter of testing. I have never seen these disconnects, so how should I test 
them? Maybe if I would dive deeper into MQTT, I would have disconnects, but 
unfortunately I do not have the time for it. There are other features, which 
have more priority for me at the moment.

davefilip, 2024-11-24:
Great, makes perfect sense.  Always a balance between what the OS does and what the application does. 

I will run my Circle-based systems with your latest MQTT patch, and if I don’t see any problems after at least 24 hours (no crashes after 5,760 messages across 4 systems), I will close this issue.

Cheers,

Dave.

davefilip, 2024-11-24:
Running since last night on 4x nodes, had a few disconnects, and OS no longer crashing with assert.  So now fixed!

closed 2024-11-24 by davefilip