rsta2/circle #517: dual display

nickverlinden, 2024-12-02:
Please forgive my ignorance, maybe i'm overlooking it in the documentation but couldn't really find it. Can we use both the DSI touchscreen and HDMI output at the same time, either mirrored or independently? I would like to use a dsi touchscreen to display a gui, and the hdmi to display a picture. Even mirroring the DSI touchscreen to the HDMI output would also be fine.

nickverlinden, 2024-12-02:
If i read this correctly, it seems that on the pi4 i can create a second instance of C2DGraphics, where i give the nDisplay parameter in the constructor to indicate the display, so if I create two C2DGrapchis instances, i can draw stuff to the Buffer of each one. Will this fail if there is no HDMI connected for instance? I guess if this is the case, HDMI is not hot pluggable unless you periodically try to create a new instance of C2DGraphics. Is that right?

```
::C2DGraphics(unsigned nWidth, unsigned nHeight, boolean bVSync = TRUE, unsigned nDisplay = 0)
Creates on instance of this class. nWidth is the screen width in pixels (0 to detect). nHeight is the screen height in pixels (0 to detect). Set bVSync to TRUE to enable VSync and HW double buffering. nDisplay is the zero-based display number (for Raspberry Pi 4).
```

rsta2, 2024-12-02:
On the RPi 4 you can connect two displays (or maybe even three, 2 HDMI, 1 DSI) 
at the same time. The config.txt parameter max_framebuffers must be set 
accordingly. But HDMI must be connected at power on, otherwise Initialize() 
will fail.

closed 2024-12-03 by nickverlinden