rsta2/circle #523: Confusion Around Clogger::Read()

davefilip, 2024-12-16:
Rene,

Trying to read the Circle log data, running into problems, and am not sure what I doing wrong.

This is the code that I am trying to use:

```
	CLogger *logger = CLogger::Get();

	char buffer[1024];
	size_t count;

	while (true) {
		count = logger->Read(buffer, sizeof(buffer), false);
		buffer[count] = '\0';
		m_Shell->Output(buffer);
		if (count < sizeof(buffer)) break;
	}
```

As I understand from the online doc, I believe this should allow me to read the logger circular buffer, but not remove any characters from it.  However, this runs indefinitely without ever breaking out of the loop, circulating through the same log entries over and over and over ... whereas I would expect count to be less than the buffer size, and then zero, when all of the data had been read once.

If I change the false to true -- meaning remove any characters read from the circular buffer -- it does work once as expected, providing log messages from the the initial boot, until the time that this code is run.  Good, but I didn't want to remove those characters, I wanted to leave them in the logger buffer.  When I run the same code a second time -- still with true -- it again gets stuck in an infinite loop, repeating the same entires over and over and over ...

So what I would like to do is similar to the dmesg command in Linux, whereby I can run a command to display all of the log messages since boot.  But if I run the same command again, possibly from another session (e.g., once from the console and once from a telnet session), I would get the same results.  Until the 16K buffer has been exceeded, and then just the last 16K messages.

Ideally, what I really want to do is use Clogger::ReadEvent(), e.g.:

```
CLogger *logger = CLogger::Get();

	TLogSeverity Severity;
	char Source[LOG_MAX_SOURCE+1];
	char Message[LOG_MAX_MESSAGE+1];
	time_t Time;
	unsigned Hundredths;
	int TZ;
	
	while (logger->ReadEvent(&Severity, Source, Message, &Time, &Hundredths, &TZ)) {
		m_Shell->OutputNL(Message);
	}
```

That works, but removes all of the messages from the buffer.  So if I run from two different sessions (e.g., once on the console and once from a telnet session), the session that runs first will remove the messages, so the second session won't get anything.

ReadEvent() would work for me, if it was not destructive and remove messages.

What is the best way to accomplish what I am trying to do?  Again, my goal is something similar to the Linux dmesg command, whereby I can run it multiple times without removing any messages.  I understand that Circle will remove old log messages after the 16K buffer fills up, which is fine, as long as I can always get the last 16K of log messages.

Any advice would be appreciated, as I don't think I am understanding how this is supposed to work.

Cheers,

Dave.

rsta2, 2024-12-17:
Dave, you have to read the logger buffer with one call with a buffer, which is big enough to receive the whole saved log. There is no other way, if you call `CLogger::Read()` with the last parameter set to `FALSE`. The buffer must have at least the size `LOGGER_BUFSIZE` (defined in `<circle/logger.h>`).

Rene

davefilip, 2024-12-17:
Rene,

Thanks, not how I initially interpreted the on-line doc.  So it is basically it is just dumping the entire buffer in one call (assuming the buffer is large enough)?

So if the buffer I provide is too small, does it provide the oldest entires until the buffer is full, or newest entries without overflowing the buffer?

Ideally, I’d rather use ReadEvent() — since then each log entry is already parsed into its components — but it is always destructive.  Not a priority, but in the future, would you consider a flag for ReadEvent() so that it was non-destructive (i.e. , did not remove the entires it had already read?).

Either that, or what I have seen in a lot of APIs is a ’since’, or ‘all entires since this date’, and I cam provide the date (or hundredths) of the last entry I previously retrieved?  It might be useful to see all one entries in the last hour?

Again, not a priority, but maybe something to consider?

Cheers,

Dave.

davefilip, 2024-12-17:
No problem, now that I understand how this is supposed to work, I can make this work for my use case.

Having ReadEvent() be non-destructive would be a nice to have, but a low priority compared to other things that are much higher priorities.

[N.B., you have mentioned IVP6 in one of our conversations, but I would ask how many requests have you received for that?  One of the first things I "typically" do on a Raspberry Pi running full Linux is disable IPV6, because it tends to get in the way, and unless you are connecting Raspberry Pis running Circle on the public Internet and having people connect directly to them (and not behind a router or firewall), is there much of a need?  Of course, I might be completely wrong and you may have received a lot of requests, I'm just sharing my opinion, so feel free to disagree.]

I am closing this issue regarding reading the system logs.

closed 2024-12-17 by davefilip

rsta2, 2024-12-17:
Dave,

yes, if you set `bClear` to `FALSE`, you have to dump the entire buffer at once to see all messages. `bClear` has been added as an additional parameter later. Initially `Read()` always removed the read characters.

If the provided buffer is too small for all messages, the newest messages will not be copied.

`ReadEvent()` has been added for the Syslog support (see *sample/33-syslog*). The intended model was, that the consumer of the log messages calls `ReadEvent()` as often as a message is returned, when it is started, and then registers a notification handler using `RegisterEventNotificationHandler()` to be informed, when another message arrives. Then it can call `ReadEvent()` again to get the new message.

> No problem, now that I understand how this is supposed to work, I can make this work for my use case.

Ah, great!

I did not receive a request for IPv6 so far. Good to know your opinion!

Rene