rsta2/circle #525: How to Terminate One Task from Another Task

davefilip, 2024-12-17:
I think this might be a moot question, and this issue closed quickly, as the online  documentation for CTask::Terminate() says:

_**Terminates the execution of the task. This method can only be called by the task itself.**_

However, what is the suggested method, or is there any suggested method, for one CTask being able to terminate another CTask?

Use case is a user's command shell that can request (now in colorOS!) a "background" task (what I am calling a Phantom Task), which runs in its own separate shell, so the user can type and run other commands while the phantom task is running in the background.  A phantom background task can be a command file which can loop.

Where am I going with all of this?  Well, for example, I have a command that establishes a connection to the MQTT server if it is dropped.  So, as just one example, a command file that runs in the background, and runs the command to re-connect to MQTT if it is disconnected, then sleeps for a few minutes, and then repeats.  Another is a command file that reads some sensor values, sends an outgoing HTTP message to an external web service, sleeps for a minute, and then repeats, so the sensor value is read and sent every minute to the web service.

So if I have a few of these background tasks running in continuous loops, how can I terminate one of them without rebooting?

If I get a pointer to the CTask, I can call Terminate() on it, but it always crashes the OS, usually with an assertion.  So before I start digging into it deeper, is there a way that I should be able to have one CTask terminate another CTask without the second CTask calling Terminate() on itself (or returning from Run())?

I ask because it might be that I am simply not cleaning up the CTasks properly in their deconstructors (~{classname}), but given the warning in the online documentation, I'm wondering if it is ever something that I should be able to get to work?  Or is it simply not possible with how Circle is currently written?

I've been doing this type of work long enough to know that the alternative is a mailbox method whereby one CTask can send a message to another CTask, and when the second CTask sees a terminate message, it exits voluntarily, but that involves building a mailbox mechanism, which I don't think Circle currently has?

To be clear, not asking you to build or change anything in Circle, just wondering if there is a way to accomplish what I am asking, before I start building a message passing / mailbox mechanism.

Of course, the equivalent of what I am asking in Linux -- and we must never forget that Circle != Linux -- is a 'kill -9' on a process.  And no need to remind me that a Circle CTask is very different from a Linux process, and more like a thread (although in the past, I have terminated threads all the time in server programs!).

If what I am asking is really not possible, please feel free to reply with one word: "Impossible"

However, in the past, when I have asked certain questions and provided the use case, you have sometimes surprised me by offering an alternative way around the problem (e.g., doing a Bind() on an outgoing UDP CSocket to receive data from it).

Cheers,

Dave.

smuehlst, 2024-12-17:
>  if there is a way to accomplish what I am asking, before I start building a message passing / mailbox mechanism.

As with Circle it is the same address space across all tasks, shouldn't it be possible to set a global variable in a controller task, and another task queries this variable in its continuous loop and then exits voluntarily on request?

davefilip, 2024-12-17:
General answer is "yes", that is correct, but since there can be any number of ("phantom/background") tasks started across different user sessions, it is not as simple as checking one global variable, and this is kinda what I was getting at with a "mailbox" for sending messages between tasks.  Need to have one location or "mail slot" created for each ("phantom/background") task started, and each shell (session) needs to know which slot to write the "terminate" message to, depending on which background task needs to be terminated.

Of course, this also means that the task being terminated has to be periodically looking for a terminate message -- and which "mail slot" to look in for a message -- so it is not preemptive.  Which is why a CTask::Terminate() would be better, if it is at all possible, since presumably it would be preemptive.

I suspect Rene is going to tell me that there is nothing in Circle that can do a preemptive termination of a task (while I can preemptively suspend or resume a task), and that I will need to build the "mail slot" approach, which is not rocket science, but means building a table for keeping track of which "mail slots" are associated with which tasks, and maintaining that table as background tasks are started and stopped.

rsta2, 2024-12-17:
Dave, as you noted, a task can only terminate itself. Stephan already 
suggested a possible solution in Circle. Circle expects tasks to be gentle to 
the system.

A question: If there would be a "kill -9" in Circle, how would you call it for 
a task, without having a reference on it? So you need the reference anyway. 
And when you have the reference, you can call just a simple method, which sets 
a flag, which terminates the task's main loop.

davefilip, 2024-12-17:
Rene,

This goes back to an earlier discussion we’ve had, about being able to call a method in a running task, which you said was not possible.

I can get the CTask, as you know, by:

CScheduler *Scheduler = CScheduler::Get();
CTask *Task = Scheduler->GetTask(name);

And I can Task->Suspend() or Task->Resume(), but I cannot Task->Terminate() successfully (I can make the call, but it never ends well, perhaps, as you suggest, because it is not ‘gentle’).

We had a prior  discussion about whether it was possible to use CTask to get access to the underlying class (with the running Run() method), and call one of that classes methods, but I believe you said that was not possible?

So this morning I started writing the “mailbox slots” that I referenced, both in my original question, and in my response to Stephan, but before I got too far, I thought I’d reach out again to confirm whether there was an easier method.

So now that I know there is not, I will use that approach.

It seems odd that there is no way to terminate a task from outside of a task.  Since Circle knows how to call the Run() method of the class, it might seem reasonable for every  class that inherits from CTask to also have a Stop() (or something like that) method that Circle could call to ask the task to terminate itself.  It would then still be up to the task itself to gently clean itself up before it terminates.

Nonetheless, since nothing like that exists, I will close this issue.  But as always, thanks for the response.

Regards,

Dave.

closed 2024-12-17 by davefilip

rsta2, 2024-12-18:
> We had a prior  discussion about whether it was possible to use CTask to get access to the underlying class (with the running Run() method), and call one of that classes methods, but I believe you said that was not possible?

When you have a pointer to the `CTask` object, you can cast is to the actual type of your custom task and if you added a method `Stop()` to it, for example, you can call it to set a flag. If this flag is checked in your task's main loop, you can "gently" stop your task by returning from `Run()`. If I wrote, that this is not possible, I was wrong.

davefilip, 2024-12-18:
Rene,

OK, I might have misunderstood what you had previously said.  But I did try this previously, and could not get it to work, which is when I reached out to you and I thought you said that the CTask object was very light and only used for scheduling, and could not call custom class functions directly.

Let me try again, and if I can’t get it to compile / work, I might send you a small code fragment of what I am trying to do.  Hopefully that will clarify any misunderstanding between us in this regards to how this might work.

Cheers,

Dave.

davefilip, 2024-12-18:
Rene,

OK, I can now get this to work.  Back around July or so I couldn’t get it to compile, but I think it was because I was still learning C++ (and typecasting syntax is a bit different between Java and C++, because Java syntax does not explicitly reference pointers).  Java and C++ are almost exactly alike except where they are entirely different … ;-)

[N.B., statics within classes vs. static classes also confused the hell out of me early on, because the terminology is similar to Java, but it works differently.]

The earlier misunderstanding is probably on my side, and I would have saved A LOT of time and effort if I could have gotten this to work earlier, and avoided passing around pointers to TASK_USER_DATA_USER vectors for exchanging data between tasks.  I probably didn’t explain to you what I was trying to do clearly enough, and/or not used the proper C++ terminology (which I was still learning at the time).

Nonetheless, now that I can call public functions on CTask derived classes, it will make my life a lot easier!

Thanks always for your patience and guiding me through all of this as I learn.

Cheers,

Dave.

rsta2, 2024-12-18:
Dave,

I understand, that it is not easy and takes time to learn C++ on the fly, after starting to use Circle. But you are making progress, and that's important. Thank you for choosing Circle for your project!

Rene

davefilip, 2024-12-18:
Nobody can argue that Circle is not an incredible and awesome OS base for the Raspberry Pi computers, or that your ongoing support and dedication is anything short of incredible and awesome!

It was very much worth the learning curve of C++, and I avoided Circle for way to long because I was afraid of learning a new language (I am old, retired, and at least in English, there is a saying: "You can’t teach an old dog new tricks”).  But I did do a bunch of C programming a few decades ago, mostly system level stuff, not writing an operating system, but system admin and management stuff on DEC VAXes that would now be called “dev ops"

Nonetheless, I feel I have come a long way learning C++ and Circle at the same time, and could not have done that without your support.  And I mean that sincerely.  Most of the other “bare metal” frameworks in the public domain for the Raspberry Pi are not actively maintained, have little or no support, and have lots of incompatibilities between different versions of the Raspberry Pi.

[And to think that I started this all with Fortran IV on punched cards nearly 45 years ago! ;-) ]

rsta2, 2024-12-18:
Thank you, Dave!