rsta2/circle #529: ST7789 Questions

davefilip, 2024-12-31:
Rene,

Not really an issue yet, as I'm still trying to figure out how to use the ST7789 driver in Circle.

My goal -- if possible -- is to keep the existing HDMI display, but be able to separately write to the LCD screen.  Taking a closer look at your sample code in /addon/display/sample/st7789chr.cpp, it looks like you are making the LCD the default display (i.e., setting m_Display to use the LCD).

Again, I would like to use both simultaneously.

So can I use the HDMI m_Display as I currently do ... and then doing something like this:

```
CST7789Display		m_LCDDisplay;  // NOT m_Display, which is still the HDMI
CST7789Device		*m_pLCD;

m_LCDDisplay.Initialize();
m_Display.SetRotation(ROT);

m_pLCD = new LCDDisplay(&m_SPIMaster, DC_PIN, RESET_PIN, BACKLIGHT_PIN, WIDTH, HEIGHT,
			    SPI_CPOL, SPI_CPHA, SPI_CLOCK_SPEED, SPI_CHIP_SELECT);

m_pLCD = new CST7789Device(&m_SPIMaster, &m_LCDDisplay, COLS, ROWS, WIDECHARS, TALLCHARS);
m_pLCD->Initialize();

m_pLCD->Write("Hello, World!", strlen("Hello, World!"));
```

to enable the ST7789 driver and write 'Hello, World!' to the LCD, while also keeping the existing HDMI display?

And if so, do I need the 'new LCDDisplay()', or can I omit that entirely if I only want to use m_pLCD->Write() to write to the LCD?

Or is the 'new LCDDisplay() necessary, because I need something to pass as the second argument when instantiating CST7789Device?  If not, then what would I pass as the second argument?

Please let me know if this is possible, that is if Circle will permit the LCD display to operate completely separate from the regular HDMI display, and if so, what is the minimum amount of work (code) do make that happen?

Cheers,

Dave.

rsta2, 2024-12-31:
Dave, I think we are talking about Circle Step48, because the version on the 
develop branch provides new features in this field.

What is important in addon/display/sample/st7789char is, that the class 
CST7789Display is the ST7789 display driver, which provides only some basic 
methods to set a pixel or to draw some static text. The class CST7789Device 
uses CST7789Display to implement a terminal (aka character) mode upon it, but 
does not include a member of the class itself. That's why CST7789Display must 
be instantiated first and initialized and after that CST7789Display must be 
instantiated using the new-operator and initialized afterwards.

This sample already provides parallel access to the standard HDMI display (as 
m_Screen, for logger output) and to a ST7789 display in terminal mode. 
Unfortunately I cannot make the sample much smaller. It should provide, what 
you want.

Rene

davefilip, 2025-01-03:
Thanks - I have this working!

closed 2025-01-03 by davefilip