rsta2/circle #530: assert fails fpr mac != 0 for WLAN when Initialize (boolean bWaitForActivate) is false

omersiar, 2025-01-02:
[image]

I see from wlan Broadcom device is ready and MAC is available but I do not know at which point it gets copied for L2 layer

I was trying to initialize the network stack first before trying to connect WLAN or getting an IP from DHCP
This is working when bWaitForActivate is true

Code looks like this.
https://github.com/probonopd/MiniDexed/blob/network/src/minidexed.cpp#L2284-L2347

rsta2, 2025-01-03:
`CWPASupplicant` must be initialized after `CNetSubSystem` in any case. Otherwise it will not work, because `CWPASupplicant` accesses `CNetSubSystem`. Have a look at [this sample](https://github.com/rsta2/circle/blob/c243194670720c8a13e6a553654b6d26ec7845a0/addon/wlan/sample/hello_wlan/kernel.cpp#L105-L127) for the expected initialization order.

omersiar, 2025-01-03:
Sorry for confusion.  That helped and solved the issue.

closed 2025-01-03 by omersiar