rsta2/circle #532: xhciendpoint error when trying to read/write files on usb drive

nickverlinden, 2025-01-07:
Hi Rene,

I ran into some trouble; i wanted to add some code to save a settings file to a usb stick. I based my code on the umsd example and do a directory listing, then a write file, after that a read file.

First i was still on Circle 47, the directory listing always went fine every time, but the write/read file gave me some specific errors.
I tried another usb stick but the same result. Once in a while, writing and reading the file would work, but very often it would not and sometimes even corrupted the filesystem on the stick.

I decided to give Circle 48 a go, and had to go to the develop branch because cflashy would not build (macos) and saw a PR that fixes this. Now i get a different error. It just crashes at the last line, and the last log line is incomplete.

I removed all other usb devices, so only the stick is present in the log output below. Usually i would also connect the esp32 device that acts as an audio interface (the one i was talking about in some of my previous issues).

Do you have any idea what could be going on?

CIRCLE 47 output:
```sh
00:00:00.95 pcie: Link up, 5.0 Gbps x1
00:00:01.46 usbdev1: Device ven2109-3431, dev9-0-1 found (HS)
00:00:01.46 usbdev1: Product: USB2.0 Hub
00:00:01.51 usbdev1: Interface int9-0-0 found
00:00:01.51 usbdev1: Using device/interface int9-0-0
00:00:02.41 usbdev1-3: Device ven18a5-245 found (HS)
00:00:02.41 usbdev1-3: Product: Verbatim STORE N GO
00:00:02.46 usbdev1-3: Interface int8-6-50 found
00:00:02.46 usbdev1-3: Using device/interface int8-6-50
00:00:03.06 umsd: Capacity is 29604 MByte
00:00:03.06 usbhub: Port 3: Device configured
00:00:03.06 xhciroot: Port 1: Device configured
00:00:03.07 kernel: Compile time: Jan  7 2025 08:11:05
00:00:03.07 mcore: CPU core 1 started
00:00:03.07 uicore: starting event loop on core 1
00:00:03.07 mcore: CPU core 2 started
00:00:03.07 uicore: starting event loop on core 2
00:00:03.07 mcore: CPU core 3 started
00:00:03.07 uicore: starting event loop on core 3
00:00:03.07 uicore: starting event loop on core 0
00:00:03.07 kernel: usb devices updated, here's a list:
c touch1       c sndpwm       b umsd1        b umsd1-1     
c uhub1        c ttyS1        c tty1         
00:00:03.08 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] listing files:

00:00:03.09 kernel: + /.Spotlight-V100
  /test.txt (0 bytes)
  /armstub8-rpi4.bin (336 bytes)
  /._config64.txt (4096 bytes)
  /bcm2710-rpi-zero-2-w.dtb (32570 bytes)
  /bcm2711-rpi-4-b.dtb (54809 bytes)
  /bcm2711-rpi-400.dtb (54813 bytes)
  /bcm2711-rpi-cm4.dtb (55449 bytes)
  /bcm2712-rpi-5-b.dtb (75445 bytes)
  /bootcode.bin (52476 bytes)
  /cmdline.txt (54 bytes)
  /._cmdline.txt (4096 bytes)
  /config.txt (606 bytes)
  /._config.txt (4096 bytes)
  /config32.txt (460 bytes)
  /config64.txt (607 bytes)
  /COPYING.linux (18693 bytes)
  /fixup_cd.dat (3204 bytes)
  /fixup.dat (7303 bytes)
  /fixup4.dat (5434 bytes)
  /fixup4cd.dat (3204 bytes)
  /kernel_2712.img (2099642 bytes)
  /kernel8-rpi4.img (2099914 bytes)
  /LICENCE.broadcom (1594 bytes)
  /Makefile (2616 bytes)
  /README (3576 bytes)
  /start_cd.elf (808892 bytes)
  /start.elf (2980544 bytes)
  /start4.elf (2256224 bytes)
  /start4cd.elf (808892 bytes)

00:00:03.09 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] debug - cpath: USB:/test.txt

00:00:04.15 xhciep: Transfer timed out
00:00:04.15 umsd: Data transfer failed
00:00:04.15 umsd: TryWrite failed
00:00:05.39 xhciep: Transfer timed out
00:00:05.39 umsd: CSW transfer failed
00:00:06.39 xhciep: Transfer timed out
00:00:06.39 umsd: CSW transfer failed twice
00:00:06.39 umsd: TryWrite failed
00:00:07.49 xhciep: Transfer timed out
00:00:07.49 umsd: CSW transfer failed
00:00:08.49 xhciep: Transfer timed out
00:00:08.49 umsd: CSW transfer failed twice
00:00:08.49 umsd: TryWrite failed
00:00:09.59 xhciep: Transfer timed out
00:00:09.59 umsd: CSW transfer failed
00:00:10.59 xhciep: Transfer timed out
00:00:10.59 umsd: CSW transfer failed twice
00:00:10.59 umsd: TryWrite failed
00:00:11.69 xhciep: Transfer timed out
00:00:11.69 umsd: CSW transfer failed
00:00:12.69 xhciep: Transfer timed out
00:00:12.69 umsd: CSW transfer failed twice
00:00:12.69 umsd: TryWrite failed
00:00:13.79 xhciep: Transfer timed out
00:00:13.79 umsd: CSW transfer failed
00:00:14.79 xhciep: Transfer timed out
00:00:14.79 umsd: CSW transfer failed twice
00:00:14.79 umsd: TryWrite failed
00:00:15.89 xhciep: Transfer timed out
00:00:15.89 umsd: CSW transfer failed
00:00:16.89 xhciep: Transfer timed out
00:00:16.89 umsd: CSW transfer failed twice
00:00:16.89 umsd: TryWrite failed
00:00:17.99 xhciep: Transfer timed out
00:00:17.99 umsd: CSW transfer failed
00:00:18.99 xhciep: Transfer timed out
00:00:18.99 umsd: CSW transfer failed twice
00:00:18.99 umsd: TryWrite failed
00:00:19.09 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] debug - cpath: USB:/test.txt

00:00:20.09 xhciep: Transfer timed out
00:00:20.09 umsd: CSW transfer failed
00:00:21.09 xhciep: Transfer timed out
00:00:21.09 umsd: CSW transfer failed twice
00:00:21.09 umsd: TryWrite failed
00:00:22.19 xhciep: Transfer timed out
00:00:22.19 umsd: CSW transfer failed
00:00:23.19 xhciep: Transfer timed out
00:00:23.19 umsd: CSW transfer failed twice
00:00:23.19 umsd: TryWrite failed
00:00:24.29 xhciep: Transfer timed out
00:00:24.29 umsd: CSW transfer failed
00:00:25.29 xhciep: Transfer timed out
00:00:25.29 umsd: CSW transfer failed twice
00:00:25.29 umsd: TryWrite failed
00:00:26.39 xhciep: Transfer timed out
00:00:26.39 umsd: CSW transfer failed
00:00:27.39 xhciep: Transfer timed out
00:00:27.39 umsd: CSW transfer failed twice
00:00:27.39 umsd: TryWrite failed
00:00:28.49 xhciep: Transfer timed out
00:00:28.49 umsd: CSW transfer failed
00:00:29.49 xhciep: Transfer timed out
00:00:29.49 umsd: CSW transfer failed twice
00:00:29.49 umsd: TryWrite failed
00:00:30.59 xhciep: Transfer timed out
00:00:30.59 umsd: CSW transfer failed
00:00:31.59 xhciep: Transfer timed out
00:00:31.59 umsd: CSW transfer failed twice
00:00:31.59 umsd: TryWrite failed
00:00:32.69 xhciep: Transfer timed out
00:00:32.69 umsd: CSW transfer failed
00:00:33.69 xhciep: Transfer timed out
00:00:33.69 umsd: CSW transfer failed twice
00:00:33.69 umsd: TryWrite failed
00:00:34.79 xhciep: Transfer timed out
00:00:34.79 umsd: CSW transfer failed
00:00:35.79 xhciep: Transfer timed out
00:00:35.79 umsd: CSW transfer failed twice
00:00:35.79 umsd: TryWrite failed
00:00:35.89 kernel: [ CIRCLE       ][ KERNEL       ][ ERROR ] file read test fatfs err: 1
```

CIRCLE 48 (develop) output:
```sh
00:00:00.93 pcie: Link up, 5.0 Gbps x1
00:00:01.44 usbdev1: Device ven2109-3431, dev9-0-1 found (HS)
00:00:01.44 usbdev1: Product: USB2.0 Hub
00:00:01.49 usbdev1: Interface int9-0-0 found
00:00:01.49 usbdev1: Using device/interface int9-0-0
00:00:02.39 usbdev1-3: Device ven18a5-245 found (HS)
00:00:02.39 usbdev1-3: Product: Verbatim STORE N GO
00:00:02.44 usbdev1-3: Interface int8-6-50 found
00:00:02.44 usbdev1-3: Using device/interface int8-6-50
00:00:03.22 umsd: Capacity is 29604 MByte
00:00:03.23 usbhub: Port 3: Device configured
00:00:03.23 xhciroot: Port 1: Device configured
00:00:03.23 kernel: Compile time: Jan  7 2025 21:36:17
00:00:03.23 mcore: CPU core 1 started
00:00:03.23 uicore: starting event loop on core 1
00:00:03.23 mcore: CPU core 2 started
00:00:03.23 uicore: starting event loop on core 2
00:00:03.24 mcore: CPU core 3 started
00:00:03.24 uicore: starting event loop on core 3
00:00:03.24 uicore: starting event loop on core 0
00:00:03.24 kernel: usb devices updated, here's a list:
c touch1       c sndpwm       b umsd1        b umsd1-1     
c uhub1        c ttyS1        c tty1         
00:00:03.24 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] listing files:

00:00:03.26 kernel: + /.Spotlight-V100
  /test.txt (0 bytes)
  /armstub8-rpi4.bin (336 bytes)
  /._config64.txt (4096 bytes)
  /bcm2710-rpi-zero-2-w.dtb (32570 bytes)
  /bcm2711-rpi-4-b.dtb (54809 bytes)
  /bcm2711-rpi-400.dtb (54813 bytes)
  /bcm2711-rpi-cm4.dtb (55449 bytes)
  /bcm2712-rpi-5-b.dtb (75445 bytes)
  /bootcode.bin (52476 bytes)
  /cmdline.txt (54 bytes)
  /._cmdline.txt (4096 bytes)
  /config.txt (606 bytes)
  /._config.txt (4096 bytes)
  /config32.txt (460 bytes)
  /config64.txt (607 bytes)
  /COPYING.linux (18693 bytes)
  /fixup_cd.dat (3204 bytes)
  /fixup.dat (7303 bytes)
  /fixup4.dat (5434 bytes)
  /fixup4cd.dat (3204 bytes)
  /kernel_2712.img (2099642 bytes)
  /kernel8-rpi4.img (2099914 bytes)
  /LICENCE.broadcom (1594 bytes)
  /Makefile (2616 bytes)
  /README (3576 bytes)
  /start_cd.elf (808892 bytes)
  /start.elf (2980544 bytes)
  /start4.elf (2256224 bytes)
  /start4cd.elf (808892 bytes)

00:00:03.26 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] debug - cpath: USB:/test.txt

00:00:04.38 xhciendpoint.cpp(282): stack[1
```

nickverlinden, 2025-01-07:
I just looked at the assert that fails, and increased the MAX_KERNEL_SIZE, recompiled everything, but the result remains the same.

I also tried commenting the assert:
```sh
                 assert (pBuffer != 0);
		assert (nBufLen > 0);
		//assert ((uintptr) pBuffer > MEM_KERNEL_END);
		CleanAndInvalidateDataCacheRange ((uintptr) pBuffer, nBufLen);

		if (!EnqueueTRB (  XHCI_TRB_TYPE_NORMAL << XHCI_TRB_CONTROL_TRB_TYPE__SHIFT
				 | XHCI_TRANSFER_TRB_CONTROL_IOC,
		     		 nBufLen | 0 << XHCI_TRANSFER_TRB_STATUS_TD_SIZE__SHIFT,
				 XHCI_TO_DMA_LO (pBuffer),
				 XHCI_TO_DMA_HI (pBuffer)))
		{
```
and then it actually works, the write creates the text file, writes the contents and the read functions reads the contents back out. but ... when starting the write operation it takes about 5 seconds and then it succeeds. the debug log with the cpath variable below is printed right before the f_write function, and right before the f_read function call, as you can see there is about 5 to 6 seconds of a delay between that (and because this code runs on the same core as the gui, the gui is also frozen during these 5 seconds). After succesfully completing that, calling `m_pUSBHCI->UpdatePlugAndPlay ();` returns an updated device list as well:

```sh
00:00:45.41 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] debug - cpath: USB:/test.txt

00:00:51.29 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] debug - cpath: USB:/test.txt

00:00:51.30 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] Text content: HELLO FROM NOS!!

00:00:51.30 kernel: tryin to get usb sound device
00:00:51.30 sndusb: USB audio streaming device not found (RX)
00:00:51.30 kernel: pSoundIn: 1f17080
00:00:51.30 sndusb: Probing sound controller failed
00:00:51.30 kernel: Cannot start input sound device
00:00:51.30 kernel: usb devices updated, here's a list:
c sndusb       b umsd1        b umsd1-1      c sndusb      
c sndusb       c touch1       c sndpwm       c uhub1       
c ttyS1        c tty1         
00:00:51.30 kernel: [ CIRCLE       ][ KERNEL       ][ DEBUG ] listing files:

00:00:51.31 kernel: + /.Spotlight-V100
  /test.txt (255 bytes)
  /armstub8-rpi4.bin (336 bytes)
...
```

rsta2, 2025-01-08:
This assertion should ensure, that the data buffer is on the stack or on the 
heap and not in a static variable, which can cause problems with the data 
cache handling. Do you use a string constant directly as a parameter to 
Write() or a static variable as buffer?

Perhaps your USB drive needs relatively long to setup. There is a timeout of
1 second in Circle 48, which has been increased to 3 seconds on the develop 
branch. Perhaps I have to increase it further. These timeouts must not occur. 
Otherwise data corruption can happen.

nickverlinden, 2025-01-08:
You are right, the write test i was doing contained a string constant. In the meantime i did some further testing, and it seems the long delay only occurs the first time you try to write on the stick, such as create a folder or write to a file. Most of my code is c code, and the settings that i write to file are structs that are created with the "extern" keyword. These are not static, but i have to say that i'm referring to 'static' in C, if i'm not mistaken static in C++ is something else. Do you think there are problems trying to write these structs? (which are not constant nor static). my settings code seems to work properly loading and saving parameter values, although a bit on the slow side. I will try to uncomment the assert again tonight now that it does not try to write the constant anymore.

rsta2, 2025-01-08:
With static data I don't mean the C++ keyword "static" but all data, which is 
located in static data segments (e.g. .data, .bss). Unfortunately I cannot 
safely remove this assertion, because I cannot guarantee, that everything will 
work then in each configuration. The alternative would be to remove it, but 
users may encounter spurious errors then, without notice. I had those errors, 
when developing the xHCI driver.

nickverlinden, 2025-01-08:
Well, good news, i reenabled the assert and all is working fine with my settings code. The problem was effectively my test code that tried to write a string to a file and read it out again. It was just for testing so i just used a string directly in the function call (making it effectively a const).  Thanks for helping me out once again!

closed 2025-01-08 by nickverlinden