rsta2/circle #535: Support for Speedlink USB Joystick SL-650212

Chris-Hal, 2025-01-17:
Hi Rene,
I don't want to overstretch your kindness :-)  this has no high importance for me but maybe for a couple of user because I just found this problem #476  in the list of issues and I have also an Speedlink USBJoystick  for PC and tested this with BMC64 and came to the same result. The new Speedlink type SL-650212 which is a recent product will not be recognized in BMC64 too. To be honest, I have also an older Speedlink SL-6603 which is working fine with BMC64

Due to the Maxi keyboard issue, I had some kernels fom you for testing purposes and found the same initial problem with the SL-650212 joystick which we had with the Maxi keyboard. I mean the configuration descriptor is not according to the USB spec. There is a descriptor with zero length in it, so the parser never came to an end. 

So the initial problem why the Speedlink joystick why the Parser hangs is already known and the status after using your Kernel with this fix is like in the screenshot. But still the stick is not detected in BMC64 , I proofed that with Randy's recent version, but if you like and feel that we should proceed, I am ready to go :-) What do you think? 

Maybe If you can provide me the same tool which you have made for  the Pi3  here
[Link]https://github.com/rsta2/circle/issues/476#issuecomment-2366788664
for the Pi2, I can immediately gather some details about the joystick? Additionally I can connect the joystick to the USB Analyzer software and make some screenshots as well? Even a comparison with the working Speedlink can be done.

Looking for your feedback

Best regards
Christian

[image]

rsta2, 2025-01-17:
Christian, we were not able to get this joystick working in issue #476, and I 
think the situation has not changed: The joystick works with Circle Step48, 
but not with Step40, which is used in BMC64. There are probably a number of 
commits between these releases, which affect the joystick behavior, but we were 
not able to find them. It's not only one commit. The originator of #476 was 
able to build a new version of BMC64 from the source code on its own, so he 
also was able to test modified BMC64 versions, but it did not help. In the end 
he modified the controller of his joystick using a small kit, and now he can 
use the joystick. I do not see a reason, why we should be more successful now.

Chris-Hal, 2025-01-17:
Hi Rene,
thank you for the feedback, that's no problem at all. For me, the new USB stick is intentionally a spare part dispenser for the old one that works. It was very cheap in the christmas sale. I only tested it initially and found that it doesn't work with BMC64. On my Linux machine it does work. 

Now that I've gained experience with the Maxi keyboard, I've repeated  the same steps on the joystick and at least I recognized the same reason for the parser hanging at the beginning, which means it has the same cause as the Maxi. So your modified kernel for the Maxi was already the solution to recognize the joystick in Circle and prevent it from hanging. So minimum that I offer to assist you if there is a demand. 

In the meantime, I had already compared the two joysticks because I was interested if there is something obvious and at least I'm attaching the data, even  I close the issue afterwards.

As I said, I can fully understand your decision.

Best regards
Christian
[Speedlink comparison.xlsx](https://github.com/user-attachments/files/18460314/Speedlink.comparison.xlsx)

closed 2025-01-17 by Chris-Hal

rsta2, 2025-01-18:
I do not understand, what I could do here, because your joystick is already 
working with Circle. Now it would be necessary to include this into BMC64, but 
this is not my project.

Chris-Hal, 2025-01-19:
I thought off, it is a matter of Circle only, that the Speedlink USB Joystick is not working but from your answer I get the impression, that related to the output, the stick should work in Circle but it is a rather a matter of BMC64, that it will not show up? 

OK, thank you

rsta2, 2025-01-19:
BMC64 uses the Circle release Step40. In this release this joystick is not 
successfully detected. In the meantime the problem has been fixed in Circle. 
The joystick works with the current release Step48, but unfortunately the 
release used in BMC64 cannot be replaced easily. It's also not clear, which 
commits in the Circle repo are necessary to solve the problem. Full story in 
issue #476.