rsta2/circle #538: RPi 5B GPIO Pins

davefilip, 2025-01-23:
Rene,

I have a question about the GPIO pins on a Raspberry Pi 5, vs. earlier models.

First, a bit of background: back with the 1B was top of the line, I built a simple ambient light sensor by connecting a CdS photo-resistor between a GPIO pin and +3.3v, and then a 10uf capacitor between the GPIO pin and GND.  I then bring the GPIO pin low to discharge the capacitor, wait for 1 second, then read the GPIO pin and wait for it to return high, counting cycles in a loop.  When later RPi models came out, I found this method unreliable -- both depending on how busy the RPi was, but also varied between RPi models / processor speeds.

One of the things that attracted me to Circle was that a task could wait for a precise number of microseconds in loops (CTimer::SimpleusDelay()), without giving up control to other tasks.  I wrote some code under Circle, which works fairly consistentently on RPi models 1B, 2B, 3B and 4B.  However, it provides very different values on a RPi 5.

```
// Create GPIO pin connected to Cds Photoresistor as Output

CGPIOPin Photo = CGPIOPin();
Photo.AssignPin(pin);
Photo.SetMode(GPIOModeOutput);

// Bring GPIO pin LOW for one second

Photo.Write(LOW);
Scheduler->Sleep(1);

// Change GPIO pin to Input and wait for HIGH
	
Photo.SetMode(GPIOModeInput);
int count = 0;

while ((Photo.Read() == LOW) && (count < GPIO_LIGHT_MAX)) {
    if (((count % 100) == 0) && debug) CLogger::Get()->Write(LogSource, LogNotice, "%d ms", count);
    CTimer::SimpleusDelay(1);
    count++;
}
```

So, for example, when RPi1->4 takes between 1900-2200 us (dark), the RPi5 takes 4000 us.  Also, when the RPi 1->4 takes 1 us (bright light), the RPi5 takes 901 us.  In-between, 101 us vs. 1900 us for the RPi5 using a lamp @ 25%.

I see on the on-line documentation for Circle that only on the RPi5, Circle can adjust the current (SetDriveStrength()) and transition speed (SetSlewRate()) of GPIO pins.  I presume that one or the other, or perhaps both, might be responsible for the differences I'm seeing on the RPi5.

So the question then becomes whether Circle running on the RPI5 sets defaults to mimic the current/transition of earlier models?  And if not, what values should I explicitly set to mimic earlier models?

Does this make sense?  If it does, the answer to this will apply to other projects, not just measuring ambient light with a photo-resistor.

Cheers,

Dave.

rsta2, 2025-01-24:
The GPIO access on the RPi 5 is slower, because the GPIO module is in the RP1 chip, which is connected via PCIe. Earlier models had this module directly in the SoC, attached to the internal bus.

`CTimer::[Simple]usDelay()`  delays approximately  and not less than the given amount of microseconds. Small deviations are possible.

I would implement this like that:

```
// Change GPIO pin to Input and wait for HIGH

Photo.SetMode(GPIOModeInput);
unsigned nStartTicks = CTimer::GetClockTicks();
unsigned nDelayMicros;

do
{
	nDelayMicros = CTimer::GetClockTicks() - nStartTicks;
	if (nDelayMicros >= GPIO_LIGHT_MAX)
	{
		break;
	}
}
while (Photo.Read() == LOW);

// Result in nDelayMicros
```

`CTimer::GetClockTicks()` returns the current value of a 32-bit microseconds up-counter. It does not necessarily start at 0.

On RPi 1-4 Circle uses the default values, which should be drive strength 8mA and slew rate not limited:

https://www.scribd.com/doc/101830961/GPIO-Pads-Control2

davefilip, 2025-01-24:
Rene,

Thanks for the feedback!

> CTimer::[Simple]usDelay() delays approximately and not less than the given amount of microseconds. Small deviations are possible.

I don’t need EXACT time, just within 10% or so … which is why I thought it was sufficient, and is certainly within range for what I need for RPi1 -> 4.

But the main reason I didn’t use:
> nDelayMicros = CTimer::GetClockTicks() - nStartTicks;

Which I first considered, is something in the Circle online doc about resetting?

So on *ix, normally I would get the system time in ms at the start and end and then subtract the two in order to get the difference in ms.

But I understand from the doc that this will reset back to zero when it hits a certain limit?  So the ending value could be less than the starting value if it circled back and reset?

It does not necessarily start at zero and may overrun after a while. It continues to count from zero then.

If so, how do I know what it looped back from?  It there a max constant so that (in pseudo code):

Start = start time
End = end time
Max = Reset

If (End < Start) {
    Delta = (Max - Start) + End
}

Otherwise, how do I know went it reset back to zero?
> The GPIO access on the RPi 5 is slower, because the GPIO module is in the RP1 chip, which is connected via PCIe. Earlier models had this module directly in the SoC, attached to the internal bus.

OK, so is it simply not possible to get a GPIO pin on a RPi5 to react as quickly as it does on earlier models, even without adjusting the slew rate?

If that is the case, then have you come across anything that also documents what the actual difference is in terms of response time?  Thanks for the link about the current, but not sure if that answers response time?  Also, not quite sure why that feature (adjustable current to meet the specified voltage) was added, and what problem it is trying to solve?

Thanks in advance for helping me understand the effect of these changes in the RPi5. I understand the RPi Foundation G-ds are trying to push everything off to the PCIe bus (e.g., SSD), and make it more like a “real” computer.

Cheers,

Dave.

rsta2, 2025-01-24:
The microseconds timer starts at some value and wraps after 0xFFFFFFFF. Next 
value is 0 then. Then it counts up until 0xFFFFFFFF and wraps again to 0. If 
you work with unsigned (or u32) values, the arithmetic automatically considers 
this:

unsigned nStartTicks;
unsigned nDelayMicros = CTimer::GetClockTicks() - nStartTicks;

For example, if nStartTicks was 0xFFFFFF00 and the current ticks value is 
0x100, the result is 0x200, which is the right value. This is used very often 
in Circle and works well.

You only have to ensure, that the loop does not run as long as about 
0xFFFFFFFF microseconds (over one hour).

> OK, so is it simply not possible to get a GPIO pin on a RPi5 to react as
> quickly as it does on earlier models, even without adjusting the slew rate?

Yes.
 
> If that is the case, then have you come across anything that also documents
> what the actual difference is in terms of response time? 

There is a test program in Circle, which displays these times in 
test/gpio-timing. Please run it to see the times.

davefilip, 2025-01-24:
Thanks Rene!  The wrap-around is clear now, and this is a great explanation that you might want to consider putting in the on-line doc somewhere (e.g., I would have guessed the internal tick counter would have been 64 bits).

Also, thanks for the GPIO speed test program - sorry I didn’t know it was there!

Cheers,

Dave.

davefilip, 2025-01-24:
Question about microseconds wrap answered (32 bit) and GPIO test program provided.

closed 2025-01-24 by davefilip

rsta2, 2025-01-25:
You are welcome.

> I would have guessed the internal tick counter would have been 64 bits).

It is, but only 32-bits are returned. For a 64-bit counter this method of 
CTimer can be used:

static u64 GetClockTicks64 (void);