rsta2/circle #542: Compatibility with Raspberry Pi Codec Zero

zanellia, 2025-02-12:
Hey! I see that the WM8960 codec is supported, but the reference hardware is the Waveshare WM8960 Hi-Fi Sound Card HAT https://www.waveshare.com/wm8960-audio-hat.htm. How much effort should I budget to get the Raspberry Pi Codec Zero https://www.raspberrypi.com/products/codec-zero/ to work with simultaneous input and output audio streams?

rsta2, 2025-02-13:
This sound card uses an other codec (DA7212) than the sound cards, which are supported by Circle so far. So one has to develop a new sound controller class for the DA7212, similar to the class [CWM8960SoundController](https://github.com/rsta2/circle/blob/master/lib/sound/wm8960soundcontroller.cpp) for the WM8960. At least the `Probe()` method must be implemented, which initializes the device registers to appropriate default values via I2C. It would be good to have an existing driver for the DA7212 as a reference. Otherwise the [datasheet](https://www.renesas.com/en/document/dst/da7212-datasheet?r=1563326) must be used.

zanellia, 2025-02-25:
Closing, since question was answered.

closed 2025-02-25 by zanellia