rsta2/circle #546: Raspberry Pi 5: MACB driver does not use all TX buffers

bjia56, 2025-03-04:
I am working on a research project that leverages Circle on the Raspberry Pi 5 (thank you for such a great project!). While we were benchmarking networking performance, we discovered that our application built with Step46 is unable to saturate the 1 gigabit ethernet connection. Part of this appears to be due to CMACBDevice's handling of m_tx_head, which is not consistently incremented:
https://github.com/rsta2/circle/blob/master/lib/macb.cpp#L1064

E.g. if SendFrame is called successfully, m_tx_head is not modified and the next call to IsSendFrameAdvisable will return false until the timeout elapses.

We are able to observe a 2-3x improvement in send throughput if the m_tx_head increment is moved from IsSendFrameAdvisable to SendFrame, similar to the uboot implementation:
https://github.com/Xilinx/u-boot-xlnx/blob/master/drivers/net/macb.c#L340-L345
This has allowed us to go from 33% to 91% saturation of the ethernet port.

The changes aren't available for upstreaming yet, but perhaps I will be able to open a PR in the future.

rsta2, 2025-03-04:
Thanks for reporting this and for appreciating Circle! I'm looking forward to 
your PR, when it is ready. I'm currently preparing a new release and can only 
accept your PR, when the release is done, so there's no hurry.

rsta2, 2025-03-28:
Now that the Circle release Step49 is done, it would be a good time to send your PR, when it is ready.

rsta2, 2025-09-18:
The originator of this issue, who wanted to improve the MACB driver, is not available any more.

closed 2025-09-18 by rsta2