rsta2/circle #552: Need help building LVGL

PanosMich, 2025-03-25:
Hello,
I am trying to get started with Raspberry PI bare-metal programming under Windows.
I have successfully setup Git, the toolchain, the make tool and the Circle framework.
Various Circle samples run fine.
Circle includes LVGL as an addon. I have included the LVGL code.
When I execute "make" to build the library, I get no errors or messages, but no files are created.
Kindly assist.

rsta2, 2025-03-25:
Building the LVGL libraries can take a longer time. Did you interrupt this 
process? There was also a problem with long path names in Circle Step48. 
Please use Step49. There may also be problems with white spaces in path names 
on Windows. Please use a directory for the Circle project, which does not have 
white spaces in its path.

PanosMich, 2025-03-25:
> Building the LVGL libraries can take a longer time. Did you interrupt this
> process? There was also a problem with long path names in Circle Step48.
> Please use Step49. There may also be problems with [...]

Thank you for your reply.
I did not interrupt the process.
I am already using Step49.
The path has no white spaces.

rsta2, 2025-03-25:
Perhaps somebody can help, who has a Windows machine and built LVGL with 
Circle? Unfortunately I cannot test this at the moment, because I do not have 
Windows.

PanosMich, 2025-03-25:
Shall I assume that, in your system, lvgl is built with no problems?

rsta2, 2025-03-26:
Yes, it successfully builds on Linux here.

rsta2, 2025-03-26:
Now I reactivated an old Windows installation, and I can confirm, that *addon/lvgl/* cannot be built with the environment, described in *doc/windows-build.txt*. The problem is an incompatibility in the file [lvgl.mk](https://github.com/lvgl/lvgl/blob/7f07a129e8d77f4984fff8e623fd5be18ff42e74/lvgl.mk) with the way, it is used in Circle's Makefile.

Currently I can only provide a workaround. Just open *addon/lvgl/lvgl/lvgl.mk* in an editor and remove all apostrophes (') from it. After saving the file, it built here successfully on Windows.

I have to see, if there is a better solution for this issue.

PanosMich, 2025-03-26:
I have already done this. Before that, make was returning a bunch of "file not found" errors.

I setup WSL and tried to build lvgl with that. I got a "make: *** No rule to make target 'lvgl/lvgl/lvgl.h', needed by 'lvgl.o'.  Stop." error, even though lvgl.h does exist. I don't know why there is an extra "lvgl/" in the path mentioned.

PanosMich, 2025-03-26:
I have already done this. Before that, make was returning a bunch of "file not found" errors.

I setup WSL and tried to build lvgl with that. I got a "make: *** No rule to make target 'lvgl/lvgl/lvgl.h', needed by 'lvgl.o'. Stop." error, even though lvgl.h does exist.

rsta2, 2025-03-26:
> I have already done this. Before that, make was returning a bunch of "file not found" errors.

OK, and what happened then after removing the apostrophes? It did work here.

*lvgl.d* is an auto-generated (perhaps stale) file and should not be edited. Remove it and try again.

You can also add the line `CHECK_DEPS=0`, before the file *Rules.mk* is included, to the *Makefile* in *addon/lvgl/*. This will switch off dependency checking and also speeds up the build.

rsta2, 2025-03-27:
LVGL should build now on Linux and Windows (environment like in *doc/windows-build.txt*) with the Circle version on the *develop* branch.

PanosMich, 2025-03-27:
Rebuilding Circle under WSL made it starting to build, but it now returns a ton of warnings and errors, making the library unusable for the samples included.
I am doing several reinstallations of Circle and LVGL and trying setups under both Windows/Git and Windows/WSL/Ubuntu hoping to find the cause.

rsta2, 2025-03-27:
Do you get any specific error messages under Windows/Git, when trying to build LVGL with Circle? I cannot help, if you do not give more information.

I'm out with Windows/WSL, because I do not have it. All what I know is, that some users are using WSL successfully to build Circle. They also provided the WSL info in some doc files in Circle.

PanosMich, 2025-03-27:
I managed to build LVGL with the Circle develop branch. I was using master so far.
However, trying to build any sample that includes LVGL fails with errors.
The LVGL branch I built was master. Should I build some other instead?

rsta2, 2025-03-28:
OK, good. You have to checkout the tag `v9.2.2` in *addon/lvgl/lvgl/*. This is automatically done, when you enter `git submodule update` from Circle project root.

By default there is only one LVGL demo build-able with Circle. It's the widget demo and you have to go to *addon/lvgl/sample/* and enter `make` there to build it. Before of course the Circle libraries (with `makeall` from project root) and the LVGL library must be build from *addon/lvgl/* with `make`.

With these modifications the sample program in *addon/lvgl/sample/* runs *lv_example_get_started_1* after rebuilding the LVGL library and the sample:

```
diff --git a/addon/lvgl/Makefile b/addon/lvgl/Makefile
index 93049014..2b30257f 100644
--- a/addon/lvgl/Makefile
+++ b/addon/lvgl/Makefile
@@ -9,7 +9,7 @@ LVGL_PATH = ./lvgl
 # From lvgl/lvgl.mk:
 CSRCS += $(shell find $(LVGL_PATH)/src -type f -name *.c)
 #CSRCS += $(shell find $(LVGL_PATH)/demos -type f -name *.c)
-#CSRCS += $(shell find $(LVGL_PATH)/examples -type f -name *.c)
+CSRCS += $(shell find $(LVGL_PATH)/examples -type f -name *.c)
 CSRCS += $(shell find $(LVGL_PATH)/demos/widgets -type f -name *.c)
 
 OBJS = lvgl.o $(CSRCS:.c=.o)
diff --git a/addon/lvgl/lv_conf.h b/addon/lvgl/lv_conf.h
index 7af96b73..8760d8e9 100644
--- a/addon/lvgl/lv_conf.h
+++ b/addon/lvgl/lv_conf.h
@@ -1086,7 +1086,7 @@
 *==================*/
 
 /*Enable the examples to be built with the library*/
-#define LV_BUILD_EXAMPLES 0
+#define LV_BUILD_EXAMPLES 1
 
 /*===================
  * DEMO USAGE
diff --git a/addon/lvgl/sample/kernel.cpp b/addon/lvgl/sample/kernel.cpp
index 93bdcb06..682b5f9b 100644
--- a/addon/lvgl/sample/kernel.cpp
+++ b/addon/lvgl/sample/kernel.cpp
@@ -18,7 +18,7 @@
 // along with this program.  If not, see <http://www.gnu.org/licenses/>.
 //
 #include "kernel.h"
-#include "../lvgl/demos/lv_demos.h"
+#include "../lvgl/examples/get_started/lv_example_get_started.h"
 
 static const char FromKernel[] = "kernel";
 
@@ -163,7 +163,7 @@ TShutdownMode CKernel::Run (void)
 #ifdef I2C_DISPLAY
 	lv_example_button ();
 #else
-	lv_demo_widgets ();
+	lv_example_get_started_1 ();
 #endif
 
 	while (1)
```

PanosMich, 2025-03-28:
Tried "git submodule update" from Circle project root.
Rebuilding Circle shows that everything is already up to date.
Rebuilding LVGL shows that liblvgl.a is up to date.

Building addon/lvgl/sample returns:

```
  LD    kernel8-32.elf
C:\Programs\ARM_GNU_Toolchain\Win\bin\arm-none-eabi-ld.exe: ../../../addon/lvgl/liblvgl.a(lv_string_clib.o): in function `lv_strnlen':
C:\programs\circle\addon\lvgl/lvgl/src/stdlib/clib/lv_string_clib.c:65:(.text+0x14): undefined reference to `strnlen'
C:\Programs\ARM_GNU_Toolchain\Win\bin\arm-none-eabi-ld.exe: C:\programs\circle\addon\lvgl/lvgl/src/stdlib/clib/lv_string_clib.c:65:(.text+0xb4): undefined reference to `strnlen'
make: *** [kernel8-32.img] Error 1
```

Building circle/sample/43-multiwindow returns

```
  CPP   main.o
In file included from ../../addon/lvgl/lvgl/lvgl.h:131,
                 from ../../addon/lvgl/lvgl.h:25,
                 from kernel.h:37,
                 from main.cpp:20:
../../addon/lvgl/lvgl/src/lv_api_map_v9_1.h:93:2: warning: #warning LV_FS_DEFAULT_DRIVE_LETTER is deprecated. Rename to LV_FS_DEFAULT_DRIVER_LETTER [-Wcpp]
   93 | #warning LV_FS_DEFAULT_DRIVE_LETTER is deprecated. Rename to LV_FS_DEFAULT_DRIVER_LETTER
      |  ^~~~~~~
In file included from kernel.h:38:
multicoredemo.h:35:1: error: expected class-name before '{' token
   35 | {
      | ^
make: *** [main.o] Error 1
```

PanosMich, 2025-03-28:
Followed a different approach. Being a hardware developer myself (and apparently git-ignorant), I did it git-lessly.
I downloaded the circle-develop and the lvgl-v.9.2.2 code and built everything under WSL.
All code and samples are built and run flawlessly.
Your specification of the required circle branch and lvgl version was VERY significant. Was this mentioned in any of the readme files?

Thank you for spending so much of your precious time with my problem. I'm certain you had better things to do.

rsta2, 2025-03-28:
That's also a working approach, good that it works now!

In addon/lvgl/sample/README is described, how the LVGL sample has to be build. 
But in fact it didn't work on Windows/Git from the beginning, because the fix 
above was necessary, which was done on the "develop" branch, where Circle 
development takes place. It will be in the next release on "master" branch. 
Thanks for pointing me on this issue!

You are welcome. It's OK, perhaps we could have been solved this a bit 
quicker, but in the end it has been solved and that's important.