rsta2/circle #554: Serial flashing problems

PanosMich, 2025-04-09:
Hello Rene,

I am sorry to bother you again, but I am having problems flashing code with the serial port.

I tried the latest Flashy and it works great with files around 200K. However, trying to upload 2M files fails returning a "timeout awaiting response" error.

I also tried cflashy, but it does not seem to work.
For example, in circle/sample/02-screenpixel, I build the sample using make and I use "objcopy -O ihex kernel8-32.elf kernel8-32.hex" to produce the .hex file.
After that, executing:
"make flash"
or "../../tools/cflashy /dev/ttyS5 --flashBaud:115200 --userBaud:115200 kernel8-32.hex"
or "../../tools/cflashy /dev/ttyS5 --flashBaud:115200 --userBaud:115200 --reboot:REBOOT kernel8-32.hex"
or "../../tools/cflashy /dev/ttyS5 kernel8-32.hex"
the command waits for ever with no feedback.

Sniffing the serial port I found out that commands without the --reboot:REBOOT option send a bunch of 0x80 characters and a "R" at the end, while the bootloader returns a "IHEX-F" message.

Commands with the --reboot:REBOOT option send a "REBOOT", then a bunch of 0x80 characters and a "R" at the end, while the bootloader returns a "IHEX-F<CR><LF>#ERR:format<CR><LF>IHEX-F" message.

Any thoughts?

rsta2, 2025-04-09:
The serial bootloader does allow to load kernel images with a size of up to 2 MBytes by default. With this patch you can load up to 8 MBytes:

```
diff --git a/tools/bootloader/vectors.s b/tools/bootloader/vectors.s
index d9cf4791..3b644f9b 100644
--- a/tools/bootloader/vectors.s
+++ b/tools/bootloader/vectors.s
@@ -6,7 +6,7 @@
 _start:
     b skip
 
-.space 0x200000-0x8004,0
+.space 0x800000-0x8004,0
 
 skip:
     mov sp,#0x08000000
diff --git a/tools/bootloader/vectors64.s b/tools/bootloader/vectors64.s
index 8122db95..c20ec546 100644
--- a/tools/bootloader/vectors64.s
+++ b/tools/bootloader/vectors64.s
@@ -6,7 +6,7 @@
 _start:
     b skip
 
-.space 0x280000-0x80004,0
+.space 0x880000-0x80004,0
 
 skip:
     mov sp,#0x08000000
```

You have to build the bootloader images again after applying the patch with `make bootloader` or `make bootloader64` in *boot/*.

It looks all right, what you have sniffed from cFlashy via the serial line. But perhaps it creates less effort to use the normal Flashy with this patch above, but to debug this problem further. BTW. This default 2 Mbytes size limit also applies to cFlashy.

PanosMich, 2025-04-11:
The patch works fine. Thank you for that.
I got flashy v.1 to work up to 230400 baud, however only with the nofast flag. For some reason, in my case, fast transfer is not working with large files at any speed. I don't know if it's because I am using WSL1.
Have you ever tried flashy v.3 with the 43-multiwindow sample? Did it work for you?

rsta2, 2025-04-11:
Unfortunately I never used WSL(1). I also did not try Flashy v3.

rsta2, 2025-04-24:
Did you find a solution?

PanosMich, 2025-04-24:
Thank you for asking.
I think I found one. It is called SDWire3. I should be receiving it in the following days.
If you want, I can send you feedback once I try it.

rsta2, 2025-04-24:
Thanks for info! That's an interesting solution and probably much quicker than a serial bootloader, but also more expensive.