rsta2/circle #555: Cannot get SD Card to work in QEMU

abeanater, 2025-04-11:
I am working on a project that requires reading of the SD card or USB drive and would like to use QEMU to prototype it.  I built the special QEMU version [here](https://github.com/smuehlst/qemu.git).

Here are the arguments I am using when calling qemu:
```
./qemu-system-aarch64.exe   -M raspi4b   -cpu cortex-a72   -smp 4   -m 2G   -kernel /c/projects/bare-metal-llm/circle-stdlib/checkout/sqlite/src/kernel8-rpi4.img   -drive id=sd,if=none,file="/c/projects/bare-metal-llm/circle-stdlib/checkout/Circle-llm/fla
sh.img",format=raw -s -S
``` 

It always fails here in emmc.c:
```
boolean CEMMCDevice::Initialize (void)
{
#ifndef USE_SDHOST
#if RASPPI == 4
	// disable 1.8V supply
	CBcmPropertyTags Tags;
	TPropertyTagGPIOState GPIOState;
	GPIOState.nGPIO = EXP_GPIO_BASE + 4;
	GPIOState.nState = 0;
	if (!Tags.GetTag (PROPTAG_SET_SET_GPIO_STATE, &GPIOState, sizeof GPIOState, 8))
	{
		return FALSE;
	}
```

Anything I am doing wrong?  My hope is to read from a file so I can validate stuff.

rsta2, 2025-04-11:
Is it possible, that you use the -M raspi3b option instead? The RPi 4
emulation is relatively new in QEMU and does not support USB and SD card.
Unfortunately there have also been problems with the multi-core support with
Circle apps, running in QEMU.

abeanater, 2025-04-12:
@rsta2 you appear to be correct.  It works with the raspi3b option.  I will continue to use that for prototyping before moving to hardware.  Thank you!

closed 2025-04-12 by abeanater