rsta2/circle #556: i2s input using aux clock results in random crashes

nickverlinden, 2025-04-19:
Hi Rene,

I've been testing my device for some time now and seems to get random crashes when i'm suing the i2s input using an auxiliary clock. To be honest, i don't even know if it's really the aux clock's fault. It's hard to tell what's going on because there is no log message, or at least it doesn't come through the serial port. Just a black screen (i'm using a DSI touchscreen). First i thought it was something else like the UDP sockets i was using, but after many hours of testing it seems that the i2s input is causing it (disabling this feature with my compile flags, the crashes don't occur anymore). The audio input works fine and sounds good, so i don't think it's a i2s settings mismatch or something. Also USB audio input works fine, no crashes. I'm using the Circle dev branch at the time the new cflashy modifications for macOS were added (because i'm running macOS).

Do you have any idea what could be going on, is it possibly related to generating the clock signal? Sometimes circle also crashes during boot when the aux clock is initialised, so that's another reason why i suspect the clock is the issue.

Many thanks for thinking with me!

This is the code i use to initialize the clock:
```c
// first set up clock pins
CKernel::CKernel (void)
:	m_Screen (m_Options.GetWidth (), m_Options.GetHeight ()),
	m_2DGraphics(m_Options.GetWidth (), m_Options.GetHeight (), false),
	m_2DGraphicsHDMI0(m_Options.GetWidth (), m_Options.GetHeight (), false, 1),
	m_Timer (&m_Interrupt),
#if RASPPI >= 4 || !defined (USE_USB_FIQ)
	m_Serial (&m_Interrupt, TRUE),
#else
	m_Serial (&m_Interrupt, FALSE),
#endif
	m_Logger (m_Options.GetLogLevel (), &m_Timer),
	m_I2CMaster (CMachineInfo::Get ()->GetDevice (DeviceI2CMaster), TRUE),
	m_USBHCI (&m_Interrupt, &m_Timer, TRUE),
	m_EMMC (&m_Interrupt, &m_Timer, &m_ActLED),
#ifdef HAL_CIRCLE_AUDIO_I2S_IN
	m_pSoundInI2S(0),
#endif
	m_pSoundIn (0),
	m_pSoundOut (0),
	//m_pMIDIDevice[HAL_CIRCLE_MIDI_MAX_USB_DEVICES] (0),
#ifdef HAL_CIRCLE_AUDIO_I2S_AUX_CLOCK
	m_SCLKPin (4, GPIOModeAlternateFunction0),
	m_Clock (GPIOClock0, GPIOClockSourceOscillator),
#endif
...

// later initialize clock
CLogger::Get ()->Write (FromKernel, LogNotice, "starting auxilary i2s clock");
		unsigned nSampleRate = HAL_CIRCLE_AUDIO_SAMPLE_RATE;
		unsigned nChanLen = 32;
		unsigned nChans = HAL_CIRCLE_AUDIO_I2S_IN_CHANNELS;
		unsigned nClockFreq =
			CMachineInfo::Get ()->GetGPIOClockSourceRate (GPIOClockSourceOscillator);
		assert (nClockFreq > 0);
		assert (8000 <= nSampleRate && nSampleRate <= 192000);
		assert (nClockFreq % (nChanLen*nChans) == 0);
		double fDiv = nClockFreq / ((float)HAL_CIRCLE_AUDIO_I2S_IN_CHUNK_SAMPLES) / nSampleRate; // 512 = frame size, on my pcm1808 board this is set using the md0 pin; set it high
		unsigned nDivI = fDiv;
		unsigned nDivF = (fDiv - nDivI) * 4096.0 + 0.5;
		assert (nDivF <= 4096);
		if (nDivF > 4095)
		{
			nDivI++;
			nDivF = 0;
		}

		if (!pKernel->m_Clock.Start (nDivI, nDivF, nDivF > 0 ? 1 : 0))
		{
			CLogger::Get ()->Write (FromKernel, LogError, "cannot start i2s aux clock");
		}
```

rsta2, 2025-04-20:
Hi Nick, I have currently no idea, how the aux clock could trigger this 
problem. But if you are using the serial device in interrupt mode for the 
logger, log messages can get lost, when the system crashes. The messages are 
written to an internal buffer in this case and send out by the interrupt 
handler. But the interrupt handler stops working, when the crash happens, so 
you cannot see all messages. Can you comment out the respective member 
initializer for m_Serial for a test. This will operate the serial device in 
non-interrupt mode.

nickverlinden, 2025-04-20:
Hi Rene,

I tried it, but got a black screen after some time of running and no log message. I also made sure there is no memory leak and tested if the screen just didn't go black and still running but unfortunately the pi is really locked up. Is it possible you think that there is an interference problem? The i2s input board is very close to the (very short, like 3cm) ribbon cable that connects the display to the dsi port. I already tried to wrap aluminium foil around the dsi cable and the short dupont cables connecting the i2s board to the pi, but nothing helps. Perhaps my clock dividers are wrong, but i don't see how this would cause a complete lockup with a black display. I would expect the audio then would just drop out or something.

Just to make sure i just commented out the clock start code from the sample above, and then the black screen lockup does not occur, so even though the i2s input code is still running and calling `pSoundInI2S->Read (Buffer, sizeof Buffer)` but not the clock. I'm quite desperate, i have 10 pi's running in parallel, everything works perfect for a very long time unless i'm using i2s input with the aux clock. All of the 10 pi's lock up at some point. Some of them more fast than others, but eventually all of them lock up.

Kind regards,
Nick

rsta2, 2025-04-20:
Hi Nick, do you have the system option REALTIME enabled? This can also 
suppress log messages from IRQ handlers, in case the crash occurs there.

I'm not an electronics engineer, so I cannot talk much about  interference. 
But when the DSI display, which is driven by the firmware on the VideoCore 
coprocessor, goes black, it's possible that this is a hardware problem.

Can you tell me more, which ADC you are using and why it needs an external 
clock? Where does the frame clock signal (FS) come from? Which are the 
parameter values for GPIO clock generation (sample rate, channels, chunk 
samples)?

You may have look at section 5.4 in this document on constraints for GPIO 
clock generation on the RPi 4:

https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf

I could do a long time test with sample/41-soundinput here to check, that 
there is no problem with the I2S sound driver, but because I do not have your 
hardware, I cannot test the aux clock. I assume, you are using RPi 4B's?

Another possibility would be to JTAG debug this, but if it is a hardware 
problem, that would also not help.

Rene

nickverlinden, 2025-04-21:
Indeed, the REALTIME option is enabled, i believe this had to be enabled because i'm also allowing an external usb audio interface to be connected and was required as stated in the README.md when using usb audio interfaces.

The ADC used is one of those PCM 1808 boards you can find on aliexpress (https://nl.aliexpress.com/item/1005007130138539.html). These don't have onboard crystals to generate the right SCLK, so i'm using the GPIO to generate this clock that is connected to the SCK pin on the board. I must be honest and say i'm not an expert on this but according to the spec sheet of the ADC it should derive the FS clock from the SCK that you feed it.

Now, i have complete faith in your code, it seems unlikely there is a problem with the I2S sound driver however i guess at this point anything is possible. If you want i can donate some of these PCM1808 boards to you (i have plenty, and could send you some). 

I had a look at the clock doc again. According to the datasheet (https://www.ti.com/lit/ds/symlink/pcm1808.pdf?ts=1745145282835&ref_url=https%253A%252F%252Fwww.google.com%252F) it states at page 15 that it needs a cock at 24.576Mhz, i'm using the 512 Fs mode and want to sample 2ch 48khz. Now looking at the doc, i don't even know how i figured out the divider to get 24.576Mhz. So i doubled checked things and the dividers i used before don't make any sense, and it's weird that it was working at all. I'm also using a different clock source now to get the exact clock rate. The correct dividers seem to be:

```sh
CKernel::CKernel (void)
:	
...
	m_SCLKPin (4, GPIOModeAlternateFunction0),
	m_Clock (GPIOClock0, GPIOClockSourcePLLD),
}
...

pKernel->m_Clock.Start (30, 2112, 0);
```

So i had high hopes with these findings and did some testing with these new clock settings but unfortunately the black screen lockup still happened. What's even more strange now is that with these clock settings i'm getting audio drop outs when writing to the PWMSoundDevice. I tried many different chunk & queue size combinations, but couldn't get rid of the dropouts entirely. I'm about to give up at this point unless you have any more ideas i can try? On the good side, usb audio is working great.

rsta2, 2025-04-21:
Thanks for the info!

I think the FS clock (GPIO19) is at the LRC pin of the module.

Can you please try:

pKernel->m_Clock.Start (30, 2120, 1);

With the MASH parameter set to 0 the MASH filter is disabled and the fractional 
part is not used.

If this does not work, I would reduce the SCK frequency, because with 24.576 
MHz it is near to the maximum allowed frequency for the MASH filter. With 
setting the MD1 pin to High (256 fs), I would try:

pKernel->m_Clock.Start (61, 144, 1);

I don't know, if it helps, but I would give it a try.

nickverlinden, 2025-04-22:
Thx Rene, i'm trying your clock settings now. I let my pi4b run overnight and interestingly this morning there was a gray screen instead of a black screen. In all the crashes the last couple of weeks that never happened. Maybe that means something different like overheating or something. With your clock settings the pwmsoundbase dropouts also disappeared so it's already an improvement. I'm going to test some more this week, and maybe also try the fs256 mode if that doesn't work out.

nickverlinden, 2025-04-22:
Ah shoot, the first clock settings crashed. I want to try the 256 fs setting, but then i need to make some modifications which will not be able to do easily. I could try to test with one board. In the meantime, i found another board on aliexpress with the same adc (pcm1808) that seems to have a clock crystal onboard (https://nl.aliexpress.com/item/1005008139232519.html). I ordered some to test. It will take some time to arrive, but i'll update when i have more info.

rsta2, 2025-04-24:
OK.

nickverlinden, 2025-04-24:
After running about 6 to 7 hours, i got a black screen (without even initialising or starting the clock), so perhaps there is an issue with i2s...
I will perform some tests again with the pi's running without i2s & clock. To confirm they don't go black when no i2s is being used.

rsta2, 2025-04-24:
:( Let me know, when I can do something (e.g. test with *sample/41-soundinput*).

In any case you should disable the interrupt mode for the serial (debug) device and the `REALTIME` system option. Otherwise eventual log messages on a crash will not appear.

nickverlinden, 2025-04-24:
Will do, thx. I will do some more extensive testing before reporting back. It might take a while though, wish me luck :-) !

nickverlinden, 2025-04-27:
So, after many hours of testing and changing things i managed to get it to run stable. The big question is what is really going on?

First i tried running sample 42-soundinput, and it ran stable for more than 16 hours. Then i was thinking, what is different compared to my project? The big ones: mine uses all 4 cores, and is updating the display at 60fps.

So then i tried running my display update code on the same core as the audio core. Now everything runs stable (tested up to ~24 hours). The big question remains why? When not using i2s input (but am using USB audio, or audio from a buffer in ram), running it on 2 seperate cores works fine, but calling pSoundInI2S->Read instead of the usb input, there's a possibility for a freeze.

My knowledge of these things are quite superficial, but could this possible be a DMA transfer problem? That at random moments the display buffer is being updated on another core and is happening right at the same the I2S copy is happening?

rsta2, 2025-04-27:
Thanks for info! I think this is good news.

The bandwidth of the bus is finite and is shared between SDRAM access of the 4 cores, if the various caches do not hit, and DMA access. So yes, there may be moments, at which for example a DMA channel tries to access the frame buffer memory in SDRAM and at the same time the I2S peripheral requests new data from SDRAM.

There is a priority parameter for DMA transfers (0-15), which is set to 1 (relatively low) for I2S transfers. I don't know the priority for video transfers, because this is programmed by the firmware, but I guess it's priority is higher. But there is also a panic priority parameter, which is used, when a DMA channel is about to run out of data. It is set to 15 (maximum) for I2S DMA.

So what could happen is, that the I2S DMA is about to run out of data and does transfers using the maximum priority. This could block the video DMA transfers at lower priority and the display freezes, if a DMA error occurs and the firmware stops working then. This is just a bit speculation, but may happen.

You could experiment with the panic priority value for I2S DMA and decrease it, to see what happens. It is set [here](https://github.com/rsta2/circle/blob/aa17b99a527274571842663d70d546f7ec30b4b8/lib/sound/dmasoundbuffers.cpp#L171).

nickverlinden, 2025-04-27:
Thx Rene, I first tried changing it to 7, then 3, then 1, but all of them lock up. I made sure to always rebuild the circle libs after changing the panic priority. I did notice some audio glitches when set to a lower value, so it has some effect.

Maybe it's something else. I'm going to try and poke around some stuff. If you think of anything else i can test, let me know!

nickverlinden, 2025-04-28:
I'm trying to find what piece of code is causing the lockup. I already found out it's not `m_p2DGraphics->UpdateDisplay();`, doing that on the audio core, but keep doing the other display stuff on another core still causes it to lock up.

At this point it's becoming very hard to debug. There are two memcpy operations that happen before updating the display, first i thought those might cause it, but even when commenting those out, lockups still happen.

Now i'm brute forcing and commenting out pieces of code for the display stuff. I try to minimise it, but i'm doing a lot of framebuffer manipulation. I wonder if this could be some memory access issue where i read and write to the same address at the same time on different cores. I was careful not to do this, but it might have sneaked into my code somewhere. It does not explain however why there does not seem to be a problem when not using i2s.

Any thoughts that help?

rsta2, 2025-04-28:
I still guess, it's the DMA operation. There are more parameters (burst size, 
normal (not panic) priority), but we can only control the DMA controllers used 
on the ARM size, not the VideoCore DMA.

Are you sure, that the whole system freezes or can it be, that only the DSI 
display update procedure stops? Can you write a character to the serial 
interface from time to time and see on a terminal, if it still arrives, when 
the display goes black?

nickverlinden, 2025-04-28:
Yes, i'm sure it's completely locked up. I tested this some time ago because i had the same thought, but unfortunately not.

Just a thought, but could increasing cpu clock speed help?

rsta2, 2025-04-28:
Can you please add the following additions to your application. This is for *sample/04-timer*, but should fit in any app. I don't know, if this works, but hopefully it shows an error message, when an ARM AXI error or ARM address error occurs. I cannot test it here, because these hardware conditions can't be easily triggered. You have to disable the `REALTIME` system option and the interrupt mode for the serial interface, so that there is a change, that it works. This is for the RPi 4 only.

```
diff --git a/sample/04-timer/kernel.cpp b/sample/04-timer/kernel.cpp
index de8c0a51..ddaf03fd 100644
--- a/sample/04-timer/kernel.cpp
+++ b/sample/04-timer/kernel.cpp
@@ -75,6 +75,9 @@ TShutdownMode CKernel::Run (void)
 {
 	m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);
 
+	m_Interrupt.ConnectIRQ (GIC_SPI (38), ARMAddressErrorHandler, this);
+	m_Interrupt.ConnectIRQ (GIC_SPI (39), ARMAXIErrorHandler, this);
+
 	m_Logger.Write (FromKernel, LogNotice, "An exception will occur after 15 seconds from now");
 
 	// start timer to elapse after 15 seconds
@@ -109,3 +112,17 @@ void CKernel::TimerHandler (TKernelTimerHandle hTimer, void *pParam, void *pCont
 	asm volatile (".word 0xFF000000");
 #endif
 }
+
+void CKernel::ARMAXIErrorHandler (void *pParam)
+{
+	CKernel *pThis = (CKernel *) pParam;
+
+	pThis->m_Logger.Write (FromKernel, LogPanic, "ARM AXI error");
+}
+
+void CKernel::ARMAddressErrorHandler (void *pParam)
+{
+	CKernel *pThis = (CKernel *) pParam;
+
+	pThis->m_Logger.Write (FromKernel, LogPanic, "ARM address error");
+}
diff --git a/sample/04-timer/kernel.h b/sample/04-timer/kernel.h
index d9ee48a1..239f546d 100644
--- a/sample/04-timer/kernel.h
+++ b/sample/04-timer/kernel.h
@@ -50,6 +50,9 @@ public:
 	
 	static void TimerHandler (TKernelTimerHandle hTimer, void *pParam, void *pContext);
 
+	static void ARMAXIErrorHandler (void *pParam);
+	static void ARMAddressErrorHandler (void *pParam);
+
 private:
 	// do not change this order
 	CActLED			m_ActLED;
```

nickverlinden, 2025-04-29:
I ran this at night, this morning it was frozen, but no message on the serial cable. I made sure REALTIME is not enabled, and that the previously mentioned Interrupt for the m_Serial initializer is also commented. I'll try again!

nickverlinden, 2025-04-29:
I just updated the Circle repo to the latest on the dev branch, and rebuilt the bootloader + new firmware. I'm letting things run until i get back from work!

rsta2, 2025-04-29:
Unfortunately this does not mean, that there is no problem with the bus. :( 
Maybe the system totally freezes and the AXIERR interrupt cannot be processed 
any more. Thanks for testing!

I had now time to think about this longer. Perhaps you really have to reduce 
the load on the bus by doing less display updates or DMA transfers at the same 
time. What is really odd is, that the display update stops, which is not 
normal.

One of us could ask a question in the Raspberry Pi bare-metal forum, if a bus 
lock can happen on high bus load on the RPi 4:

https://forums.raspberrypi.com/viewforum.php?f=72

But this makes only sense with a description of what your application is doing 
in more detail. Especially which devices (e.g. displays) are accessed by the 
different cores and which DMA transfers are running.

nickverlinden, 2025-04-29:
Thx Rene. Well, the test with updated Circle repo and freshly built bootloader & firmware still caused a lockup so it's not something that was fixed in a firmware update. Unfortunately the USB Serial dongle that i'm using sometimes decides to stop working while monitoring and that happened somewhere during the day. There was no message on crash, but can't be sure the dongle was still receiving messages when the crash happened.

I think i don't really want to invest a lot of time in this anymore. I'm already happy running them both on the same core, albeit with a larger queue so more latency but i'm willing to go with that for now.

What i am still going to do:
-Do a test halving the frame rate from 60 to 30
-Do a test with a longer i2s queue

rsta2, 2025-04-30:
OK Nick, understood. In case I have another idea, what can be tested, I will 
let you know.

nickverlinden, 2025-05-01:
Thx for all the help so far, really appreciate it! Going to close this one for now.

closed 2025-05-01 by nickverlinden

nickverlinden, 2025-05-14:
Hi Rene, I've got some more updates around this issue. So running the code on the same core still caused the black screen, it just occured a lot less fast. I noticed this after deploying the updated code to 9 other pi's, and after some hours a couple of them start giving me the black screen at random moments. Initially i gave up, but later one i got the idea to test one more thing. I was researching the possibility to use the esp32 for another project and was reading that the PWM also used DMA on that mcu. That's when i though, maybe the pi also uses DMA for the PWM audio output, so perhaps i can disable PWM audio output in Circle (by not initializing it) and see what happens. So after testing a lot, i found out that when i2s input is enabled, and pwm sound output is disabled the black screen does not occur. I did 2 tests, one for 12 hours, and one for 18 hours and not a single one crashed. This makes me happy because i was not satisfied by the quality of the PWM output anyway. But perhaps this could give us a clue for the future. Maybe there is an issue using PWM, I2s and display updates at the same time.
At some point in the future i want to use audio output as well, so i'm probably going to try i2s RX & TX, and hopefully that doesn't give any black screen. I'll let you know in this issue.

rsta2, 2025-05-14:
Thanks for info, Nick!