rsta2/circle #560: Ableton Link Support

DatanoiseTV, 2025-04-30:
Hi,

I have started porting some examples of Ableton Link which would add nice functionality to circle, but it need some wrappers.
So far, I have wrappers for the timer etc, but we need some kind of asio implementation.

Is anyone interested in helping with this? Perhaps a good starting point is looking the ESP32 port:
https://github.com/Ableton/link/tree/master/include/ableton/platforms/esp32

rsta2, 2025-05-05:
I'm familiar with Circle (but not with Ableton Link) and I would help. But I'm afraid Ableton Link requires the ability to receive multicast network frames? This is currently not supported by Circle.

DatanoiseTV, 2025-05-18:
That is correct. Could maybe lwip be used as TCP/IP stack for circle? Afaik, it supports multicast.

rsta2, 2025-05-19:
Circle has its own TPC/IP stack, which cannot easily replaced with an other stack. This is also not the main problem. The (five different) low-level NIC drivers in Circle do not support multicast receive yet. I'm thinking about implementing multicast receive in Circle, but it will take time.

rsta2, 2025-05-22:
IP multicast send and receive support is now on the *develop* branch. There is a test program in *test/mdns-receiver*, which demonstrates the receiver. After calling `Connect()` for a socket, one can also send to the respective multicast address. Using `SendTo()` is also possible. This all works only in the local network (LAN or WLAN), because the IGMP is not implemented.

DatanoiseTV, 2025-05-22:
Nice, I tried Jules the other day and it started implementing IGMPv2 support, but didn't test yet: https://github.com/DatanoiseTV/circle/tree/igmp-support-v1

I think one problem for Ableton Link might be also that it seems to need asio and pthread.

rsta2, 2025-05-22:
You should have told, that you are working on this. ;) My code is slightly 
tested and comes with multicast support for all supported NICs, but without 
IGMP. GPT told me, it is not really necessary, but maybe I did not understand 
Ableton Link right to ask the right question.

Circle has some kind of pthread emulation in addon/linux/. But it has only a 
subset of features and Circle has a cooperative scheduler, which means it 
cannot be fully compatible. Asio support is not available at all.

DatanoiseTV, 2025-05-23:
> You should have told, that you are working on this. ;) My code is slightly
> tested and comes with multicast support for all supported NICs, but without
> IGMP. GPT told me, it is not really necessary, but maybe I did not understand
> Ableton Link right to ask the right question.

It was jules and it was just my first tryout with jules. Not sure if it works at all.

> 
> Circle has some kind of pthread emulation in addon/linux/. But it has only a
> subset of features and Circle has a cooperative scheduler, which means it
> cannot be fully compatible. Asio support is not available at all.

Ok good to know. Maybe then it can be also implemented without asio. Would be great if we get it running. Would open tons of new possibilities.

rsta2, 2025-05-23:
I had a look at your code and it's astonishing, what these tools can do today. But I never used them to code for Circle so far. I would also do some things differently because of some constraints for Circle, which you and the AI could not know. For example the base library *libcircle.a* in *lib/* must not use references to other Circle libraries (like in *macaddress.h*), because it's the lowest layer. I don't want to have circular dependencies here.

I did some further reading and discovered, that there are IGMP snooping switches out there, which require IGMP implemented, even in a local network environment. So I will implement this too, and maybe will use some of your code. Unfortunately it's a little bit difficult to test this without such switch.

Unfortunately I'm not familiar with the whole thing (Ableton Link, ASIO etc.). I did read a paper about Ableton Link and understood, what it is for, but not more. But I can imagine, that would be great to get it running on Circle. So let me know, when I can do something towards this.

DatanoiseTV, 2025-05-24:
> I did some further reading and discovered, that there are IGMP snooping switches out there, which require IGMP implemented, even in a local network environment. So I will implement this too, and [...]

I asked ChatGPT and it seems there is several ones supporting IGMP snooping in a very affordable price range:

1. Netgear GS105Ev2 / GS108Ev3 (Smart Managed Plus)

    Ports: 5 or 8

    Price: ~$30–$50

    IGMP: Supports IGMP snooping v1/v2/v3 (configurable via web interface)

    Features: VLANs, QoS, port mirroring

    Note: Requires Windows or web GUI for setup

2. TP-Link TL-SG105E / TL-SG108E (Easy Smart Switch)

    Ports: 5 or 8

    Price: ~$25–$45

    IGMP: Supports IGMP Snooping (v1/v2)

    Features: VLANs, QoS, basic management

    Note: Older versions required Windows utility, newer ones have web interface

3. Zyxel GS1200-5 / GS1200-8

    Ports: 5 or 8

    Price: ~$30–$50

    IGMP: IGMP Snooping v1/v2/v3

    Features: Web-based management, VLAN, QoS

    Web GUI: Yes, no software needed

4. D-Link DGS-1100-05 / DGS-1100-08

    Ports: 5 or 8

    Price: ~$40–$60

    IGMP: Supports v1/v2/v3 Snooping

    Features: Web-managed, VLANs, QoS, loopback detection

rsta2, 2025-05-25:
The IGMP support in Circle is making progress. I'm testing this with Wireshark 
at the moment. Circle is responding to Membership Queries, sends a Query by 
itself on joining a host group and a Leave Group on leaving it. This seems to 
be all OK according to Wireshark. I will push it to GitHub next week.

Thanks for this list. I have a TP-Link TL-SG605E here, which also supports 
IGMP Snooping, but I cannot see any effect in it. The multicast list only shows 
one entry, which does not come from myself. I think the reason is, that there 
is another router (my gate to the Internet) on my net, which supports IGMPv3 
and which has a smaller IP address then the TL-SG605E, so the other router is 
the querying router on my net. But I can see any multicasts to 224.0.0.251 
(mDNS/DNS-SD), with or without IGMP support in Circle. Perhaps I have to setup 
a small closed network for testing.

rsta2, 2025-05-26:
The IGMPv2 support is on the *develop* branch. Unfortunately I wasn't able to setup a test network, where the IGMP function was really verified. It was only tested using Wireshark. Hopefully we will see in the process of supporting Ableton Link (or #564), if problems occur with the IGMP support. I guess most users won't see a difference, regardless of IGMP is implemented or not.

rsta2, 2025-06-09:
I had a closer look now onto the Ableton Link source. I think somebody would have to port the Asio library to Circle first, which seems to be a huge effort to me. I don't think this will be available in the near future.

rsta2, 2026-03-18:
We (@smuehlst and myself) are working on porting Ableton Link to Circle again and we are making progress. Ableton Link including Asio and the linkhut tool can be built successfully already and runs to some degree. This is mainly possible with the improved POSIX-like network support in the [circle-stdlib project](https://codeberg.org/larchcone/circle-stdlib), which will allow to build Ableton Link without much modifications. More information will follow.