rsta2/circle #572: circle/boot - downloads not working (anymore?)

pottendo, 2025-05-08:
Hi,
i found it super-convenient to download the proper Pi bootfiles using:
cd circle/boot
make

however since a few days this seems to be broken somehow, I tested from 2 sites and 2 different Linux distros, and get this:
```
circle/boot$ make
rm -f bootcode.bin fixup*.dat start*.elf bcm*.dtb *.dtbo kernel*.img armstub*.bin LICENCE.broadcom COPYING.linux
make clean -C "../tools/bootloader"

[...]
Downloading bcm2712-rpi-cm5-cm5io.dtb ...
Downloading bcm2712-rpi-cm5l-cm5io.dtb ...
make: *** [Makefile:32: firmware] Error 8

circle/boot$ ls -l
total 3012
drwxrwxr-x 2 pottendo pottendo    4096 Mai  8 17:03 armstub
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2710-rpi-zero-2-w.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2711-rpi-400.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2711-rpi-4-b.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2711-rpi-cm4.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2712d0-rpi-5-b.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2712-rpi-500.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2712-rpi-5-b.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 bcm2712-rpi-cm5-cm5io.dtb
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:09 bcm2712-rpi-cm5l-cm5io.dtb
-rw-rw-r-- 1 pottendo pottendo   52476 Mai  8 17:08 bootcode.bin
-rw-rw-r-- 1 pottendo pottendo     476 Mai  8 17:03 config32.txt
-rw-rw-r-- 1 pottendo pottendo     734 Mai  8 17:03 config64.txt
-rw-rw-r-- 1 pottendo pottendo   18693 Mai  8 17:08 COPYING.linux
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 fixup4cd.dat
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 fixup4.dat
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 fixup_cd.dat
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 fixup.dat
-rw-rw-r-- 1 pottendo pottendo    1594 Mai  8 17:08 LICENCE.broadcom
-rw-rw-r-- 1 pottendo pottendo    3152 Mai  8 17:03 Makefile
-rw-rw-r-- 1 pottendo pottendo    3834 Mai  8 17:03 README
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 start4cd.elf
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 start4.elf
-rw-rw-r-- 1 pottendo pottendo       0 Mai  8 17:08 start_cd.elf
-rw-rw-r-- 1 pottendo pottendo 2984320 Mai  8 17:08 start.elf

```

Probably not a circle issue, but a bit weird. Some files can be loaded others not.
Some details:
```
Downloading bcm2712-rpi-cm5-cm5io.dtb ...
--2025-05-08 17:14:16--  https://github.com/raspberrypi/firmware/blob/4bb9d889a9d48c7335ebe53c0b8be83285ea54b1/boot/bcm2712-rpi-cm5-cm5io.dtb?raw=true
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 429 Too Many Requests
2025-05-08 17:14:16 ERROR 429: Too Many Requests.

```

Any idea?
thanks, pottendo

pottendo, 2025-05-08:
Ah thx, didn't see that - sorry for not checking properly. bye, pottendo

rsta2, 2025-05-09:
I think we should keep this open, so that other user can quickly find it. I will close it after the next release is done.

Andy2No, 2025-06-15:
I'm in the process of setting it up in Windows, following https://github.com/rsta2/circle/blob/master/doc/windows-build.txt and then https://github.com/rsta2/circle/blob/master/doc/bootloader.txt  This leads to the same problem - I got as far as 

make all

on line 39 of https://github.com/rsta2/circle/blob/master/doc/bootloader.txt, which caused this type of error (I actually got Error 1) with all the files that should have been downloaded left empty.

So, following the introductory documents leads to this impass, and having to find this issue for the solution.  Perhaps something in the docs needs to be updated so that people can currently get started?  Or is the plan to leave it like that and fix it with a new release, perhaps?

rsta2, 2025-06-15:
The Makefile in boot/ does currently not work on the master branch with GitHub, 
because they changed something on their platform.

Please checkout the develop branch and proceed as described. Checking out 
develop works like this:

git checkout develop

This will be fixed with the next release on master branch too. So it's not 
necessary to change the doc files.

rsta2, 2025-06-18:
This issue has been solved in Circle 49.0.1. It's not necessary to checkout *develop* any more.

closed 2025-06-18 by rsta2