rsta2/circle #580: Circle LVGL with SquareLines

naturalmechanics, 2025-05-23:
This is not exactly an error, but rather a tutorial request

Idea: 
[Squareline](https://squareline.io/) outputs UI.h and UI.c 
I need to somehow use these to render a UI using RPI CM4 emmc variant.

**Goal: 
Take the ui.c file (and its dependencies ) from squareline output and compile it using Circle LVGL to generate a bootable Kernel for RPI CM4 emmc.** 

I can run the lvgl sample code, as we discussed before. I can also render a simple image (half screen white, other half black) - but copying the large array from UI.c throws everything off. First I get 

```
m_pDisplay->GetDepth() == LV_COLOR_DEPTH 
``` 

(
I have also copied this part:
```
const lv_image_dsc_t ui_img_tacho_rpm_90_final_png = {
   .header.w = 800,
   .header.h = 800,
   .data_size = sizeof(ui_img_tacho_rpm_90_final_png_data),
   .header.cf = LV_COLOR_FORMAT_NATIVE_WITH_ALPHA,
   .header.magic = LV_IMAGE_HEADER_MAGIC,
   .data = ui_img_tacho_rpm_90_final_png_data};
```
)

I also tried this: 

1. I convert a image using [LVGL converter](https://lvgl.io/tools/imageconverter). This produced only a `*.c` file. Then, I modified the kernel.cpp 

```
extern "C" { // Ensure C linkage for LVGL symbols
  #include "lvgl.h" // Include LVGL headers if not already done
  extern const lv_img_dsc_t my_image; // Declare the image descriptor
}

// Use the image in your code
void show_image() {
  lv_obj_t *img = lv_img_create(lv_scr_act());
  lv_img_set_src(img, &my_image);
  lv_obj_align(img, LV_ALIGN_CENTER, 0, 0);
}
```

2. Then i place the image binary data in my_image.c (copied from ui.c)
3. I added `my_image.c`  to CSRCS in makefile of the sample folder under addon/lvgl/sample
4. I edited `lv_conf.h`

```
#define LV_USE_IMG 1
#define LV_COLOR_DEPTH 16
```
This throws the error: 

`LV_IMG_CF_RGB565 not defined`

Then I changed the cf  in the my_image.c file to LV_IMG_CF_RGB888 
But that led to a lot of errors in lvgl.c

I am completely thrown off course
Please can we have a tutorial of sorts to help me out of the situation? 

Thank you.

rsta2, 2025-05-23:
First you must not change the display depth from 16, otherwise you will get this failed assertion: `m_pDisplay->GetDepth() == LV_COLOR_DEPTH`. LVGL is configured to be used with RGB565 for HDMI displays.

Unfortunately I cannot talk about SquareLine, because I never used it, but I got your second try to work with this patch to the sample in *addon/lvgl/sample/*:

```
diff --git a/addon/lvgl/sample/Makefile b/addon/lvgl/sample/Makefile
index 45174b35..1384e2ba 100644
--- a/addon/lvgl/sample/Makefile
+++ b/addon/lvgl/sample/Makefile
@@ -9,7 +9,7 @@
 
 CIRCLEHOME = ../../..
 
-OBJS   = main.o kernel.o
+OBJS   = main.o kernel.o my_image.o
 
 LIBS   = $(CIRCLEHOME)/addon/lvgl/liblvgl.a \
          $(CIRCLEHOME)/lib/usb/libusb.a \
diff --git a/addon/lvgl/sample/kernel.cpp b/addon/lvgl/sample/kernel.cpp
index 93bdcb06..893d5288 100644
--- a/addon/lvgl/sample/kernel.cpp
+++ b/addon/lvgl/sample/kernel.cpp
@@ -156,6 +156,18 @@ static void lv_example_button (void)
 
 #endif
 
+extern "C" { // Ensure C linkage for LVGL symbols
+  #include "lvgl.h" // Include LVGL headers if not already done
+  extern const lv_img_dsc_t my_image; // Declare the image descriptor
+}
+
+// Use the image in your code
+static void show_image() {
+  lv_obj_t *img = lv_img_create(lv_scr_act());
+  lv_img_set_src(img, &my_image);
+  lv_obj_align(img, LV_ALIGN_CENTER, 0, 0);
+}
+
 TShutdownMode CKernel::Run (void)
 {
        m_Logger.Write (FromKernel, LogNotice, "Compile time: " __DATE__ " " __TIME__);
@@ -163,7 +175,8 @@ TShutdownMode CKernel::Run (void)
 #ifdef I2C_DISPLAY
        lv_example_button ();
 #else
-       lv_demo_widgets ();
+       // lv_demo_widgets ();
+       show_image ();
 #endif
 
        while (1)
```

You also need to create a new file *lvgl.h* in the sample's directory with this contents:

```
#include <lvgl/lvgl/lvgl.h>
```

Now, when you copy your converted image file *my_image.c* into the sample's directory, it should build and work.

rsta2, 2025-05-27:
Did this help? Regarding SquareLines you could post the generated UI.\* files of a simple example here, so that I can have a look at it.

naturalmechanics, 2025-05-27:
hi, tomorrow i will be back at this, Berlin time 1200.
sorry for the delay.

rsta2, 2025-06-06:
This was not a real Circle issue. I answered the question as good, as I was able to. So this can be closed.

closed 2025-06-06 by rsta2