rsta2/circle #601: Hostapd CVEs

soyersoyer, 2025-07-20:
Hi,
Circle uses a [fork](https://github.com/rsta2/hostap/tree/hostap_0_7_0-circle) of hostapd 0.7, which is affected by pretty much all [Hostapd CVEs](https://www.cvedetails.com/vulnerability-list/vendor_id-12005/product_id-22495/W1.fi-Hostapd.html?page=1&order=1).
This makes Wi-Fi enabled devices using Circle vulnerable.

rsta2, 2025-07-20:
Unfortunately I'm unable to solve this at the moment. Thanks for reporting!

**It's not recommended to use the WLAN support in Circle until this has been fixed!**

rsta2, 2025-07-23:
First something to clarify: For the Circle WLAN support the wpa_supplicant product is used, not hostapd. Both are build from the same code base, but with different options, so the resulting binary code is much different and different security advisories apply. wpa_supplicant for Circle is based on version 0.7.0 of this code base, so on cvedetails.com the following selection applies:

[CVEs for Wpa Supplicant 0.7.0](https://www.cvedetails.com/vulnerability-list/vendor_id-12005/product_id-29296/version_id-539687/W1.fi-Wpa-Supplicant-0.7.0.html?page=1&order=2)

On this website only the CVEs of the last ten years are listed, that's why I searched for "wpa_supplicant" also on cve.org:

[CVEs for wpa_supplicant](https://www.cve.org/CVERecord/SearchResults?query=wpa_supplicant)

This lists six additional CVEs, which all do not apply for wpa_supplicant 0.7.0.

Now one can try to identify specific CVEs to solve from the list, provided by cvedetails.com, but the problem is, that there is no relation form a CVE description to the wpa_supplicant codes, which is affected. That's why I came to the conclusion, that this is not the way to go.

Luckily there is a security advisory list on w1.fi, the website, where wpa_supplicant and hostapd is hosted:

[hostapd and wpa_supplicant security advisories](http://w1.fi/security/)

This provides a much more detailed description of the different vulnerabilities and patches for the source code to fix them. I analyzed all listed security advisories with this result:

[circle_wpa_supplicant_security_advisories.pdf](https://github.com/user-attachments/files/21389276/circle_wpa_supplicant_security_advisories.pdf)

The last column gives the reason, why an advisory does not affect the Circle WLAN support. This is mostly, because version 0.7.0 or wpa_supplicant at all is not affected, or because specific required build options are not enabled for Circle. I found only two advisories, which affect Circle:

* [2017-1](http://w1.fi/security/2017-1/)
* [2018-1](http://w1.fi/security/2018-1/)

Because FT (Fast Transition), WNM (Wireless Network Management) and TDLS (Tunneled Direct Link Setup) are not implemented or enabled in wpa_supplicant 0.7.0 for Circle, the advisory 2017-1 comes only with one source code patch (in two variants), which applies to Circle. Advisory 2018-1 has only patch (in two variants). Please see the .pdf for the patches.

The problem is, that these patches are for a newer source code base than version 0.7.0 and they cannot be directly applied, if they can be applied at all (manually). I'm still looking into this.

soyersoyer, 2025-08-08:
Thank you. This looks good and works well with WPA2.
I tried enabling WPA3-SAE, but the hostapd internal crypto library doesn't have all the bignum bindings yet.
One silly question: Why do you need to reimplement the stdlib functions (abs, isblank, isprint, isspace, strrchr, strdup, strtol, sscanf, snprintf, vsnprintf, printf, vprintf, qsort, abort)?  Can circle be used without stdlib?

rsta2, 2025-08-09:
Thanks for the feedback. I will try to enable WPA3-SAE, including the 
necessary modules for crypto. Yes, Circle can be used without the circle-
stdlib-project, which is an option.

rsta2, 2025-08-15:
The issue has been solved in Circle Step50. If you use the WLAN support, you should upgrade to this Circle release.