rsta2/circle #602: Outdated Broadcom Wi-Fi firmwares

soyersoyer, 2025-07-20:
The Broadcom Wi-Fi firmwares are from 2021, and there are newer versions of each.
I read somewhere that Broadcom only updates them when there is a security flaw, so presumably all of them are vulnerable.

rsta2, 2025-07-20:
Does this mean, that even Raspberry Pi OS is vulnerable in your opinion regarding to Wi-Fi access, or is the problem, that the firmware, which is downloaded in *addon/wlan/firmware/* in Circle is out-of-date?

rsta2, 2025-07-29:
The WLAN firmware has been updated on the *develop* branch. The new firmware does only work in conjunction with the modified ether4330 driver on the *develop* branch. Issue #601 still remains.

closed 2025-07-29 by soyersoyer