rsta2/circle #607: Possible OOM issue in Circle 50

IanCassTwo, 2025-08-24:
I'm trying to chase an issue in our application after we updated to Circle 50. It manifests as a ```heaplow: Out of memory``` error followed by a hard lock up. I've made extensive searches for memory leaks and there's nothing obvious. I've added debugging to the main kernel loop to print memory stats every 10 secs to track memory usage. Everything looks normal for 10mins or so then there'll be a 2min stall followed by some ```heaplow: Trying to free large block``` errors then a hard lock. Is this likely to be caused by anything added to Circle 50 or should I continue to debug the app? 

Log snippets below

Logging every 10 secs seems consistent up until the issue occurs.
```
Aug 23 18:41:16.49 heaplow: malloc(64): 163 blocks (max 169)
Aug 23 18:41:16.50 heaplow: malloc(1024): 67 blocks (max 76)
Aug 23 18:41:16.50 heaplow: malloc(4096): 8 blocks (max 28)
Aug 23 18:41:16.51 heaplow: malloc(16384): 5 blocks (max 5)
Aug 23 18:41:16.51 heaplow: malloc(65536): 36 blocks (max 36)
Aug 23 18:41:16.52 heaplow: malloc(262144): 4 blocks (max 4)
Aug 23 18:41:16.52 heaplow: malloc(524288): 0 blocks (max 0)
<snip>
Aug 23 18:52:39.50 heaplow: malloc(64): 163 blocks (max 169)
Aug 23 18:52:39.50 heaplow: malloc(1024): 60 blocks (max 76)
Aug 23 18:52:39.51 heaplow: malloc(4096): 8 blocks (max 28)
Aug 23 18:52:39.51 heaplow: malloc(16384): 5 blocks (max 5)
Aug 23 18:52:39.52 heaplow: malloc(65536): 36 blocks (max 36)
Aug 23 18:52:39.52 heaplow: malloc(262144): 4 blocks (max 4)
Aug 23 18:52:39.53 heaplow: malloc(524288): 0 blocks (max 0)
Aug 23 18:52:39.53 kernel: Heap Free Memory 453991904
Aug 23 18:54:46.50 heaplow: Trying to free large block (size 524332)
Aug 23 18:54:46.53 heaplow: Trying to free large block (size 524333)
Aug 23 18:54:46.55 heaplow: Trying to free large block (size 524418)
Aug 23 18:54:46.57 heaplow: Trying to free large block (size 524419)
Aug 23 18:54:46.59 heaplow: Trying to free large block (size 524504)
Aug 23 18:54:46.61 heaplow: Trying to free large block (size 524505)
<snipped 1642 lines>
Aug 23 18:55:03.97 heaplow: Trying to free large block (size 559937)
Aug 23 18:55:04.00 heaplow: Trying to free large block (size 560022)
Aug 23 18:55:04.02 heaplow: Trying to free large block (size 560023)
Aug 23 18:55:04.04 heaplow: malloc(64): 163 blocks (max 169)
Aug 23 18:55:04.04 heaplow: malloc(1024): 108 blocks (max 110)
Aug 23 18:55:04.04 heaplow: malloc(4096): 9 blocks (max 28)
Aug 23 18:55:04.05 heaplow: malloc(16384): 5 blocks (max 7)
Aug 23 18:55:04.06 heaplow: malloc(65536): 36 blocks (max 38)
Aug 23 18:55:04.06 heaplow: malloc(262144): 4 blocks (max 6)
Aug 23 18:55:04.07 heaplow: malloc(524288): 0 blocks (max 2)
heaplow: Out of memory

rsta2, 2025-08-24:
By default Circle can only reuse memory blocks of up to 512K after freeing. This can be configured with the system option [HEAP_BLOCK_BUCKET_SIZES](https://github.com/rsta2/circle/blob/f3b6515e0f116db7e76423594e46656aca7b2b1c/include/circle/sysconfig.h#L72). Because you are allocating memory blocks, which are slightly larger, you should add a bucket for 1M blocks by adding the following line to your *Config.mk* file:

```
DEFINE += -DHEAP_BLOCK_BUCKET_SIZES=0x40,0x400,0x1000,0x4000,0x10000,0x40000,0x80000,0x100000
```

This should fix the problem. Please rebuild all after adding the option.

IanCassTwo, 2025-08-24:
This was working without issue in Circle 49. Did this change in 50?

rsta2, 2025-08-24:
There were no changes according to dynamic memory management in Circle 50. But 
the newlib version in circle-stdlib has changed.

rsta2, 2025-08-27:
There is a fix for the problem on the *develop* branch. @IanCassTwo would you please test, if this solves the issue? Thanks.

IanCassTwo, 2025-08-27:
I think this is fixed. I've been running it for at least 45mins and I've seen this packet which is the one that caused it to crash before...

```
Aug 27 14:06:57.77 ParseResponse: pMsg: Dumping 0x27 bytes starting at 0xAAC74C
Aug 27 14:06:57.77 ParseResponse: pMsg: C74C: 05 70 72 6F 78 79 05 6C-6F 63 61 6C 00 00 1C 80
Aug 27 14:06:57.78 ParseResponse: pMsg: C75C: 01 00 00 00 78 00 10 FE-80 00 00 00 00 00 00 65
Aug 27 14:06:57.79 ParseResponse: pMsg: C76C: DC 79 A5 23 44 9A 79 7A-C0 0C 00 2F 80 01 00 00
Aug 27 14:06:57.80 mdnsd: Name is proxy
Aug 27 14:06:57.80 mdnsd: m_HostnameWithDomain is usbode.local
Aug 27 14:06:57.81 mdnsd: Name is proxy.local
Aug 27 14:06:57.81 mdnsd: m_HostnameWithDomain is usbode.local
Aug 27 14:06:57.82 mdnsd: RRTrailer.nType is 7168
```

I'll leave it running for a while longer but this is looking great so far.

rsta2, 2025-09-18:
Circle Step 50.0.1 has been released, which fixes this issue.