rsta2/circle #609: USB Ethernet w/RPI Zero 2W

davefilip, 2025-09-04:
I am interested in adding a USB Ethernet dongle to a Raspberry Pi Zero 2W.

First, anything I should look for in a USB Ethernet dongle beyond using a Realtek 8152 chip?  I believe your notes suggest:

- CDC Ethernet devices (RTL815x)

Although CDC (Communication Device Class) is not something that typically appears in the description when selling these devices, not sure if that is now "standard" or only supported on some hardware?

Also, can I assume that using an OTG USB Ethernet dongle is not a problem?  Or for a Zero would I need to use an OTG USB adapter (micro USB to USB A) and plug a USB Ethernet dongle into that (as I don't think I understand enough about which creates a new USB hub device, and whether that is a problem)?

For example, thinking of trying Smays OTG Micro USB Ethernet Adapter for Raspberry Pi Zero for sale on Amazon (know I can't post links in GitHub), but understand you might not be able to recommend any individual devices, rather just the overall device requirements.

Moreover, while I know that Circle does not do network routing, is there any way that I can use both WiFi and Ethernet at the same time, by manually opening a separate CSocket?  Or is that simply not supported.

Thanks in advance for any information you can provide.  My overall goal -- if it is possible -- is to do "normal" communications (Telnet, MQTT, HTTP, NTP, etc.) on WiFi as I currently do, but be able to open a socket on an Ethernet network and send TCP packets.  If that is not possible, I can also make all -- "normal" and socket -- communications through a single Ethernet connection work with some minor adjustments.

Cheers,

Dave.

davefilip, 2025-09-05:
> rsta2
>  left a comment
> (rsta2/circle#609)
> Hi, the USB CDC Ethernet support has been tested with RTL8152 and RTL8153 devices only (and with QEMU). This is a device class, which is specified by the official USB specs. Devices, which support it, are often offered with the info, that they work without a special driver on Windows. So more devices may work, but they have not been tested so far. In any case you should define the optionUSE_NAK_USB_FIX, if you connect them directly to a RPi Zero 2 W (without USB hub).
>
> Circle does not support the USB OTG (On-The-Go) specification, but normal OTG controllers should work in USB host mode, if there is a driver for the chipset, they use (or CDC, if this is used). Unfortunately I did not found "Smays OTG Micro USB Ethernet Adapter".
>
> No, simultaneous operation of two network interfaces is not supported by Circle.

Thanks Rene!

I purchased this on Amazon (which might not be available in Germany?):

https://www.amazon.com/dp/B01AT4C3KQ?ref=ppx_yo2ov_dt_b_fed_asin_title

which seems to work fine with Circle. It is sold as OTG, and does not EXPLICITLY mention CDC, but apparently provides that device class, because it works.  I did need to use:

	-DUSE_NAK_USB_FIX

for this to work, but there is not a lot of description on what this define parameter does:

sysconfig.h:// USE_NAK_USB_FIX enables a fix for SetCompleteOnNAK() in no-hub

So is this parameter just needed for USB OTG devices connected directly to a RPi Zero?  If I had a USB Ethernet device that plugged into USB-A, and used a micro-USB to USB cable on my RPi Zero (which I have, but I think *it* is called a USB OTG adapter cable?), would I still need this define?

Also, can I used this define when building non-RPi Zero images (e.g., for RPi3, RPi4, RPi5), or will it break USB devices on those models?

Sorry, have not tried all permutations, but trying to understand how to code my build scripts, when this parameter is needed, and if/when I can use it without breaking anything else?

Not asking you to test various permutations, just looking for a better understanding of what this define changes in Circle?

Thanks,

Dave.

rsta2, 2025-09-06:
USE_NAK_USB_FIX enables the detection of USB NAK packets. With this an USB 
transfer is aborted, when the device repondes with NAK and the driver returns, 
that nothing has been received. This should be generally enabled, but with USB 
MIDI it may lead to the loss of MIDI events for unknown reason. That's why it 
is an option.  It has nothing to do with OTG support. Circle does not support 
the OTG protocol. Use can define this option also for other RPi models. Only if 
you want to use USB MIDI, there would be an exception.

rsta2, 2025-09-06:
> so what does USE_NAK_USB_FIX do that permits this device to work?

Yes, USE_NAK_USB_FIX is required for your device, but not because of OTG. 
There is a problem of how NAK packets from an USB device are handled in the 
USB host controller driver for the RPi 1-3 and there are two options: one 
works with all devices, but may lead to the loss of MIDI events from USB MIDI 
devices and the other does not work with devices, which use USB Bulk-In-
Transfers in a synchronous way. This is very technical and I don't want to 
explain it here.

For you is only important: The option USE_NAK_USB_FIX must be defined and it 
works with your adapter. As long you do not use USB MIDI, this option can be 
generally defined. You will not run into problems because of this.

> Or is the issue that Circle does not support the OTG protocol, so that
> indirectly, this permits an OTG device to operate anyway, by discarding any
> data packets it doesn’t support?

Devices, which support USB OTG, can be used in a host or peripheral mode. The 
USB controller of the RPi 1-3 for example supports this. The OTG protocol 
allows to automatically detect, if an USB host or a peripheral is connected 
and the USB controller enters the right opposite mode then. But this is not 
supported by Circle. You have to decide, if you want to use the host or 
peripheral mode for the USB controller in the RPi 1-3, when you program your 
application. Your application uses the host mode most likely and you can 
connect USB devices, which run in peripheral mode to it.

Honestly I'm not aware, why this Ethernet adapter is named "... OTG ...", 
because I cannot imagine, that it can work in USB host mode. I guess this only 
applies to the plug on it and it does not support the OTG protocol too. So 
there won't be any problem with this adapter in terms of OTG support. It will 
work in the suggested configuration.

Rene

davefilip, 2025-09-06:
Thanks for the feedback.  No need to go into further details, I understand what I need to understand.

**_Honestly I'm not aware, why this Ethernet adapter is named "... OTG ...",
because I cannot imagine, that it can work in USB host mode._**

The device can work in either Host or Peripheral mode, based on the description on Amazon, so I presume it is working in Host mode, in order to work with Circle.  I suspect the 'OTG' in the Amazon description is related to the micro USB connector, and not needing a Micro USB to USB A OTG adapter (which I have, but am not using).  Nonetheless, I'm just glad that it works ... ;-)

Thanks for your feedback.  I am closing this ticket.  As always, thanks for your support, and cheers!

Dave.

closed 2025-09-06 by davefilip